espressif/esp-idf #6644: WIFI connect error WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT

kormilicinkostia, 2021-03-01:
- Module or chip used: ESP32-WROOM-32UE(16MB)
- IDF version V4.1
- Build System: CMake
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- environment type: VS Code
- Using an IDE?: VS Code
- Power Supply: external 3.3V

**Problem Description**
I have 2 esp32 devices. They are manually set to 2 mac-addresses 04: C3: E6: 02: 14: 7d and 04: C3: E6: 02: 14: 7e. A device with a mac address 04: C3: E6: 02: 14: 7d connects to Wi-Fi without any problems. But when connecting a device with mac-address 04: C3: E6: 02: 14: 7e, I get errors in the logs: WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT and WIFI_REASON_CONNECTION_FAIL. If I disconnect device 04: C3: E6: 02: 14: 7d, then my device 04: C3: E6: 02: 14: 7e connects to the network without any problems.

**Expected Behavior**
Both devices connect to the network

**Actual Behavior**
Only a device with a poppy address is connected 04: C3: E6: 02: 14: 7d

**Steps to reproduce**

1. Connect the device 04: C3: E6: 02: 14: 7d to the wifi 
2. Connect the device 04: C3: E6: 02: 14: 7e to the wifi 

**Code to reproduce this issue**

```
bool start_wifi(char* serial)
{
   uint8_t mac_addr[8] = {0x04,0xC3,0xE6,0x02,0x14,0x7e};
    esp_base_mac_addr_set(mac_addr);
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    initialise_mdns(serial);
    esp_netif_create_default_wifi_ap();
    esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &event_handler, NULL));
    esp_wifi_set_ps(WIFI_PS_NONE);
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_FLASH));

    char name_ap[32] = DEFAULT_AP_SSID;
    strncat (name_ap, serial,SERIAL_SIZE);

    ESP_LOGI(TAG,"esp_wifi_get_config AP %d", esp_wifi_get_config(ESP_IF_WIFI_AP, &wifi_config_ap));
    ESP_LOGI(TAG,"esp_wifi_get_config STA %d", esp_wifi_get_config(ESP_IF_WIFI_STA, &wifi_config_sta));

    if(strcmp((char*)wifi_config_ap.ap.ssid,name_ap)
    ||  wifi_config_ap.ap.ssid_len != strlen(name_ap)
    || strcmp((char*)wifi_config_ap.ap.password,DEFAULT_AP_PASS))
    {
        memset(wifi_config_ap.ap.ssid,0,sizeof(wifi_config_ap.ap.ssid));
        memcpy(wifi_config_ap.ap.ssid,name_ap,sizeof(name_ap));
        wifi_config_ap.ap.ssid_len = strlen((char*)wifi_config_ap.ap.ssid);
        wifi_config_ap.ap.authmode = WIFI_AUTH_WPA_PSK;
        memset(wifi_config_ap.ap.password,0,sizeof(wifi_config_ap.ap.password));
        memcpy(wifi_config_ap.ap.password,DEFAULT_AP_PASS,strlen(DEFAULT_AP_PASS)+1);
    }

    ESP_LOGI(TAG, "wifi_init_softap finished.SSID:%s password:%s",
             wifi_config_sta.sta.ssid, wifi_config_sta.sta.password);

    ESP_LOGI(TAG, "wifi_init_sta finished.");
    ESP_LOGI(TAG, "connect to ap SSID:%s password:%s",
             wifi_config_ap.ap.ssid, wifi_config_ap.ap.password);

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config_ap));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config_sta));
    ESP_ERROR_CHECK(esp_wifi_start());
    ESP_LOGI(TAG, "start esp_wifi_set_protocol");
    ESP_ERROR_CHECK(esp_wifi_set_protocol(WIFI_MODE_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N));

    return true;

}

```

negativekelvin, 2021-03-01:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html?#mac-address

In apsta mode you are using two mac addresses and creating an address conflict

zhangyanjiaoesp, 2021-03-02:
@kormilicinkostia 
Do you mean you have two devices, both in AP+STA mode, device1's mac address is: sta (04:c3:e6:02:14:7d) + softAP (04:c3:e6:02:14:7e), device2's mac address is : sta (04:c3:e6:02:14:7e) + softAP (04:c3:e6:02:14:7f). device1 connect to an AP (a router or a device3 with softAP mode), and then device2 try to connect to the same AP ?

kormilicinkostia, 2021-03-02:
@zhangyanjiaoesp zhangyanjiaoesp
Both devices are in AP + STA mode. Both devices are connected to the same router.device1's mac address is: sta (04:c3:e6:02:14:7d) + softAP (04:c3:e6:02:14:7e), device2's mac address is : sta (04:c3:e6:02:14:7e) + softAP (04:c3:e6:02:14:7f)

zhangyanjiaoesp, 2021-03-03:
@kormilicinkostia 
I have tested locally using your example code, but I didn't reproduce the problem. Here is my test info:
1. IDF v4.1, commit ad938156
2. Router: TP-LINK Archer C2, WPA2-PSK
3. Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

And my test log is here:
device1
> I (618) wifi:config nano formating: disabled
I (618) wifi:Init data frame dynamic rx buffer num: 32
I (618) wifi:Init management frame dynamic rx buffer num: 32
I (628) wifi:Init management short buffer num: 32
I (628) wifi:Init dynamic tx buffer num: 32
I (638) wifi:Init static rx buffer size: 1600
I (638) wifi:Init static rx buffer num: 10
I (638) wifi:Init dynamic rx buffer num: 32
I (648) wifi_init: rx ba win: 6
I (648) wifi_init: tcpip mbox: 32
I (648) wifi_init: udp mbox: 6
I (658) wifi_init: tcp mbox: 6
I (658) wifi_init: tcp tx win: 5744
I (668) wifi_init: tcp rx win: 5744
I (668) wifi_init: tcp mss: 1440
I (668) wifi_init: WiFi IRAM OP enabled
I (678) wifi_init: WiFi RX IRAM OP enabled
I (678) wifi:Set ps type: 0
I (688) wifi station: esp_wifi_get_config AP 0
I (688) wifi station: wifi_init_sta finished.
I (788) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (798) wifi:mode : sta (04:c3:e6:02:14:7d) + softAP (04:c3:e6:02:14:7e)
I (798) wifi:Total power save buffer number: 16
I (798) wifi:Init max length of beacon: 752/752
I (808) wifi:Init max length of beacon: 752/752
I (808) wifi station: start esp_wifi_set_protocol
I (928) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (928) wifi:state: init -> auth (b0)
I (938) wifi:state: auth -> assoc (0)
I (948) wifi:state: assoc -> run (10)
I (968) wifi:connected with test, aid = 1, channel 1, BW20, bssid = ac:84:c6:7e:31:2b
I (968) wifi:security: WPA2-PSK, phy: bgn, rssi: -45
I (978) wifi:pm start, type: 0
I (1138) wifi:AP's beacon interval = 204800 us, DTIM period = 1
I (1588) esp_netif_handlers: sta ip: 192.168.0.187, mask: 255.255.255.0, gw: 192.168.0.1
I (1588) wifi station: got ip:192.168.0.187
I (1588) wifi station: connected to ap SSID:test password:12345678

device2
> I (598) wifi:wifi driver task: 3ffc0d78, prio:23, stack:6656, core=0
I (618) wifi:wifi firmware version: bcf771b
I (618) wifi:config NVS flash: enabled
I (618) wifi:config nano formating: disabled
I (618) wifi:Init data frame dynamic rx buffer num: 32
I (618) wifi:Init management frame dynamic rx buffer num: 32
I (628) wifi:Init management short buffer num: 32
I (628) wifi:Init dynamic tx buffer num: 32
I (638) wifi:Init static rx buffer size: 1600
I (638) wifi:Init static rx buffer num: 10
I (638) wifi:Init dynamic rx buffer num: 32
I (648) wifi_init: rx ba win: 6
I (648) wifi_init: tcpip mbox: 32
I (658) wifi_init: udp mbox: 6
I (658) wifi_init: tcp mbox: 6
I (658) wifi_init: tcp tx win: 5744
I (668) wifi_init: tcp rx win: 5744
I (668) wifi_init: tcp mss: 1440
I (668) wifi_init: WiFi IRAM OP enabled
I (678) wifi_init: WiFi RX IRAM OP enabled
I (678) wifi:Set ps type: 0
I (688) wifi station: esp_wifi_get_config AP 0
I (688) wifi station: wifi_init_sta finished.
I (788) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (788) wifi:mode : sta (04:c3:e6:02:14:7e) + softAP (04:c3:e6:02:14:7f)
I (798) wifi:Total power save buffer number: 16
I (798) wifi:Init max length of beacon: 752/752
I (798) wifi:Init max length of beacon: 752/752
I (798) wifi station: start esp_wifi_set_protocol
I (928) wifi:ap channel adjust o:13,0 n:1,0
I (928) wifi:new:<1,0>, old:<13,0>, ap:<1,0>, sta:<1,0>, prof:13
I (928) wifi:state: init -> auth (b0)
I (938) wifi:state: auth -> assoc (0)
I (948) wifi:state: assoc -> run (10)
I (968) wifi:connected with test, aid = 2, channel 1, BW20, bssid = ac:84:c6:7e:31:2b
I (968) wifi:security: WPA2-PSK, phy: bgn, rssi: -46
I (978) wifi:pm start, type: 0
I (1108) wifi:AP's beacon interval = 204800 us, DTIM period = 1
I (1588) esp_netif_handlers: sta ip: 192.168.0.186, mask: 255.255.255.0, gw: 192.168.0.1
I (1588) wifi station: got ip:192.168.0.186
I (1588) wifi station: connected to ap SSID:test password:12345678

**Please provide your commit ID, router info and log info. I will try to test it based on your commit ID again, or you can try to update your v4.1 version to test.**

negativekelvin, 2021-03-03:
Even if it may work sometimes, you can't reuse Mac addresses like that.  That's why espressif assigns 4 Mac addresses per chip.

kormilicinkostia, 2021-03-03:
Here is my download log:
```
I (3828) wifi:wifi driver task: 3ffcd608, prio:23, stack:6656, core=0
I (3848) wifi:wifi firmware version: 3ea4c76
I (3848) wifi:config NVS flash: enabled
I (3848) wifi:config nano formating: disabled
I (3848) wifi:Init dynamic tx buffer num: 32
I (3848) wifi:Init data frame dynamic rx buffer num: 32
I (3858) wifi:Init management frame dynamic rx buffer num: 32
I (3858) wifi:Init management short buffer num: 32
I (3868) wifi:Init static rx buffer size: 1600
I (3868) wifi:Init static rx buffer num: 10
I (3868) wifi:Init dynamic rx buffer num: 32
I (3878) wifi:Set ps type: 0
I (3878) WIFI: esp_wifi_get_config AP 0
I (3888) WIFI: esp_wifi_get_config STA 0
I (3888) WIFI: wifi_init_softap finished.SSID:FisGoWIFI 
I (3898) WIFI: wifi_init_sta finished.
I (3898) WIFI: connect to ap SSID:viki_print_wifi_0493091627 
I (4008) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (4008) wifi:mode : sta (04:c3:e6:02:14:7e) + softAP (04:c3:e6:02:14:7f)
I (4008) wifi:Total power save buffer number: 16
I (4008) wifi:Init max length of beacon: 3008/3008
I (4018) wifi:Init max length of beacon: 3008/3008
I (4018) WIFI: start esp_wifi_set_protocol
I (4028) WIFI: SYSTEM_EVENT_STA_START
I (4158) wifi:ap channel adjust o:1,1 n:7,2
I (4158) wifi:new:<7,2>, old:<1,1>, ap:<7,2>, sta:<7,2>, prof:1
I (4158) wifi:state: init -> auth (b0)
I (4158) wifi:state: auth -> assoc (0)
I (4168) wifi:state: assoc -> run (10)
I (4378) wifi:state: run -> init (6c0)
I (4378) wifi:new:<7,0>, old:<7,2>, ap:<7,2>, sta:<7,2>, prof:1
I (4388) wifi:new:<7,0>, old:<7,0>, ap:<7,2>, sta:<7,2>, prof:1
E (4388) WIFI: connect to the AP fail 15
E (6838) WIFI: connect to the AP fail 205
```

I am using commit: 5ef1b390026270503634ac3ec9f1ec2e364e23b2
Router: zyxel keenetic ultra 2

zhangyanjiaoesp, 2021-03-03:
@kormilicinkostia 
According to your log `I (4378) wifi:state: run -> init (6c0)` ,it means the router send deauth to the ESP32 device. I think you need to change your mac address according to @negativekelvin 's suggestion, and test again.

closed 2021-04-22 by kormilicinkostia