espressif/esp-idf #6645: building

Mode79, 2021-03-01:
building the prebuilt examples is generating .a file but it give couple of errors as shown in picture
[image]

igrr, 2021-03-01:
This is Windows 10, right?
It's pretty odd that it can't find cmd.exe given the fact that the command prompt (cmd.exe) is already running.
Can you please run `echo %PATH%` and `where cmd` commands and see what they output?

Mode79, 2021-03-01:
right!

when i run the echo %PATH%, i got the following
C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files\CMake\bin;C:\MinGW\bin;C:\Program Files\dotnet\;C:\Users\mfawz\AppData\Local\Programs\Python\Python38-32\Scripts\;C:\Users\mfawz\AppData\Local\Programs\Python\Python38-32\;C:\Users\mfawz\AppData\Local\Microsoft\WindowsApps;C:\Users\mfawz\AppData\Local\particledev\bin;C:\Users\mfawz\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\mfawz\AppData\Local\GitHubDesktop\bin;C:\Users\mfawz\.dotnet\tools;C:\Users\mfawz\AppData\Roaming\npm;C:\Users\mfawz\AppData\Local\Box\Box Edit\;C:\Users\mfawz\.dotnet\tools

Mode79, 2021-03-01:
the problem is fixed once I added %SystemRoot%\system32 to the windows PATH

closed 2021-03-01 by Mode79