espressif/esp-idf #6646: esp_websocket_client_send_text() crashes

0xFEEDC0DE64, 2021-03-01:
Since I implemented remote coredump uploads for all the ESP32 at our customer's fields, I get to see a lot of crashes which have not been reproducable (yet) on a esp32 usb dev board, but maybe someone knows what I can look at and try?

I am sending text websocket messages to a https cloud server and sometimes `esp_websocket_client_send_text()` just crashes randomly somewhere in mbedtls code. I am sure that the buffers provided are valid and not deleted yet (memory owned by a c++ std::string, which gets deleted after `esp_websocket_client_send_text()` finishes)

esp-idf: 4.4-dev (plus a lot of my pull-requests merged already)

crashing thread stack:

```
==================== CURRENT THREAD STACK =====================
#0  0x4016a109 in mbedtls_debug_print_ret (ssl=0x3ffeade4, level=1, file=0x3f429b7c \"../esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c\", line=8752, text=0x3f42a434 \"mbedtls_ssl_write_record\", ret=-78) at ../esp-idf/components/mbedtls/mbedtls/library/debug.c:154
#1  0x40168d34 in ssl_write_real (len=8, buf=<optimized out>, ssl=0x3ffeade4) at ../esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:8752
#2  mbedtls_ssl_write (ssl=0x3ffeade4, buf=<optimized out>, len=<optimized out>) at ../esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:8830
#3  0x40169e35 in __wrap_mbedtls_ssl_write (ssl=0x3ffeade4, buf=0x3ffe6c10 \"\\201\\376\\001\\027\\\"\\204*O0\	\\006\", len=8) at ../esp-idf/components/mbedtls/port/dynamic/esp_ssl_tls.c:75
#4  0x4015e867 in esp_mbedtls_write (tls=0x3ffeade4, data=0x3ffe6c10 \"\\201\\376\\001\\027\\\"\\204*O0\	\\006\", datalen=8) at ../esp-idf/components/esp-tls/esp_tls_mbedtls.c:185
#5  0x4015f6a2 in esp_tls_conn_write (datalen=8, data=0x3ffe6c10, tls=<optimized out>) at ../esp-idf/components/esp-tls/esp_tls.h:476
#6  ssl_write (t=0x3ffe12a0, buffer=0x3ffe6c10 \"\\201\\376\\001\\027\\\"\\204*O0\	\\006\", len=8, timeout_ms=2000) at ../esp-idf/components/tcp_transport/transport_ssl.c:147
#7  0x401d0264 in esp_transport_write (t=0x3ffe12a0, buffer=0x3ffe6c10 \"\\201\\376\\001\\027\\\"\\204*O0\	\\006\", len=8, timeout_ms=2000) at ../esp-idf/components/tcp_transport/transport.c:178
#8  0x40160561 in _ws_write (t=<optimized out>, opcode=<optimized out>, b=0x3ffe1e2c <error: Cannot access memory at address 0x3ffe1e2c>, len=279, timeout_ms=2000, mask_flag=128) at ../esp-idf/components/tcp_transport/transport_ws.c:277
#9  0x401605f0 in esp_transport_ws_send_raw (t=0x3ffe136c, opcode=129, b=0x3ffe1e2c <error: Cannot access memory at address 0x3ffe1e2c>, len=279, timeout_ms=2000) at ../esp-idf/components/tcp_transport/transport_ws.c:305
#10 0x4012ec02 in esp_websocket_client_send_with_opcode (client=0x3ffe0b38, opcode=<optimized out>, data=0x3ffefa68 <error: Cannot access memory at address 0x3ffefa68>, len=279, timeout=<optimized out>) at ../esp-idf/components/esp_websocket_client/esp_websocket_client.c:868
#11 0x4012f82c in esp_websocket_client_send_text (client=0x3ffe0b38, data=0x3ffefa68 <error: Cannot access memory at address 0x3ffefa68>, len=279, timeout=200) at ../esp-idf/components/esp_websocket_client/esp_websocket_client.c:820
#12 0x401013c3 in espcpputils::websocket_client::send_text (this=0x3ffbad98 <goe::(anonymous namespace)::websocketClient+4>, timeout=200, len=<optimized out>, data=<optimized out>) at ../components/espcpputils/src/wrappers/websocket_client.h:27
#13 goe::(anonymous namespace)::cloud_send_task () at ../main/cloud.cpp:181

======================== THREADS INFO =========================
  Id   Target Id         Frame 
* 1    process 1073620544 0x4016a109 in mbedtls_debug_print_ret (ssl=0x3ffeade4, level=1, file=0x3f429b7c \"../esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c\", line=8752, text=0x3f42a434 \"mbedtls_ssl_write_record\", ret=-78) at ../esp-idf/components/mbedtls/mbedtls/library/debug.c:154
  2    process 1073626292 0x400892e6 in xQueueSemaphoreTake (xQueue=0x3ffaf03c, xTicksToWait=<optimized out>) at ../esp-idf/components/hal/esp32/include/hal/cpu_ll.h:39
  3    process 1073518944 0x401cf9be in esp_pm_impl_waiti () at ../esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
  4    process 1073521624 0x401cf9be in esp_pm_impl_waiti () at ../esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
  5    process 1073516264 vTaskDelay (xTicksToDelay=1) at ../esp-idf/components/freertos/tasks.c:1526
  6    process 1073506784 xEventGroupWaitBits (xEventGroup=0x3ffc6b48, uxBitsToWaitFor=16, xClearOnExit=0, xWaitForAllBits=1, xTicksToWait=5) at ../esp-idf/components/freertos/event_groups.c:415
  7    process 1073540484 0x4008477c in esp_crosscore_int_send_yield (core_id=0) at ../esp-idf/components/esp32/crosscore_int.c:112
  8    process 1073625524 xEventGroupWaitBits (xEventGroup=0x3ffe3748, uxBitsToWaitFor=2, xClearOnExit=1, xWaitForAllBits=1, xTicksToWait=100) at ../esp-idf/components/freertos/event_groups.c:415
  9    process 1073620124 xEventGroupWaitBits (xEventGroup=0x3ffe2468, uxBitsToWaitFor=2, xClearOnExit=1, xWaitForAllBits=1, xTicksToWait=100) at ../esp-idf/components/freertos/event_groups.c:415
  10   process 1073495940 0x400892e6 in xQueueSemaphoreTake (xQueue=0x3ffafe64, xTicksToWait=<optimized out>) at ../esp-idf/components/hal/esp32/include/hal/cpu_ll.h:39
  11   process 1073413172 0x4008477c in esp_crosscore_int_send_yield (core_id=0) at ../esp-idf/components/esp32/crosscore_int.c:112
  12   process 1073505528 0x4000bff0 in ?? ()
  13   process 1073549380 0x4008912a in xQueueReceive (xQueue=0x3ffcefe8, pvBuffer=0x3ffd0fa0, xTicksToWait=<optimized out>) at ../esp-idf/components/hal/esp32/include/hal/cpu_ll.h:39
  14   process 1073544908 0x4008477c in esp_crosscore_int_send_yield (core_id=0) at ../esp-idf/components/esp32/crosscore_int.c:112
  15   process 1073608920 0x4008477c in esp_crosscore_int_send_yield (core_id=0) at ../esp-idf/components/esp32/crosscore_int.c:112
  16   process 1073593572 0x4008477c in esp_crosscore_int_send_yield (core_id=0) at ../esp-idf/components/esp32/crosscore_int.c:112
  17   process 1073524048 0x4008477c in esp_crosscore_int_send_yield (core_id=0) at ../esp-idf/components/esp32/crosscore_int.c:112
  18   process 1073560464 0x4008477c in esp_crosscore_int_send_yield (core_id=0) at ../esp-idf/components/esp32/crosscore_int.c:112

==================== THREAD 2 (TCB: 0x3ffe3cb4, name: 'websocket_task') =====================
#0  0x400892e6 in xQueueSemaphoreTake (xQueue=0x3ffaf03c, xTicksToWait=<optimized out>) at ../esp-idf/components/hal/esp32/include/hal/cpu_ll.h:39
#1  0x40089371 in xQueueTakeMutexRecursive (xMutex=0x3ffaf03c, xTicksToWait=4294967295) at ../esp-idf/components/freertos/queue.c:681
#2  0x4008298d in lock_acquire_generic (lock=<optimized out>, delay=4294967295, mutex_type=4 '\\004') at ../esp-idf/components/newlib/locks.c:152
#3  0x40082a88 in _lock_acquire_recursive (lock=0x3ffaeee8) at ../esp-idf/components/newlib/locks.c:166
#4  0x401c2d95 in _vfprintf_r (data=0x3ffe3d14, fp=0x3ffaee90, fmt0=0x3f417fc4 \"\\033[0;32mI (%u) %s: Reconnect after %d ms\\033[0m\
\", ap=...) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853
#5  0x401c5df4 in vprintf (fmt=0x3f417fc4 \"\\033[0;32mI (%u) %s: Reconnect after %d ms\\033[0m\
\", ap=...) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34
#6  0x401d4c36 in esp_log_writev (level=<optimized out>, tag=<optimized out>, format=0x3f417fc4 \"\\033[0;32mI (%u) %s: Reconnect after %d ms\\033[0m\
\", args=...) at ../esp-idf/components/log/log.c:208
#7  0x4008dd01 in esp_log_write (level=ESP_LOG_INFO, tag=0x3f417f70 \"WEBSOCKET_CLIENT\", format=0x3f417fc4 \"\\033[0;32mI (%u) %s: Reconnect after %d ms\\033[0m\
\") at ../esp-idf/components/log/log.c:218
#8  0x4012e670 in esp_websocket_client_abort_connection (client=0x3ffe0b38) at ../esp-idf/components/esp_websocket_client/esp_websocket_client.c:165
#9  0x4012ea70 in esp_websocket_client_task (pv=0x3ffe0b38) at ../esp-idf/components/esp_websocket_client/esp_websocket_client.c:683
```

yuanjianmin, 2021-03-15:
Could you please apply this patch
[0001-mbedtls-Fix-mbedtls_ssl_send_alert_message-crash-due.zip](https://github.com/espressif/esp-idf/files/6140910/0001-mbedtls-Fix-mbedtls_ssl_send_alert_message-crash-due.zip)
and check whether could reproduce this issue.

0xFEEDC0DE64, 2021-03-29:
Thank you, I applied the patch and I haven't received such a crashdump from our test devices yet. But it also was rare before, so I am not 100% sure it's fixed. Would you mind explaining your change to the code? I took a quick look but didn't really know why that change fixes things :)

mahavirj, 2021-04-01:
@0xFEEDC0DE64 

This was speculative fix (based on information supplied over this issue). We are further looking into test code or scenario that can help us reproduce this problem.

- From issue report (backtrace), we understood that you might be using feature `CONFIG_MBEDTLS_DYNAMIC_BUFFER`, can you please re-confirm?
- Is it possible to share your test-code and sdkconfig file?

0xFEEDC0DE64, 2021-04-05:
We have `CONFIG_MBEDTLS_DYNAMIC_BUFFER` enabled, and I can share our sdkconfig, but I don't have a small code example ready to share, because I never experienced this crash on my developer setup, only very rare in production setups when the crashdumps were sent in over the internet.

We still haven't received such a crash stack anymore, so I guess your fix really works :)

[sdkconfig_hw4](https://github.com/espressif/esp-idf/files/6257243/sdkconfig_hw4.txt)
(keep in mind that that sdkconfig lists a lot of configs of my pull-requested changes to esp-idf itself, so running the menuconfig might change a lot, but mbedtls was not touched)

mahavirj, 2021-04-28:
Fix commit is now available at https://github.com/espressif/esp-idf/commit/93604b1a11fec5bb9c3d1ad419647c16319b839c. Please feel free to re-open in case you still observe the issue.

closed 2021-04-28 by mahavirj