espressif/esp-idf #6649: ESP32 hf cona; leads core dump

chwwu, 2021-03-02:
## Environment

- Development Kit :  ESP32-DevKitC
- Kit version : DevKitC v4
- Module or chip used : ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.2
- Build System:   idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [Plain Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
My goal is using ESP32 as audio gateway and connects to a BT headset(speaker), receives audio from BT headset, 
hf cona; will lead core dump

//Detailed problem description goes here.

### Expected Behavior
ESP32 acts as Audio gateway and receive/sent audio data sucessfully
### Actual Behavior
hf cona; will lead core dump

### Steps to reproduce

1. Download the esp-idf (v4.2) from https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/get-started/windows-setup.html

2. Enter idf.py menuconfig and:
    In Component config > Bluetooth > Bluetooth controller:
        Set BR/EDR Sync (SCO/eSCO) Max Connections to 1,
        Set BR/EDR Sync (SCO/eSCO) default data path to HCI,
    In Component config > Bluetooth > Bluedroid Options:
        Set Hands Free/Handset Profile audio(SCO) data path to HCI,
        Disable Wide Band Speech
    Save the changes and exit menuconfig.

3. in app_hf_msg_set.c uncomment line 16 and provide the Bluetooth address of the device you want to connect with.
4. Build the firmware and flash the ESP32.
5. Connect to your headset using hf con; and hf cona; commands in UART interface.

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

Just original HFP-AG example. I attached my temporary fix as a git patch file.

// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7100
load:0x40078000,len:13212
ho 0 tail 12 room 4
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (30) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (30) boot: compile time 18:51:17
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
[... 126 lines cut ...]
0x400927ba: multi_heap_assert at C:/Users/chwwu/esp/esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) get_prev_free_block at C:/Users/chwwu/esp/esp-idf/components/heap/multi_heap.c:194

0x40092d01: multi_heap_free_impl at C:/Users/chwwu/esp/esp-idf/components/heap/multi_heap.c:547

0x40081442: heap_caps_free at C:/Users/chwwu/esp/esp-idf/components/heap/heap_caps.c:305

0x400932e9: free at C:/Users/chwwu/esp/esp-idf/components/newlib/heap.c:47

0x400f16ca: bta_sys_event at C:/Users/chwwu/esp/esp-idf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:505

0x400edc68: osi_thread_run at C:/Users/chwwu/esp/esp-idf/components/bt/common/osi/thread.c:68

0x4008fae5: vPortTaskWrapper at C:/Users/chwwu/esp/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: 490a0cd3176f16bf

Rebooting...
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

chwwu, 2021-03-02:
I found the core dump log is similar as #6218 , but after patching [hfp_ag](https://github.com/espressif/esp-idf/issues/6218#issuecomment-765174168) the [core dump](https://github.com/espressif/esp-idf/issues/6218#issuecomment-753948294) still happened

chwwu, 2021-03-03:
Hi @ypiotrp 

This patch didn't work, but thanks for your kindly response.

Best regards,
Don

munho-espressif, 2021-03-03:
Hi @chwwu 
Where is your patch for your temporary fix?
Thanks

chwwu, 2021-03-08:
hi @munho-espressif 

Where is your patch for your temporary fix?
>> I didn't patch anything, just using v4.2 and the config changing I mentioned

I found this crash depends on HF devices, some can work some will lead crashing.

There are few more questions.
1. The ESP32 acts as AG, which HF devices is tested by your team and is stable for developing? 
2. the audio format is 8K/16K as sample rate which means 8000/16000 frames per second? and frames size is 16bit? Does the audio format can be set by hard code just for demo?
3. outgoing flow : PCM data sent in bt_app_hf_outgoing_cb-> CVSD codec -> transmit by BT to handsfree device
   incoming flow : handsfree device transmit CVSD format audio data via BT -> CVSD codec -> AG received PCM data in bt_app_hf_incoming_cb?
Am I correct about the flow?

Best,
Don

munho-espressif, 2021-03-08:
Hi @chwwu
Firstly, I will answer your question. 
1. We did not test devices that did not support 2edr and 3edr.
2. 8K and 16K represent 8K or 16K sampling points in a second, and audio format can be set in idf.py menuconfig
3. outgoin_cb calls back after the esp_hf_ougoingt_data_ready is called.
    incoming_cb is called after AG received audio data.
And then, I can't repodure your issue, I give you a debug patch, you can test again and provide reproduce log to me.
[debug_ag.zip](https://github.com/espressif/esp-idf/files/6099636/debug_ag.zip)

Thanks

chwwu, 2021-03-08:
Hi @munho-espressif 

We did not test devices that did not support 2edr and 3edr.
>>So, did you means if the HF devices support 2edr and 3edr, it is "stable" for AG developing? 

8K and 16K represent 8K or 16K sampling points in a second, and audio format can be set in idf.py menuconfig
>> How many bit per sampling points? and could u give me the config path (EX: In Component config > Bluetooth > Bluedroid Options:)

incoming_cb is called after AG received audio data
>> Could you give me the API acts as " AG received audio data"? by the way if I chose CVSD codec, the example just pass the pcm data without codec am I right? If I am correct, I can receive pcm data without any codec from HF devices via BT?

I will try the patch

Best,
Don

munho-espressif, 2021-03-08:
Hi @chwwu
1. if the HF devices support 2edr and 3edr, it is "stable" for AG developing
2. Per sampling points is 16bit.
3.  Component config > Bluetooth > Bluedroid Options > Wide Band Speech, If this option is enabled, hfp will use msbc(16k), otherwise cvsd(8k)
4. If you choose HCI data path, bta_ag_sco_read_cback will be called when ag receive data, and cvsd codec in controller, it isn't in host.
Thanks

chwwu, 2021-03-09:
Hi @munho-espressif 

The attachment is the log with your debug patch

[log.txt](https://github.com/espressif/esp-idf/files/6106383/log.txt)

If you choose HCI data path, bta_ag_sco_read_cback will be called when ag receive data, and cvsd codec in controller, it isn't in host.
>> This means we didn't consider about the codec, the bta_ag_sco_read_cback finally will call bt_app_hf_incoming_cb and the received data in bt_app_hf_incoming_cb is raw data (PCM data), am I right?

Best Regards,
Don

munho-espressif, 2021-03-09:
Hi @chwwu

> This means we didn't consider about the codec, the bta_ag_sco_read_cback finally will call bt_app_hf_incoming_cb and the received data in bt_app_hf_incoming_cb is raw data (PCM data), am I right?

Yes, the data we receive in host is decoded in Controller if the code is cvsd.

Thanks

munho-espressif, 2021-03-09:
Hi @chwwu
you can test again and provide reproduce log to me.
[debug_hfp.zip](https://github.com/espressif/esp-idf/files/6106773/debug_hfp.zip)

Thanks

chwwu, 2021-03-09:
Hi @munho-espressif 

The attachment is the log with your debug patch

[log.txt](https://github.com/espressif/esp-idf/files/6106914/log.txt)

Best Regards,
Don

munho-espressif, 2021-03-09:
Hi @chwwu
I'm sorry, but you need to test again and provide reproduce log to me.
[debug_hfp.zip](https://github.com/espressif/esp-idf/files/6107439/debug_hfp.zip)
Thanks

munho-espressif, 2021-03-09:
Hi @chwwu

>  A further question, what situation will lead the following message
W (18370) BT_BTM: SCO xmit Q overflow, pkt dropped
W (18380) BT_BTM: stat 10 

About this issue, you can try this patch
[hfp_warning_overflow.zip](https://github.com/espressif/esp-idf/files/6108408/hfp_warning_overflow.zip)
By the way, last time log.txt you give me, is the content of the same print all the time? Is there a log like this:

> ***<u> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Watchpoint 0 triggered </u>*** 
Core  0 register dump:
PC      : 0x400fa54a  PS      : 0x00060036  A0      : 0x800f6864  A1      : 0x3ffd07b0  
0x400fa54a: bta_sys_event at /home/liaowenhao/esp/esp-idf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:492
A2      : 0x3ffd0e60  A3      : 0x3ffd0e70  A4      : 0x3ffcf468  A5      : 0x00060023  
A6      : 0x00000001  A7      : 0x00060323  A8      : 0x800fa54a  A9      : 0x3ffd0790  
A10     : 0x00000000  A11     : 0x3ffd07b0  A12     : 0x00000004  A13     : 0x80000000  
A14     : 0x3ffcf588  A15     : 0x00001200  SAR     : 0x0000001e  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
Backtrace:0x400fa547:0x3ffd07b0 0x400f6861:0x3ffd07e0 0x4008fb2d:0x3ffd0800
0x400fa547: bta_sys_event at /home/liaowenhao/esp/esp-idf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:489
0x400f6861: osi_thread_run at /home/liaowenhao/esp/esp-idf/components/bt/common/osi/thread.c:68
0x4008fb2d: vPortTaskWrapper at /home/liaowenhao/esp/esp-idf/components/freertos/xtensa/port.c:143
ELF file SHA256: 0d23dc04fb30cd4b
Rebooting...
>
Thanks

chwwu, 2021-03-10:
Hi @munho-espressif

W (18370) BT_BTM: SCO xmit Q overflow, pkt dropped
W (18380) BT_BTM: stat 10
>> Could you explain why the pkt dropped?

By the way, last time log.txt you give me, is the content of the same print all the time? Is there a log like this:
>> The log always the same as I attached before event the backtrace addr the same too. 

Guru Meditation Error: Core 0 panic'ed (Unhandled debug exception).
>>Actually, I had tried this patch[](https://github.com/espressif/esp-idf/issues/6218#issuecomment-742931029), It will lead the error you mentioned, but the following solution mentioned in that issue didn't work for me.

Best,
Don

munho-espressif, 2021-03-10:
Hi @chwwu

- When the buffer space that send sco data is exhausted in controller, it will happen this situation.
- We don't analyze how much heap size the AG should use.
- You need to test again and provide reproduce log to me.
[debug_hfp.zip](https://github.com/espressif/esp-idf/files/6112925/debug_hfp.zip)
- In above debug_hf.patch, I add a function named `esp_set_watchpoint()`, it can watch the addr where change the content of `p_msg` in `bta_sys_event`, your issue is `p_msg` has been modified somewhere, so when free(p_msg), it will crash, you can modify `osi_free(p_msg)` to `osi_free(param)` in `bta_sys_event`, and see how it works.

Thanks

chwwu, 2021-03-10:
Hi @munho-espressif

When the buffer space that send sco data is exhausted in controller, it will happen this situation
>> Using the example code, It seems abnormal, right?

 We don't analyze how much heap size the AG should use
>> In other words, do you know, how much default memory is used when I enable AG ?

You need to test again and provide reproduce log to me.
>>
[log.txt](https://github.com/espressif/esp-idf/files/6113452/log.txt)

you can modify osi_free(p_msg) to osi_free(param)
>> It still crash
[log.txt](https://github.com/espressif/esp-idf/files/6113458/log.txt)

Best,
Don

chwwu, 2021-03-15:
Hi @munho-espressif 

Any update about this?

Best,
Don

munho-espressif, 2021-03-16:
Hi @chwwu 
You need test again and provide reproduce log to me.
[debug_hfp.zip](https://github.com/espressif/esp-idf/files/6147449/debug_hfp.zip)
And you can  increase the heap corruption detection level can give more detailed information about heap corruption errors.
path:
Component config → Heap memory debugging → Heap corruption detection->Light Impact
reference: https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-reference/system/heap_debug.html#light-impact
Thanks

chwwu, 2021-03-17:
Hi @munho-espressif 

That crash becomes minor now, but there is a question,

If I want to receive the volume up/down event from speaker when the volume up/down button on speaker is pressed
Is this possible to do this on AG?
if it is possible, could you tell me how?
I want to use that event to develop.

Best,
Don

munho-espressif, 2021-03-17:
Hi @chwwu

If you are on a call, AG can receive the volume control event, otherwise, other volume control is not available like media volume control because HFP profile only deals with the volume of the call.
When spaker adjusts volume, AG callback will receive a event called **ESP_HF_VOLUME_CONTROL_EVT**, you can register callback to receive the event. In AG example, you can see how to use the event. 
path: esp-idf/examples/bluetooth/bluedroid/classic_bt/hfp_ag/main/bt_app_hf.c

chwwu, 2021-03-17:
Hi @munho-espressif 

I have added the following code on callback event.

`        case ESP_HF_VOLUME_CONTROL_EVT:
        {
            printf("Don volune test\n");
            break;
        }`

when [hf con;]
Connect.
the log prints one time.
I (8859) BT_APP_HF: --Volume Target: SPEAKER, Volume 10
Don volune test

And after hf cona;
Connect Audio
I (17809) BT_APP_HF: APP HFP event: AUDIO_STATE_EVT
I (17809) BT_APP_HF: --Audio State connecting
E (17809) BT_BTM: btm_sco_connected, handle 181
I (17809) BT_APP_HF: APP HFP event: AUDIO_STATE_EVT
I (17819) BT_APP_HF: --Audio State connected

When I press volume up/down button, the log didn't show, but the volume up/down works on speaker.
So I think this callback didn't receive the volume control event from speaker.
I am sure the speaker will sent event, because I connect with pc, the volume up/down UI will show on the pc when speaker up/down button is pressed

Best,
Don

munho-espressif, 2021-03-17:
Hi @chwwu
Are you on the phone? Events can only receive when you are on the phone.

Thanks

chwwu, 2021-03-17:
Hi @munho-espressif 

My use case is esp32 act as audio gateway and connect to the hands free speaker (via hf cona;), and when I press the volume up/down button, I expect esp32 could receive some event.

Is that case matches "on the phone" you mentioned?  As I understanding, after hf cona; the connection is built and is "on the phone"

If no, is there another method I could receive the button event?

Best,
Don

munho-espressif, 2021-03-17:
Hi @chwwu

“On the phone" means you are calling. I try to use equipment to capture airlog, I find the headset does't sent data about volume control If you're not on the phone, even audio had connected. You may be can use AG to control volume by using `esp_bt_hf_volume_control`

Thanks

Alvin1Zhang, 2021-07-27:
Thanks for reporting, will close now, feel free to reopen.

closed 2021-07-27 by Alvin1Zhang