espressif/esp-idf #6650: ESP32-S2 WPS Example Problem

yusufbavas, 2021-03-02:
## Environment

- Development Kit:      ESP32-S2 Saola-1
- Kit version v1.2
- Module or chip used:  ESP32-S2-WROOM-1
- IDF version v4.3-beta1 and v4.2
- Compiler version xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- Using an IDE: Visual Studio Code with esp-idf extension
- Power Supply:         USB

## Problem Description

I face a problem on using WPS connection. I use wps example in github(https://github.com/espressif/esp-idf/tree/master/examples/wifi/wps). Board can find wps connection but it can't connect it. I tried esp-idf v4.2 and v4.3-beta1. 
Both has same issue. What could be the reason of this problem.

[image]

sagb2015, 2021-03-10:
@yusufbavas We have tested the WPS functionality on 4.2 and latest branch and there does not seem to be any problem. Can you help to run the same with supplicant debug logs enabled and provide the console logs?

Alvin1Zhang, 2021-03-16:
@yusufbavas Thanks for reporting, would you please help share if any updates for this issue? Thanks.

yusufbavas, 2021-03-22:
Sorry for late reply. I still have same problem. I tried latest branch and master branch again but result is same. I upload program in secure mode with --no-stub option. Can it be the reason of this problem? 

Here is my console log. I enabled some debug features but I am not fimiliar with esp idf debug options. I can enable or disable some options if you want. Its seems like board get name of the network but it cant get the password. Logs says password is empty.

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x22e0
load:0x4004c000,len:0xb5c
load:0x40050000,len:0x3334
SHA-256 comparison failed:
Calculated: c0e44b984a7122f885bb3184d8e8b061e44c2ee16287b1d4e60b0a2d6f0e7c4f
Expected: b4f114a6fe54dd6dc1dd3454d0ec2efaa45e9efb7d3f8c6eb495ce5127db1cd4
Attempting to boot anyway...
entry 0x4004c22c
I (68) boot: ESP-IDF v4.3-beta1 2nd stage bootloader
I (68) boot: compile time 15:12:25
D (68) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (73) boot: chip revision: 0
D (77) boot.esp32s2: magic e9
D (80) boot.esp32s2: segments 04
D (83) boot.esp32s2: spi_mode 02
D (86) boot.esp32s2: spi_speed 0f
D (90) boot.esp32s2: spi_size 02
I (93) boot.esp32s2: SPI Speed      : 80MHz
I (98) boot.esp32s2: SPI Mode       : DIO
I (103) boot.esp32s2: SPI Flash Size : 4MB
D (107) boot: Enabling RTCWDT(9000 ms)
I (111) boot: Enabling RNG early entropy source...
D (117) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0
D (126) boot: mapped partition table 0x8000 at 0x3f008000
D (131) flash_parts: partition table verified, 4 entries
I (137) boot: Partition Table:
I (140) boot: ## Label            Usage          Type ST Offset   Length
D (148) boot: load partition table entry 0x3f008000
D (153) boot: type=1 subtype=2
I (156) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (163) boot: load partition table entry 0x3f008020
D (168) boot: type=1 subtype=1
I (171) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (179) boot: load partition table entry 0x3f008040
D (184) boot: type=0 subtype=0
I (187) boot:  2 factory          factory app      00 00 00010000 00100000
I (195) boot: End of partition table
D (199) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (205) esp_image: reading image header @ 0x10000
D (210) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (217) esp_image: image header: 0xe9 0x06 0x02 0x01 400231f8
V (222) esp_image: loading segment header 0 at offset 0x10018
V (228) esp_image: segment data length 0x189cc data starts 0x10020
V (235) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3f000020
I (243) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=189cch (100812) map
D (251) esp_image: free data page_count 0x0000003f
D (256) bootloader_flash: mmu set paddr=00010000 count=2 size=189cc src_addr=10020 src_addr_aligned=10000
V (285) esp_image: loading segment header 1 at offset 0x289ec
D (285) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
V (286) esp_image: segment data length 0x3f04 data starts 0x289f4
V (293) esp_image: segment 1 map_segment 0 segment_data_offs 0x289f4 load_addr 0x3ffc6520
I (301) esp_image: segment 1: paddr=000289f4 vaddr=3ffc6520 size=03f04h ( 16132) load
D (309) esp_image: free data page_count 0x0000003f
D (314) bootloader_flash: mmu set paddr=00020000 count=1 size=3f04 src_addr=289f4 src_addr_aligned=20000
V (327) esp_image: loading segment header 2 at offset 0x2c8f8
D (330) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
V (336) esp_image: segment data length 0x404 data starts 0x2c900
V (343) esp_image: segment 2 map_segment 0 segment_data_offs 0x2c900 load_addr 0x40022000
0x40022000: _WindowOverflow4 at C:/Users/Administrator/esp/esp-idf-v4.3-beta1/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1730
[... 715 lines cut ...]
D (9702) wifi:clear scan ap list
D (9712) event: running post WIFI_EVENT:5 with handler 0x40095008 and context 0x3ffd44a8 on loop 0x3ffd3278
0x40095008: wifi_default_action_sta_disconnected at C:/Users/Administrator/esp/esp-idf-v4.3-beta1/esp-idf/components/esp_wifi/src/wifi_default.c:107

D (9722) esp_netif_handlers: esp_netif action disconnected with netif0x3ffd41a0 from event_id=5
D (9732) esp_netif_lwip: check: remote, if=0x3ffd41a0 fn=0x4008beac
0x4008beac: esp_netif_down_api at C:/Users/Administrator/esp/esp-idf-v4.3-beta1/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1226

D (9732) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffd41a0
D (9742) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffd41a0
D (9752) esp_netif_lwip: if0x3ffd41a0 start ip lost tmr: no need start because netif=0x3ffd4220 interval=120 ip=0
D (9762) esp_netif_lwip: check: local, if=0x3ffd41a0 fn=0x4008c604
0x4008c604: esp_netif_update_default_netif_lwip at C:/Users/Administrator/esp/esp-idf-v4.3-beta1/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:174

D (9762) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffd41a0
V (9772) esp_netif_objects: esp_netif_next_unsafe 0x0
V (9772) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffd41a0
V (9782) esp_netif_objects: esp_netif_next_unsafe 0x3ffd41a0
D (9792) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (9792) event: running post WIFI_EVENT:5 with handler 0x4008636c and context 0x3ffd4d8c on loop 0x3ffd3278
0x4008636c: wifi_event_handler at C:\Users\Administrator\Desktop\ESP32S2\wps-v2\wps\build/../main/wps.c:57

I (9802) example_wps: WIFI_EVENT_STA_DISCONNECTED
I (9812) example_wps: Failed to connect!
```

nishanth-radja, 2021-09-21:
@yusufbavas can you also pls share the AP model and  configuration of the AP which you are using for the WPS connection.Also Pls share the wireless sniffer capture during the failure of the wps connection.

AxelLin, 2022-02-13:
@yusufbavas 
Please check if v4.3 branch works or not.
I'm wondering if this is related to https://github.com/espressif/esp-idf/issues/8087
The fix is in v4.3 branch but not yet in v4.4 and v4.2 branch.

nishanth-radja, 2022-02-15:
@yusufbavas, As AxelLin suggested pls try on 4.3 branch, 

From the logs shared looks like the wps handshake failed during the building of M1 packet. 
"D (4392) wifi:wps handle eapol succeed
D (4452) wifi:wps handle eapol succeed
D (5092) wifi:wps handle eapol succeed
V (5102) esp_adapter: thread sem get: sem=0x3ffdf0b8
D (5102) event: running post WIFI_EVENT:8 with handler 0x4008636c and context 0x3ffd4d8c on loop 0x3ffd3278
0x4008636c: wifi_event_handler at C:\Users\Administrator\Desktop\ESP32S2\wps-v2\wps\build/../main/wps.c:57

I (5102) example_wps: WIFI_EVENT_STA_WPS_ER_FAILED"

We are supposed to have "wps handle eapol succeed" 8 times for a successful wps session.Sniffer capture will help.

closed 2022-11-16 by Alvin1Zhang