espressif/esp-idf #6651: pppos_client example sending SMS results in "unknown line received >"

kluverp, 2021-03-02:
When running the "pppos_client" example on a BG96 modem, I run into the following issues:

1)
I cannot set rx number 34 in menuconfig. I know gpio pin 34 is a read-only pin. But I've used it with success as a rx pin number. Is this not a recommended pin number?

2)
The retrieval of the "Operator" is slow on my device, causing the function to return "+COPS: 0" this is somehow not recognized as a valid response on the "bg96_get_operator_name()" function. It seems a legitimate response to me, that there are no Oparators to choose from.
When I run the "bg96_get_operator_name()" in loop, after 5 times or so, I do get an operator.

``` 
I (475) uart: queue free spaces: 30
I (625) bg96: PIN set ok
E (635) esp-modem: esp_dte_handle_line(114): handle line failed
W (635) pppos_example: Unknow line received: +COPS: 0

I (645) pppos_example: Module: BG96
```

3)
When sending the SMS, then example gives the following error:

```
I (17385) pppos_example: User interrupted event from netif:0x0
I (17385) esp-modem-netif: PPP state changed event 5
E (17405) esp-modem: esp_modem_dte_send_wait(360): wait prompt [
> ] timeout                                                                                                   
E (18055) esp-modem: esp_dte_handle_line(114): handle line failed
W (18055) pppos_example: Unknow line received: > 

I (18055) pppos_example: Send send message [Welcome to ESP32!] ok
I (18075) pppos_example: Power down
```
I've tracked the error, and somehow in the function "esp_modem_dte_send_wait()" in the file "esp_modem.c", the uart pattern detection is switched off (is OK). Then the uart sends the "AT+CMGS" command, which also works fine. Then the "uart_read_bytes()" function receives nothing. Even if I increase the timeout.
After that, the pattern detection is switched on again, and the interrupt fires, receiving the ">" char from the modem. I don't know why this is. 
That's where the "Unknown line" warning is coming from, since there is no handler for "> ".

The SMS gets send anyway, but the above is strange behavior.

------------------

- Development Kit:      none (custom hardware)
- Kit version NA
- Module or chip used:  ESP32-WROOM-32E
- IDF version : both 4.2 and master branch
- Build System:         idf.py
- Compiler version: crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?:  No
- Power Supply:         USB & external 3.3V

david-cermak, 2021-03-04:
@kluverp Could you please check if this patch fixes your issue with sending SMS?

https://github.com/espressif/esp-idf/issues/6428#issuecomment-765370001

kluverp, 2021-03-04:
> @kluverp Could you please check if this patch fixes your issue with sending SMS?
> 
> [#6428 (comment)](https://github.com/espressif/esp-idf/issues/6428#issuecomment-765370001)

Thanks for your reply. 
Yes, this patch seems to work. 
I've added the improvements made by @jonathandreyer as well.

This seems to solve the failing "send_wait()" function as mentioned in my issue # 3.

Issue # 2 I've mentioned above still remains.

yuanjianmin, 2021-03-05:
@kluverp For issue#2, could you have another BG96 modem to test it. Or you can send the command `AT+COPS?` separately to query the operator and see the content of the reply.

lhespress, 2021-03-05:
@kluverp For issue#2, you can reference the code as follows which Waiting for Operator success:

```
        ESP_LOGI(TAG, "Battery voltage: %d mV", voltage);
        /* Waiting for Operator */
        int retry = 0;
        const int retry_count = 10;
        while (!strlen(dce->oper) && ++retry < retry_count) {
            ESP_ERROR_CHECK(dce->get_operator_name(dce));
            ESP_LOGI(TAG, "Waiting for Operator: %s (%d/%d)", dce->oper, retry, retry_count);
        }
        /* setup PPPoS network parameters */
```

kluverp, 2021-03-08:
> @kluverp For issue#2, could you have another BG96 modem to test it. Or you can send the command `AT+COPS?` separately to query the operator and see the content of the reply.

Here is some debugged output:
```
I (8859) bg96: get imei number ok
I (8859) esp_modem_dte_send_cmd: command: AT+CIMI
W (8909) esp-modem: esp_handle_uart_pattern: '
'                                                                                                             
W (8959) esp-modem: esp_handle_uart_pattern: '204086457032097
'                                                                                                             
W (9009) esp-modem: esp_handle_uart_pattern: '
'                                                                                                             
W (9059) esp-modem: esp_handle_uart_pattern: 'OK
'                                                                                                             
I (9059) esp_modem_dte_send_cmd: command: AT+COPS?
W (9109) esp-modem: esp_handle_uart_pattern: '
'                                                                                                             
E (9159) esp-modem: esp_dte_handle_line(91): handle line failed
W (9159) esp-modem: esp_handle_uart_pattern: '+COPS: 0
'                                                                                                             
W (9159) modem: Unknown line received: |+COPS: 0
|                                                                                                             
W (9209) esp-modem: esp_handle_uart_pattern: '
'                                                                                                             
W (9259) esp-modem: esp_handle_uart_pattern: 'OK
'                                                                                                             
I (9259) esp_modem_dte_send_cmd: command: AT+IFC=0,0
W (9309) esp-modem: esp_handle_uart_pattern: '
'                                                                                                             
W (9359) esp-modem: esp_handle_uart_pattern: 'OK
'                                                                                                             
I (9359) esp_modem_dte_send_cmd: command: AT&W
W (9409) esp-modem: esp_handle_uart_pattern: '

```

kluverp, 2021-03-08:
> @kluverp For issue#2, you can reference the code as follows which Waiting for Operator success:
>
> ```
>         ESP_LOGI(TAG, "Battery voltage: %d mV", voltage);
>         /* Waiting for Operator [...]

Yes, this is exactly what I did!
And it works, the 3rd or 4th time in the loop the command succeeds.

The reason I metioned the issue, is that a response of "+COPS: 0" is causing an error. In my opinion this response is valid, although it is an empty response.

yuanjianmin, 2021-03-15:
@kluverp Unfortunately,now we could only cyclic query the operator until the BG96 response the correct name. Do you have any other problem, if not, please help close this issue. Thanks.

closed 2021-04-15 by espressif-bot