espressif/esp-idf #6653: [ESP32-C3] uart_read_bytes misbehaves ?

tobozo, 2021-03-02:
- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      [ESP32-C3-Dev]
- Module or chip used:  [ESP32-C2]
- IDF version : v4.4-dev-4-g73db142
- Build System:         [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux|
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

`uart_read_bytes` misbehaves

### Expected Behavior

buffer is filled with data

### Actual Behavior

buffer is filled with zeroes

### Code to reproduce this issue

See this [example code](https://gist.github.com/tobozo/398ed7acb65fb5f143a3b9c1a2805372) tested on ESP32 and failing on ESP32-C3

ginkgm, 2021-03-12:
Hi @tobozo ,

We have confirmed that on C3, when the configured baudrate is below about 19500, the registers are not configured correctly.

We have a fix for it in commit (fc61e60948e22e5e866e7a3e12e0dc6959dc09a1 or 7dca6b742815d29ebb2531981b134445b0224711 on v4.3 branch). But you may need for some time until it's synced to Github.

Here's the patch, you can have a try first. We will close this issue after some day. If you still have problems, please feel free to reopen it.

diff --git a/components/hal/esp32c3/include/hal/uart_ll.h b/components/hal/esp32c3/include/hal/uart_ll.h
index a06cc72d7090604047838d1c4c461c37ba59d7aa..931134f71f8488b0011bf5b6706736704e1e83dc 100644
--- a/components/hal/esp32c3/include/hal/uart_ll.h
+++ b/components/hal/esp32c3/include/hal/uart_ll.h
@@ -145,20 +145,24 @@ static inline uint32_t uart_ll_get_sclk_freq(uart_dev_t *hw)
  * @brief  Configure the baud-rate.
  *
  * @param  hw Beginning address of the peripheral registers.
- * @param  baud The baud rate to be set. When the source clock is APB, the max baud rate is `UART_LL_BITRATE_MAX`
+ * @param  baud The baud rate to be set.
  *
  * @return None
  */
 static inline void uart_ll_set_baudrate(uart_dev_t *hw, uint32_t baud)
 {
-    const int sclk_div = 1;
+#define DIV_UP(a, b)    (((a) + (b) - 1) / (b))
     uint32_t sclk_freq = uart_ll_get_sclk_freq(hw);
-    uint32_t clk_div = ((sclk_freq) << 4) / baud;
+    const uint32_t max_div = BIT(12) - 1;   // UART divider integer part only has 12 bits
+    int sclk_div = DIV_UP(sclk_freq, max_div * baud);
+
+    uint32_t clk_div = ((sclk_freq) << 4) / (baud * sclk_div);
     // The baud rate configuration register is divided into
     // an integer part and a fractional part.
     hw->clk_div.div_int = clk_div >> 4;
     hw->clk_div.div_frag = clk_div &  0xf;
-    hw->clk_conf.sclk_div_num = sclk_div - 1;//7;//255;
+    hw->clk_conf.sclk_div_num = sclk_div - 1;
+#undef DIV_UP
 }
 
 /**
@@ -172,7 +176,7 @@ static inline uint32_t uart_ll_get_baudrate(uart_dev_t *hw)
 {
     uint32_t sclk_freq = uart_ll_get_sclk_freq(hw);
     typeof(hw->clk_div) div_reg = hw->clk_div;
-    return ((sclk_freq << 4)) / ((div_reg.div_int << 4) | div_reg.div_frag);
+    return ((sclk_freq << 4)) / (((div_reg.div_int << 4) | div_reg.div_frag) * (hw->clk_conf.sclk_div_num + 1));
 }
 
 /**
diff --git a/components/hal/esp32s3/include/hal/uart_ll.h b/components/hal/esp32s3/include/hal/uart_ll.h
index c5a167b2981813a6a8f17586fcc342798f48c958..7e5baeef2c1f2cda15463350d3850925d0317a18 100644
--- a/components/hal/esp32s3/include/hal/uart_ll.h
+++ b/components/hal/esp32s3/include/hal/uart_ll.h
@@ -129,20 +129,24 @@ static inline uint32_t uart_ll_get_sclk_freq(uart_dev_t *hw)
  * @brief  Configure the baud-rate.
  *
  * @param  hw Beginning address of the peripheral registers.
- * @param  baud The baud rate to be set. When the source clock is APB, the max baud rate is `UART_LL_BITRATE_MAX`
+ * @param  baud The baud rate to be set.
  *
  * @return None
  */
 static inline void uart_ll_set_baudrate(uart_dev_t *hw, uint32_t baud)
 {
-    uint32_t sclk_freq, clk_div;
+#define DIV_UP(a, b)    (((a) + (b) - 1) / (b))
+    uint32_t sclk_freq = uart_ll_get_sclk_freq(hw);
+    const uint32_t max_div = BIT(12) - 1;   // UART divider integer part only has 12 bits
+    int sclk_div = DIV_UP(sclk_freq, max_div * baud);
 
-    sclk_freq = uart_ll_get_sclk_freq(hw);
-    clk_div = ((sclk_freq) << 4) / baud;
+    uint32_t clk_div = ((sclk_freq) << 4) / (baud * sclk_div);
     // The baud rate configuration register is divided into
     // an integer part and a fractional part.
     hw->clk_div.div_int = clk_div >> 4;
     hw->clk_div.div_frag = clk_div &  0xf;
+    hw->clk_conf.sclk_div_num = sclk_div - 1;
+#undef DIV_UP
 }
 
 /**
@@ -156,7 +160,7 @@ static inline uint32_t uart_ll_get_baudrate(uart_dev_t *hw)
 {
     uint32_t sclk_freq = uart_ll_get_sclk_freq(hw);
     typeof(hw->clk_div) div_reg = hw->clk_div;
-    return ((sclk_freq << 4)) / ((div_reg.div_int << 4) | div_reg.div_frag);
+    return ((sclk_freq << 4)) / (((div_reg.div_int << 4) | div_reg.div_frag) * (hw->clk_conf.sclk_div_num + 1));
 }
 
 /**

tobozo, 2021-03-12:
Hey Michael, thanks for the prompt reply :+1: 

I have patched my local `uart_ll.h` file with the diff you provided and I can confirm this fixed the problem with my GPS at 9600 bauds :tada: 

[image]

Closing this as solved

be well,

tobozo

Alvin1Zhang, 2021-04-06:
Thanks for reporting, fix on master and release/4.3 already available on GitHub, feel free to reopen. Thanks.

closed 2021-04-06 by Alvin1Zhang