espressif/esp-idf #6654: ULP RISC-V Example keeps rebooting

markus-seidl, 2021-03-03:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      none
- Module or chip used: ESP32S2-WROVER-I
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-4-g73db14240
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

I'm trying to run the included "ulp_riscv" example and have the following errors. The example isn't modified and executed vanilla. It should wake up the esp32s2 after a change on GPIO0. I have a pull up resistor on that port, if that matters.

1) Endless loop and unable to initialise the ULP. Sometimes a reset fixes this - sometimes not.
2) Endless loop after an event (change on GPIO0) has occurred.

Log for 

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. Download esp idf
2. Compile and flash the ulp_riscv example

### Code to reproduce this issue

See the example ulp_riscv

## Debug Logs

Endless loop (If I reset the board, sometimes the CPU crashes on startup)
```
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x12d4
load:0x4004c000,len:0x8c4
load:0x40050000,len:0x2ca4
[... 162 lines cut ...]
I (356) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
D (363) FLASH_HAL: extra_dummy: 0
D (366) spi_flash: trying chip: issi
D (369) spi_flash: trying chip: gd
D (373) spi_flash: trying chip: mxic
D (377) spi_flash: trying chip: winbond
D (380) spi_flash: trying chip: boya
D (384) spi_flash: trying chip: generic
I (388) spi_flash: detected chip: generic
I (393) spi_flash: flash io: dio
W (397) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
D (410) cpu_start: calling init function: 0x40080b58
0x40080b58: esp_ota_init_app_elf_sha256 at /home/parallels/esp/esp-idf/components/app_update/esp_app_desc.c:76

D (415) intr_alloc: Connected src 73 to int 10 (cpu 0)
D (420) intr_alloc: Connected src 28 to int 2 (cpu 0)
I (425) cpu_start: Starting scheduler on PRO CPU.
D (431) heap_init: New heap initialised at 0x3fffc000
D (431) intr_alloc: Connected src 17 to int 3 (cpu 0)
ULP-RISC-V woke up the main CPU! 
ULP-RISC-V read changes in GPIO_0 current is: Low 
Entering in deep sleep
```

## Other items if possible

renzbagaporo, 2021-03-29:
Hi @markus-seidl can you try this patch? 
[0001-ulp-clear-rtc-int-at-initialization.patch.txt](https://github.com/espressif/esp-idf/files/6220780/0001-ulp-clear-rtc-int-at-initialization.patch.txt)

markus-seidl, 2021-04-07:
Thanks! The example now works much more than I expect it to.

closed 2021-04-13 by markus-seidl