espressif/esp-idf #6659: ESP32-C3F  GPIO19 GPIO20 invalid

zenghi, 2021-03-04:
ESP-IDF v4.3-beta1-24-g0bbc721a6
gpio_pad_select_gpio(19);
gpio_set_direction(19, GPIO_MODE_OUTPUT);
gpio_set_level(19, 1);
gpio_pad_select_gpio(20);
gpio_set_direction(20, GPIO_MODE_OUTPUT);
gpio_set_level(20, 1);

Cannot output high level

chegewara, 2021-03-04:
Dont know about pin 19, but pin20 is UART0 pin:
```
U0RXD 27 I/O/T VDD3P3_CPU GPIO20, U0RXD
```
Page 11:
https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf

o-marshmallow, 2021-03-10:
Hello @zenghi ,
As @chegewara said, by default pin 20 is used for UART0. Two ways are possible to use it back as a GPIO:
* Using `gpio_pad_select_gpio`, as you did.
* Using `gpio_reset_pin` to restore back its GPIO default state.

I was unable to reproduce your issue despite using a ESP32-C3F and the same ESP-IDF version. The code you provide sets both output to high (verified with a logic analyzer). 

Two small things that may help you:
* Try using `gpio_reset_pin` instead of `gpio_pad_select_gpio`.
* Check the return values of the function calls you made. Maybe you are missing an error.

This is another example I used to test these two pins:
```
void test_gpio(void) {
    const int pins[] = { 19, 20 };
    const int length = sizeof(pins)/sizeof(int);
    esp_err_t err = ESP_OK;

    for (int i = 0; i < length; i++) {
        /* Resetting the pins will reconfigure them as GPIOs. */
        err = gpio_reset_pin(pins[i]);
        if (err != ESP_OK) {
            printf("Cannot reset GPIO %d (error %d)\n", pins[i], err);
            return;
        }

        err = gpio_set_direction(pins[i], GPIO_MODE_OUTPUT);
        if (err != ESP_OK) {
            printf("Cannot set GPIO %d direction (error %d)\n", pins[i], err);
            return;
        }

    }

    for (int out = 1; ; out = !out) {
        /* Set the same output level to all the pins we are testing. */
        for (int i = 0; i < length; i++) {
            err = gpio_set_level(pins[i], out);
            if (err != ESP_OK) {
                printf("Cannot set GPIO %d level (error %d)\n", pins[i], err);
                return;
            }
        }

        vTaskDelay(500 / portTICK_PERIOD_MS);
    }
}

void app_main(void)
{
    test_gpio();
    /* The function returned, an error occured. */
    printf("An error occured while testing GPIOs.\n");
    while(1) {
        vTaskDelay(10000 / portTICK_PERIOD_MS);
    }
}
```

If the problem persists, please contact the business support to get more details about the board you are using. This would help us reproducing the issue.

zenghi, 2021-03-11:
Hello @o-marshmallow  @chegewara   Thank you for your work. GPIO works normally according to the above code

closed 2021-03-11 by zenghi