espressif/esp-idf #6660: DevKit SPIRAM Init Bug in IDF 4.3, 4.4 and master (just work with 4.2)

drony, 2021-03-04:
```
I (28) boot: ESP-IDF v4.3-beta1-24-g0bbc721a6-dirty 2nd stage bootloader
I (29) boot: compile time 18:55:41
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (46) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00011000 00006000
I (78) boot:  1 phy_init         RF data          01 01 00017000 00001000
I (86) boot:  2 factory          factory app      00 00 00020000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=0b2e0h ( 45792) map
I (129) esp_image: segment 1: paddr=0002b308 vaddr=3ffb0000 size=02ad0h ( 10960) load
I (134) esp_image: segment 2: paddr=0002dde0 vaddr=40080000 size=02238h (  8760) load
I (139) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=1b66ch (112236) map
I (183) esp_image: segment 4: paddr=0004b694 vaddr=40082238 size=0d690h ( 54928) load
I (216) boot: Loaded app from partition at offset 0x20000
I (216) boot: Disabling RNG early entropy source...
D (228) efuse: coding scheme 0
C:/SysGCC/esp32/esp-idf/v4.3/components/freertos/queue.c:1441 (xQueueSemaphoreTake)- assert failed!
Re-enable cpu cache.

abort() was called at PC 0x4008a5a8 on core 0

Backtrace:0x400893b6:0x3ffe3970 0x40089b0d:0x3ffe3990 0x4008e7c6:0x3ffe39b0 0x4008a5a8:0x3ffe3a20 0x4008e6c9:0x3ffe3a60 0x400eb43b:0x3ffe3a80 0x4008e683:0x3ffe3ab0 0x400d720f:0x3ffe3b00 0x400d70fd:0x3ffe3b60 0x400d70b9:0x3ffe3b80 0x400837b8:0x3ffe3bb0 0x400d44a9:0x3ffe3be0 0x400813c0:0x3ffe3c00 0x40079795:0x3ffe3c30 |<-CORRUPTED

ELF file SHA256: 21b37dc60d2819b6

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7156
load:0x40078000,len:16848
ho 0 tail 12 room 4
load:0x40080400,len:4164
entry 0x400806dc
I (28) boot: ESP-IDF v4.3-beta1-24-g0bbc721a6-dirty 2nd stage bootloader
I (29) boot: compile time 18:55:41
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (46) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
```

```
Decoding stack results
0x400893b6: panic_abort at C:/SysGCC/esp32/esp-idf/v4.3/components/esp_system/panic.c line 356
0x40089b0d: esp_system_abort at C:/SysGCC/esp32/esp-idf/v4.3/components/esp_system/system_api.c line 112
0x4008e7c6: abort at C:/SysGCC/esp32/esp-idf/v4.3/components/newlib/abort.c line 46
0x4008a5a8: xQueueSemaphoreTake at C:/SysGCC/esp32/esp-idf/v4.3/components/freertos/queue.c line 1441
0x4008e6c9: esp_log_impl_lock_timeout at C:/SysGCC/esp32/esp-idf/v4.3/components/log/log_freertos.c line 45
0x400eb43b: esp_log_writev at C:/SysGCC/esp32/esp-idf/v4.3/components/log/log.c line 170
0x4008e683: esp_log_write at C:/SysGCC/esp32/esp-idf/v4.3/components/log/log.c line 199
0x400d720f: esp_efuse_utility_process at C:/SysGCC/esp32/esp-idf/v4.3/components/efuse/src/esp_efuse_utility.c line 71
0x400d70fd: esp_efuse_read_field_blob at C:/SysGCC/esp32/esp-idf/v4.3/components/efuse/src/esp_efuse_api.c line 47
0x400d70b9: esp_efuse_get_pkg_ver at C:/SysGCC/esp32/esp-idf/v4.3/components/efuse/src/esp32/esp_efuse_fields.c line 65
0x400837b8: psram_enable at C:/SysGCC/esp32/esp-idf/v4.3/components/esp32/spiram_psram.c line 808
0x400d44a9: esp_spiram_init at C:/SysGCC/esp32/esp-idf/v4.3/components/esp32/spiram.c line 150
0x400813c0: call_start_cpu0 at C:/SysGCC/esp32/esp-idf/v4.3/components/esp_system/port/cpu_start.c line 347

```

```
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

void app_main(void)
{
	while (true)
	{

		vTaskDelay(1000 / portTICK_PERIOD_MS);	
		printf("Hello world!\n");
	
	}
}

```
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/6084953/sdkconfig.zip)

mahavirj, 2021-03-05:
@drony This was fixed with https://github.com/espressif/esp-idf/commit/2cdc1fcdbf881fe9cecfb834289de6c496f51975#diff-ed775424e11115a26dae63b7478eeb08177aaae845aed6d22d98a09f22457f82, I see it as part of v4.3-beta1 release as well. Can you please confirm if you have this change once?

drony, 2021-03-05:
I checked the idf version installed on me. I have the same piece of code. The same problem with master branch and v4.4dev release.

please create simple app and replace sdkconfig file with the one i attached above post. 

thank you

drony, 2021-03-05:
**Fixed:**

look like missed replace another ESP_LOGD with ESP_EARLY_LOGD at line 71
`esp_err_t esp_efuse_utility_process(const esp_efuse_desc_t* field[], void* ptr, size_t ptr_size_bits, efuse_func_proc_t func_proc)
{
    esp_err_t err = ESP_OK;
    int bits_counter = 0;

    // get and check size.
    int field_len = esp_efuse_get_field_size(field);
    int req_size = (ptr_size_bits == 0) ? field_len : MIN(ptr_size_bits, field_len);

    int i = 0;
    while (err == ESP_OK && req_size > bits_counter && field[i] != NULL) {
        if (check_range_of_bits(field[i]->efuse_block, field[i]->bit_start, field[i]->bit_count) == false) {
            ESP_EARLY_LOGE(TAG, "Range of data does not match the coding scheme");
            err = ESP_ERR_CODING;
        }
        int i_reg = 0;
        int num_reg;
        while (err == ESP_OK && req_size > bits_counter &&
                (num_reg = get_reg_num(field[i]->bit_start, field[i]->bit_count, i_reg)) != -1) {

            int start_bit = get_starting_bit_num_in_reg(field[i]->bit_start, i_reg);
            int num_bits = get_count_bits_in_reg(field[i]->bit_start, field[i]->bit_count, i_reg);
            if ((bits_counter + num_bits) > req_size) { // Limits the length of the field.
                num_bits = req_size - bits_counter;
            }
            **ESP_EARLY_LOGD(TAG, "In EFUSE_BLK%d__DATA%d_REG is used %d bits starting with %d bit",
                    (int)field[i]->efuse_block, num_reg, num_bits, start_bit);**
            err = func_proc(num_reg, field[i]->efuse_block, start_bit, num_bits, ptr, &bits_counter);
            ++i_reg;
        }
        i++;
    }
    assert(bits_counter <= req_size);
    return err;
}`

mahavirj, 2021-03-05:
@drony Looks like this was reverted in https://github.com/espressif/esp-idf/commit/f5f046126468ff27c8ec2c4e2e06bdf74c5b1e5c#diff-ed775424e11115a26dae63b7478eeb08177aaae845aed6d22d98a09f22457f82 causing this regression. We will fix it.

mahavirj, 2021-03-09:
@drony Issue will be (auto) closed when fix reaches to github

closed 2021-04-01 by espressif-bot