espressif/esp-idf #6661: esp_http_client_perform fails to return response body data in 4.3-beta1

flyinggorilla, 2021-03-04:
## Environment

- Development Kit:      [ESP32-Wrover-Kit] --> ESP32_core_board_v2 as well as on Adafruit Huzzah32
- Module or chip used:  [ESP32-WROOM-32]
- IDF version v4.3-beta1
- Build System:  idf.py driving CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows
- (Windows only) environment type:  Plain Command Prompt
- Using an IDE: Visual Code 
- Power Supply: USB

## Problem Description

* **ESP-IDF 4.2 stable has a known bug with scrambling buffers of HTTP response body when using `esp_http_client_perform()`**. I ran into this issue with HTTP response bodies of approx >500 bytes. My examples worked fine with response bodies that where approx <100 bytes. 
* **ESP-IDF 4.3-beta1** release notes mentions this v4.2 bug as fixed. However, **now there is NO HTTP response data by `esp_http_client_read_response()` at all now** irrespective whether using the user_data `user_data` and/or `esp_http_client_read().

* what DOES work in v4.3-beta1 however is to use the combination of `esp_http_client_open()`, `esp_http_client_write()`, `esp_http_client_read_response()`  and **NOT USE use `esp_http_client_perform()` at all**

### Expected Behavior

* either `user_data` buffer be filled with response data (however, i didnt find a way to limit writing too much into this buffer)
* or calling `esp_http_client_read_response()` after `esp_http_client_perform()` fills the buffer with the data in same amount as HTTP Content-Length header.

### Code to reproduce this issue

```

//###### THIS FUNCTION FAILS TO RETREIVE RESPONSE BODY DATA ##############
void testHttpClient_CallPerformUserData(const char* tag)
{
    char sBuf[2048] = {0};
    esp_http_client_config_t espHttpClientConfig = {0};
    espHttpClientConfig.url = "https://ptsv2.com/t/espidf43beta/post";
    espHttpClientConfig.method = HTTP_METHOD_POST; // HTTP_METHOD_GET LEADS TO SAME PROBLEM
    espHttpClientConfig.user_data = sBuf;  // PROVIDING A BUFFER DOESNT HELP EITHER
    esp_http_client_handle_t hEspHttpClient = NULL;
    if (!hEspHttpClient)
    {
        hEspHttpClient = esp_http_client_init(&espHttpClientConfig);
    }
    const char *sPostData = "request: testdata\r\nsystem: testWeatherbuoy,0,765\r\n";
    esp_err_t err;
    esp_http_client_set_header(hEspHttpClient, "Content-Type", "text/plain");
    err = esp_http_client_set_post_field(hEspHttpClient, sPostData, strlen(sPostData));
    if (err != ESP_OK)
    {
        ESP_LOGE(tag, "esp_http_client_set_post_field() %d bytes %s", strlen(sPostData), esp_err_to_name(err));
    } 
    err = esp_http_client_perform(hEspHttpClient);
    if (err == ESP_OK)
    {
        int iContentLength = esp_http_client_get_content_length(hEspHttpClient);
        ESP_LOGI(tag, "esp_http_client_fetch_headers(): Status %d, Content-Length %d", esp_http_client_get_status_code(hEspHttpClient), iContentLength);
        if (iContentLength >= 0) {
            if (iContentLength == strlen(sBuf)) {
                ESP_LOGI(tag, "esp_http_client_read_response(): %d>>>\r\n%s<<<", iContentLength, sBuf);
            } else {
                ESP_LOGE(tag, "esp_http_client_read_response(): user_data does not contain expected %d bytes>>>\r\n%s<<<", iContentLength, sBuf);
            }
        } else {
            ESP_LOGE(tag, "esp_http_client_read_response(): %s", esp_err_to_name(err));
        }
    }
    else
    {
        ESP_LOGE(tag, "esp_http_client_open() %s", esp_err_to_name(err));
    }

    esp_http_client_cleanup(hEspHttpClient);
    hEspHttpClient = NULL;
}

//###### THIS FUNCTION FAILS TO RETREIVE RESPONSE BODY DATA ##############
void testHttpClient_CallPerformRead(const char* tag)
{
    char sBuf[2048] = {0};
    esp_http_client_config_t espHttpClientConfig = {0};
    espHttpClientConfig.url = "https://ptsv2.com/t/espidf43beta/post";
    espHttpClientConfig.method = HTTP_METHOD_POST; // HTTP_METHOD_GET LEADS TO SAME PROBLEM
    //espHttpClientConfig.method = HTTP_METHOD_POST; // HTTP_METHOD_GET LEADS TO SAME PROBLEM
    espHttpClientConfig.user_data = sBuf;  // PROVIDING A BUFFER DOESNT HELP EITHER
    //espHttpClientConfig.user_data = NULL;  // 
    esp_http_client_handle_t hEspHttpClient = NULL;
    if (!hEspHttpClient)
    {
        hEspHttpClient = esp_http_client_init(&espHttpClientConfig);
    }
    const char *sPostData = "request: testdata\r\nsystem: testWeatherbuoy,0,765\r\n";
    esp_err_t err;
    esp_http_client_set_header(hEspHttpClient, "Content-Type", "text/plain");
    err = esp_http_client_set_post_field(hEspHttpClient, sPostData, strlen(sPostData));
    if (err != ESP_OK)
    {
        ESP_LOGE(tag, "esp_http_client_set_post_field() %d bytes %s", strlen(sPostData), esp_err_to_name(err));
    } 
    err = esp_http_client_perform(hEspHttpClient);
    if (err == ESP_OK)
    {
        int iContentLength = esp_http_client_get_content_length(hEspHttpClient);
        ESP_LOGI(tag, "esp_http_client_fetch_headers(): Status %d, Content-Length %d", esp_http_client_get_status_code(hEspHttpClient), iContentLength);
        int len = esp_http_client_read_response(hEspHttpClient, sBuf, sizeof(sBuf) - 1);
        if (len >= 0) {
            if (iContentLength == len) {
                ESP_LOGI(tag, "esp_http_client_read_response(): %d>>>\r\n%s<<<", iContentLength, sBuf);
            } else {
                ESP_LOGE(tag, "esp_http_client_read_response(): content missing %d != %d bytes>>>\r\n%s<<<", len, iContentLength, sBuf);
            }
        } else {
            ESP_LOGE(tag, "esp_http_client_read_response(): %s", esp_err_to_name(err));
        }
    }
    else
    {
        ESP_LOGE(tag, "esp_http_client_open() %s", esp_err_to_name(err));
    }

    esp_http_client_cleanup(hEspHttpClient);
    hEspHttpClient = NULL;
}

//###### THIS FUNCTION ***SUCCEEDS** TO PROVIDE RESPONSE BODY DATA ##############
void testHttpClient_CallOpenWriteRead(const char* tag)
{
    char sBuf[2048] = {0};
    esp_err_t err;
    esp_http_client_config_t espHttpClientConfig = {0};
    espHttpClientConfig.url = "https://ptsv2.com/t/espidf43beta/post"; //
    espHttpClientConfig.method = HTTP_METHOD_POST;
    esp_http_client_handle_t hEspHttpClient = NULL;
    if (!hEspHttpClient)
    {
        hEspHttpClient = esp_http_client_init(&espHttpClientConfig);
    }

    const char *sPostData = "request: testdata\r\nsystem: testWeatherbuoy,0,765\r\n";
    esp_http_client_set_header(hEspHttpClient, "Content-Type", "text/plain");
    err = esp_http_client_open(hEspHttpClient, strlen(sPostData));
    if (err == ESP_OK)
    {
        int sent = esp_http_client_write(hEspHttpClient, sPostData, strlen(sPostData));
        if (sent >= 0) {
            ESP_LOGI(tag, "esp_http_client_write(): OK, sent: %d", sent);
        } else {
            ESP_LOGE(tag, "esp_http_client_write()");
        }

        int iContentLength = esp_http_client_fetch_headers(hEspHttpClient);
        ESP_LOGI(tag, "esp_http_client_fetch_headers(): Status %d, Content-Length %d", esp_http_client_get_status_code(hEspHttpClient), iContentLength);
        //ESP_LOGI(tag, ">>>(%d)\r\n%s\r\n<<<", strlen(sBuf), sBuf);

        int len = esp_http_client_read_response(hEspHttpClient, sBuf, sizeof(sBuf) - 1);
        if (len >= 0) {
            ESP_LOGI(tag, "esp_http_client_read_response(): %d>>>\r\n%s<<<", len, sBuf);
        } else {
            ESP_LOGE(tag, "esp_http_client_read_response(): %s", esp_err_to_name(err));
        }
    }
    else
    {
        ESP_LOGE(tag, "esp_http_client_open() %s", esp_err_to_name(err));
    }

    esp_http_client_cleanup(hEspHttpClient);
    hEspHttpClient = NULL;
}
```

## Debug Logs

```
 testHttpClient_CallPerformUserData("PerformUserData");
 testHttpClient_CallPerformRead("PerformRead");
 testHttpClient_CallOpenWriteRead("OpenWriteRead");
```

I (4914) PerformUserData: esp_http_client_fetch_headers(): Status 200, Content-Length 137
**E (4914) PerformUserData: esp_http_client_read_response(): user_data does not contain expected 137 bytes>>>**
<<<
I (6754) PerformRead: esp_http_client_fetch_headers(): Status 200, Content-Length 137
**E (6754) PerformRead: esp_http_client_read_response(): content missing 0 != 137 bytes>>>**
<<<
I (8214) OpenWriteRead: esp_http_client_write(): OK, sent: 50
I (8494) OpenWriteRead: esp_http_client_fetch_headers(): Status 200, Content-Length 137
I (8494) OpenWriteRead: esp_http_client_read_response(): 137>>>
<BEGIN>One morning, when Gregor Samsa woke from troubled dreams, he found himself transformed in his bed into a horrible vermin.<THE END><<<

## Other items if possible

sdkconfig changes from defaults:
```
#
# Flash
#
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"

#
# Partition Table
#
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions.csv"

# ESP HTTP client
#
CONFIG_ESP_HTTP_CLIENT_ENABLE_HTTPS=y

#
# Bootloader config
#
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
```

yuanjianmin, 2021-03-05:
@flyinggorilla Function `esp_http_client_perform` include `esp_http_client_open` -> `esp_http_client_write` -> `esp_http_client_fetch_headers` -> `esp_http_client_read` (and option) `esp_http_client_close`. So if you want to get data in `user_data`, you should register a `event_handler` in `esp_http_client_config_t`, and in this `event_handler` callback, you should combine the response data in `HTTP_EVENT_ON_DATA` to `evt->user_data`. And after `esp_http_client_perform` returned, the response data is in your `user_data` pointer. The more usage you can refer to example `esp_http_client_example.c`.

flyinggorilla, 2021-03-05:
Thank you for your response. 
* my preferred use is the `esp_http_client_perform()` variant, that I had assumed I can combine with `esp_http_client_read_response()` and intentionally not need the event handler (as it partially/erroneously worked in ESP-IDF v4.2).
  * if this use is NOT intended, i suggest the best action item is to __mention in the documentation of `esp_http_client_read_response()` that it should only be used in context of stream mode__ with `esp_http_client_open()`->`esp_http_client_write()`->`esp_http_client_read_response()`. 
  * And __updating the documentation of `esp_http_client_perform()` that it is mandatory to use an event handler for receiving HTTP body data__ (i.e. using `esp_http_client_read_*()` api is not intended/wanted). 
* i see now my mistake on the variant with `user_data`. thanks for pointing that out. i am really sorry, totally my bad. no excuses. 

thank you.

yuanjianmin, 2021-03-15:
@flyinggorilla Thank you for your humanized advice.We will update the doc in future version.If you have no other problem, could you please help close this issue. Thanks.

flyinggorilla, 2021-03-15:
thanks, closing as future doc update should prevent this issue.

closed 2021-03-15 by flyinggorilla