espressif/esp-idf #6662: Error: invalid register number (97) for 'wsr' instruction

roger-, 2021-03-04:
## Environment

- Development Kit:     ESP32-S2-Saola-1RI
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-S2-WROVER-I
- IDF version: v4.4-dev-4-g73db14240
- Build System:   idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:   Linux
- Using an IDE?: No
- Power Supply:    USB

## Problem Description

Running `make` for my project gives me:

```
CC build/bootloader/bootloader_support/src/bootloader_mem.o
/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h: Assembler messages:
/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:65: Error: invalid register number (97) for 'wsr' instruction
make[2]: *** [/esp/esp-idf/make/component_wrapper.mk:292: src/bootloader_mem.o] Error 1
```

Prior to this I ran ` idf.py set-target esp32s2`.

Building on a regular ESP32 (non S2) works fine.

igrr, 2021-03-05:
@roger- GNU Make based build system only supports the esp32 target. To build for esp32s2, you need to use CMake based build system `idf.py build`. Please follow the getting started guide starting from here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/index.html#step-7-configure.

roger-, 2021-03-05:
You're right, missed that. Thanks!

closed 2021-03-05 by roger-