espressif/esp-idf #6663: http response: send partial data w/o chunk encoding

lal12, 2021-03-05:
**Is your feature request related to a problem? Please describe.**

The only way of sending unbuffered data as a http response is currently by using `httpd_resp_send_chunk`. However a chunk encoding is not always wished.

**Describe the solution you'd like**

It is possible to just send a response without `content-length` header. So something like 
```
httpd_resp_send_status(req);
httpd_resp_send_headers(req);
httpd_resp_send_data(req, mybuf, buflen);
```
should be added.

lal12, 2021-03-05:
I noticed that by default `Connection: keep-alive` is used so this obviously would needed to be changed for a case like described above, to indicate the end of the transfer.

shubhamkulkarni97, 2021-03-18:
Hi @lal12,

Thanks for raising this feature request.

Could you share use case in which you need this type of behaviour?

lal12, 2021-03-18:
For one I had some issues in the past (not specifically with the ESP IDF) where chunk encoding wasn't support by simple http libraries. (Yeah I know chunk encoding is required in the spec, but I don't think it is that commonly used).

But for the most part just for efficiency purposes and ease of uses. I imagine that the LWIP TCP_SND_BUF would be much better utilized, currently when constructing JSON many very short parts (like brackets or single words) are transmitted each per chunk, which adds a lot of unnecessary overhead.

shubhamkulkarni97, 2021-11-11:
@lal12 Do you have any patch to support this change?

This feature request seems to be use case specific and it would be great if you can contribute with this feature.

closed 2021-12-10 by Alvin1Zhang