espressif/esp-idf #6664: Store multiple SSID/password combinations in NVS

franz-ms-muc, 2021-03-05:
Store multiple SSID/password combinations in NVS

today it handle 1 SSID in STA Mode, and cannot be used as a mobile device travelling from WIFI-to-WIFI and automatically connect to one of the known Wifis. 

This Feature Request is a pro for all mobile and Vehicle applications.

Alvin1Zhang, 2021-03-08:
Thanks for raising this feature request.

MaxwellAlan, 2021-04-14:
Hi @franz-ms-muc thanks for request this feature.
It's a big breaking changes for WiFi libs, we need more time to evaluate this feature, thanks.
In additon, this features can be realized in application according to existing IDF apis, such as you can save some ssid and password into NVS, and before connect, scan for the best rssi ssid and then using NVS saved password to connect, all can be done in applications.

willmoffat, 2021-04-14:
@MaxwellAlan it's true that it can be implemented in the app, but that means it gets reimplemented (badly) many times over. I'd also love to see IDF support for multiple WiFi credentials (and reconnect).

MaxwellAlan, 2021-11-17:
Hi @willmoffat @franz-ms-muc , this feature will break the NVS design for our other customer, we will not support this feature. Thanks for your request.

Alvin1Zhang, 2021-11-17:
Thanks for reporting.

closed 2021-11-17 by Alvin1Zhang