espressif/esp-idf #6668: Unclear comment in blufi example

KonstantinSchubert, 2021-03-06:
## Environment

- Development Kit: All
- IDF version : HEAD
- Build System:   All
## Problem Description

There is a comment in the `blufi_wireless.c` file in the `esp32-blufi` example which is unclear. 

https://github.com/wolfeidau/esp32-blufi/blob/6c822a6792f3997838f1e22259f53d110b11e71e/blufi_wireless.c#L249

It refers to a `bluefi_task` entity, but it is not explained what this is or why it should be used.

### Expected Behavior

The comment should explain what `blufi_task` is referring to, and why it should be used.
If the comment is incorrect, it should be removed.

Alvin1Zhang, 2021-03-08:
Thanks for reporting and sorry for the inconvenience, we will look into.

xulongzhuang, 2021-04-07:
Hello, @KonstantinSchubert , I think that when there is a blocking behavior or a long-term execution of the action, a new thread `blufi_task` should be created to execute it. Calling the above behavior in the callback may cause the watchdog timeout interruption error，`blufi_task` is created by the user according to the actual situation

closed 2022-12-01 by Alvin1Zhang