espressif/esp-idf #6669: CAN bus driver error, Arduino core 1.0.4 -> 1.0.5

obdevel, 2021-03-06:
- Development Kit: ESP32-DevKitC
- Kit version: see below
- Module or chip used: ESP32-WROOM-32
- IDF version: see below
- Build System: 
- Compiler version:
- Operating System: macOS 10.13.6
- Using an IDE?: Arduino 1.8.13
- Power Supply: USB

Problem Description

Upgrading from Arduino core 1.0.4 to 1.0.5 causes a CAN driver error with previously-working code.

Specifically, the can_driver_install() function returns ESP_ERR_INVALID_STATE, which implies the driver is already installed. The code compiles correctly but fails with this error at runtime. Reverting to core 1.0.4 is a workaround but not a long-term solution.

Arduino core => IDF versions:
1.0.4 => v3.2.3-14-gd3e562907
1.0.5 => v3.3.4-432-g7a85334d8

Has this error been fixed in a later version ? I want to guide the Arduino project to the minimum IDF version they need to resolve this.

Thanks.

---------------
esptool.py v2.8-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:71:bf:f1:1e:f0
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 3c:71:bf:f1:1e:f0

Dazza0, 2021-09-07:
@obdevel  Could to check what error code is being returned form `can_driver_install()`?

obdevel, 2021-09-07:
@Dazza0 It seems to be compiling and working ok in 1.0.6. I'll do some more testing to prove this.

I can't test 2.0.0 because esptool.py fails on macOS 10.13. I've logged a separate issue for this.

Dazza0, 2021-09-23:
@obdevel Closing this issue as it seems to be working. Please feel free to reopen if more issues occur

closed 2021-09-23 by Dazza0