espressif/esp-idf #6671: Install.sh fail on path that include spaces (ubuntu 20.04)

kolergy, 2021-03-06:
esp-idf V4.4-dev (obtained via git clone on the 06/03/2021) install.sh script fails due to a name including spaces in the path

`(base) xx@linux:~/My Drive/Projects/esp-idf$ ./install.sh
 Installing ESP-IDF tools
./install.sh: line 9: /home/xx/My: No such file or directory`

The fix is to put the full install path between comas as follows in install.sh:

`#!/usr/bin/env bash

set -e
set -u

export IDF_PATH=$(cd $(dirname $0); pwd)

echo "Installing ESP-IDF tools"
"${IDF_PATH}/tools/idf_tools.py" install

echo "Installing Python environment and packages"
"${IDF_PATH}/tools/idf_tools.py" install-python-env

basedir="$(dirname $0)"
echo "All done! You can now run:"
echo ""
echo "  . ${basedir}/export.sh"
echo ""`

igrr, 2021-03-09:
@kolergy ESP-IDF build system currently doesn't support spaces in paths, please see the note on the getting started page: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html?highlight=spaces#step-1-install-prerequisites.

dobairoland, 2023-08-04:
We have retested this issue with latest version of the v4.4 branch and it looks like the problem has been solved over time and we have forgot to close this.

closed 2023-08-04 by dobairoland