espressif/esp-idf #6673: esp_wifi_set_config returns incorrect error code for invalid password

boarchuz, 2021-03-07:
## Environment

- IDF version: v4.3-beta1-24-g0bbc721a6

## Problem Description

When setting WiFi AP config with an invalid password, esp_wifi_set_config() returns a meaningless esp_err_t.

### Expected Behavior

Returns ESP_ERR_WIFI_PASSWORD (ESP_ERR_WIFI_BASE + 0xb).

### Actual Behavior

Returns 0xb

### Code to reproduce this issue

```
#include "esp_event.h"
#include "nvs_flash.h"
#include "esp_wifi.h"

void app_main(void)
{
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    wifi_config_t ap_config = {
        .ap = {
            .ssid = "MYSSID",
            .password = "", // <-- invalid password for WIFI_AUTH_WPA2_PSK
            .authmode = WIFI_AUTH_WPA2_PSK,
            .max_connection = 5,
        },
    };

    ESP_ERROR_CHECK(esp_event_loop_create_default());
    ESP_ERROR_CHECK(nvs_flash_init());

    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &ap_config));
}
```

## Debug Logs

```
W (685) wifi:password less than 8

ESP_ERROR_CHECK failed: esp_err_t 0xb (ERROR) at 0x40086fb4
0x40086fb4: _esp_error_check_failed at /home/matt/espressif/esp-idf/components/esp_common/src/esp_err.c:41

file: "../main/main.c" line 22
func: app_main
expression: esp_wifi_set_config(WIFI_IF_AP, &ap_config)
```

espxiehang, 2021-03-08:
Hi @boarchuz ,The error reported by `esp_wifi_set_config` is `W (685) wifi:password less than 8`.
At this time, the return value of `esp_wifi_set_config`  is not `ESP_OK`.
Because you use `ESP_ERROR_CHECK`, it will trigger `abort ()`.
```
#define ESP_ERROR_CHECK(x) do {                                         \
        esp_err_t __err_rc = (x);                                       \
        if (__err_rc != ESP_OK) {                                       \
            _esp_error_check_failed(__err_rc, __FILE__, __LINE__,       \
                                    __ASSERT_FUNC, #x);                 \
        }                                                               \
    } while(0)
void _esp_error_check_failed(esp_err_t rc, const char *file, int line, const char *function, const char *expression)
{
    printf("ESP_ERROR_CHECK failed: esp_err_t 0x%x", rc);
#ifdef CONFIG_ESP_ERR_TO_NAME_LOOKUP
    printf(" (%s)", esp_err_to_name(rc));
#endif //CONFIG_ESP_ERR_TO_NAME_LOOKUP
    printf(" at %p\n", __builtin_return_address(0));
    printf("file: \"%s\" line %d\nfunc: %s\nexpression: %s\n", file, line, function, expression);
    abort();
}

```

espxiehang, 2021-03-08:
I'll check the reason why the return value is not correct

boarchuz, 2021-03-08:
@espxiehang Thanks, I've noticed another one too so there may be even more that need ESP_ERR_WIFI_BASE added:

Change the above `esp_wifi_set_mode(WIFI_MODE_AP)` -> `esp_wifi_set_mode(WIFI_MODE_STA)` and then the final line will return 5 instead of the expected ESP_ERR_WIFI_MODE (ESP_ERR_WIFI_BASE + 5)

espxiehang, 2021-03-08:
Hi @boarchuz ,Thank you for your feedback.
Use `uint8_ t` receiving the return value causes this issue.
I'll fix it as soon as possible.

closed 2021-03-12 by espressif-bot