espressif/esp-idf #6677: ESP32-WROVER boot strapping pin

tatsutaigu, 2021-03-08:
About data sheet of ESP32-WROVER,
[https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf](https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf)

page 10: VDD_SDIO default is 3.3V(pull-down)
page 19: on schematic, VDD_SDIO is 1.8V as description.

if page 19 is correct, can I set pull-up on IO12pin on PCBA?
[https://dl.espressif.com/dl/schematics/esp32-lyrat-v4.3-schematic.pdf](https://dl.espressif.com/dl/schematics/esp32-lyrat-v4.3-schematic.pdf)
IO12 pull-up resister(R34) is removed on LyraT board.
But, can I mount R34?

negativekelvin, 2021-03-08:
VDD_SDIO is 1.8V for ESP32-WROVER and it has internal pullup
VDD_SDIO is 3.3V for ESP32-WROVER-B and ESP32-WROVER-E

Alvin1Zhang, 2021-03-08:
@tatsutaigu Checked with our internal team, VDD_SDIO is 1.8v on ESP32-WROVER , it has 10K ohm pull-up resistor at IO12 on module.

tatsutaigu, 2021-03-08:
@Alvin1Zhang 
Thank you for your reply.

I found R9 on schematics of WROVER.
[https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf](https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf)

closed 2021-03-08 by tatsutaigu