espressif/esp-idf #6678: idf monitor problem with esp32

cxjcxggg, 2021-03-08:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROVER-E]
- IDF version : v4.2
- Build System:  [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

idf monitor doesn't work well. I was running hello-world example. After I ran ``idf.py set-target esp32`` and ``idf.py -p COM9 flash monitor``, building and flashing succeeded but there seemed to be a problem with the monitor. It was supposed to print like the Expected Behavior but only print like the Actual Behavior. 

I ran ``idf.py fullclean``, but it doesn't help.

### Expected Behavior

[image]

This picture was taken one month ago. Today I manually deleted and reinstalled the ESP-IDF and ESP-IDF-TOOLS to the v4.2, and the problem arose. Everything is the same except the ESP-IDF. As you can see in the log, the ESP-IDF I used was v4.1-dirty.

### Actual Behavior

[image]

### Steps to reproduce

1. connect the board
2. move to the hello-world directory
3. run ``idf.py set-target esp32``
4. run ``idf.py -p COM9 flash monitor``

cxjcxggg, 2021-03-09:
I run XCOM to monitor the serial but there is no output either. So it doesn't seem to be a monitor problem.

waterfox207, 2021-03-09:
Hey,
I have a similar problem with the actual master (73db142). I am not able to monitor or flash my esp. I have tested it on two computers with the same problem. 
For flashing I am using the ESP-Prog. With IDF-V4.2 flashing and monitor the device works fine. Sadly my code is somehow not backwards compatible with IDF-V4.2.

beta-tester, 2021-03-09:
i had the same issue (73db142403c6e5b763a0e1c07312200e9b622673) on linux (Raspberry Pi OS / Debian buster).
it had no problems before.

it was using two different serial ports...
/dev/ttyUSB0
/dev/ttyAMA0 (permission denied)
one was working the second not, but it was using the two serial ports randomly(?) in the same run...
before compiling, to flash and to start monitor... there was always the chance that the process took the wrong port and failt.

thanks to @waterfox207 i added `-p /dev/ttyUSB0` to the parameters and now it is working again.

cxjcxggg, 2021-03-10:
> Hey,
> I have a similar problem with the actual master ([73db142](https://github.com/espressif/esp-idf/commit/73db142403c6e5b763a0e1c07312200e9b622673)). I am not able to monitor or flash my esp. I have tested it on two computers with the same problem.
> For flashing I am using the ESP-Prog. With IDF-V4.2 flashing and monitor the device works fine. Sadly, my code is somehow not backwards compatible with IDF-V4.2.
> 
> EDIT: I found a "workaround" if I add -p com4 (the serial port of my programmer) flashing and monitoring works like expected.

cxjcxggg, 2021-03-12:
Also, the serial ports detected on my win10 were COM3 and COM4 one month ago when the example was working, but now they are COM8 and COM9. I have no idea why they changed and whether it has anything to do with the problem.

martinespgano, 2021-03-16:
Hi, @cxjcxggg, after trying the steps provided by ESP_Minatel (https://esp32.com/viewtopic.php?f=13&t=19878), I'd suggest building, flashing and monitoring another simple example, e.g. blink in the `get-started` folder. Please, let me know if the problem persists.

cxjcxggg, 2021-03-18:
Hi @martinespgano , I ran the blink example. Flashing seemed to be successful, but no LED on the board blinked. The monitor acted the same as before. 

I manually put the board on DFU mode by holding the RESET button and pressing the EN button, the monitor printed 

```
rst:0x1 (POWERON_RESET),boot:0x6 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
waiting for download
```
It seems that the connection between serial port and monitor is good. So I start to think maybe it's a hardware damage problem rather than a monitor failure.

martinespgano, 2021-03-19:
Hi, @cxjcxggg, thank you for the information. Do you have a chance to try another device? If you haven't changed the `BLINK_GPIO` number in the blink example, the GPIO 5 is used by default. Is some LED connected to GPIO 5?

beta-tester, 2021-03-19:
for example, my ESP32 board has its built-in LED on GPIO 2.
it is this ESP32 board https://joy-it.net/en/products/SBC-NodeMCU-ESP32

cxjcxggg, 2021-03-20:
Hi @beta-tester and @martinespgano, my board is a [ESP-WROVER-KIT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html). The LEDs connect to GPIO 0, 2 and 4. I did forget to check the blink example configuration before, so it was set to GPIO 5 by default. But even if I changed it to GPIO 0, 2 and 4, no LEDs blinked after flashing. 

I have some more information here that might be helpful:
- Three LEDs just remain lighting up all the time when I build ,flash and monitor hello-world and blink example. 
- The green LED goes off when I press RESET button and lights up when I release it. The blue and red LEDs are on all the time.
- The LCD screen lights up when I press EN button and goes off when I release it.

cxjcxggg, 2021-03-20:
Hi @martinespgano, I have another ESP32. I will give it a try as soon as my colleague finishes his work with it. By the way, I have an ESP8266 available, does the ESP-IDF I have work with it?

martinespgano, 2021-03-23:
Hi, @cxjcxggg, thank you for the provided information. ESP-IDF is for ESP32, thus it'd be better to wait for another ESP32. Please, let us know as soon as you'll have a chance to try it.

martinespgano, 2021-03-23:
Hi, @cxjcxggg, thank you for the provided information. ESP-IDF is for ESP32, thus it'd be better to wait for another ESP32. Please, let us know when you'll have a chance to try it.

cxjcxggg, 2021-03-26:
Hi @martinespgano , thanks for your help. I will let you know the result as soon as I get hand on another ESP32.

cxjcxggg, 2021-03-29:
Hi @martinespgano , I have tried my ESP32-DevKitC V4. The hello_world example works fine and the monitor successfully reads the output.

[image]

So it seems that the esp-idf is good. There may be a hardware problem with my ESP32-Wrover-Kit, or the latest version of esp-idf doesn't work with the board.

martinespgano, 2021-04-06:
Hi, @cxjcxggg, thank you very much for your feedback. In that case, I suggest closing this issue.

cxjcxggg, 2021-04-07:
Hi @martinespgano, thank you very much for your help.

tom-borcin, 2021-04-09:
Hi @cxjcxggg, thank you for reporting this. I am closing this issue now to keep our backlog small and manageable. If you don't agree with this action, please let me know. Thank you!

closed 2021-04-09 by tom-borcin

Zeni241, 2022-10-24:
0

**Press the reset(enable)** button after you see:

`MONITOR --- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 --- --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---`

kerms, 2023-03-09:
Using wemos esp32-s2 with USB-CDC interface, I have the same behavior with `idf.py monitor -p /dev/ttyACM0`. While esp32-c3 USB-CDC works just fine as esp32 with USB-UART chip(ch340/cp2102).

Pressing RESET works for esp32-s2.
Also works when using simple `idf.py monitor` (without -p port), but this command will try to connect other port. when it comes to the right port, UART output shown correctly. (Pressing RST is a way quicker though :/