espressif/esp-idf #6679: ESPTouch does not work for the second time!

mohammadjavadpirhadi, 2021-03-08:
## Environment

- Development Kit: ESP32-DevKit
- Kit version : [v1]
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.3-beta1
- Build System: idf.py
- Compiler version: 8.4.0
- Operating System: Windows
- (Windows only) environment type: VS Code Terminal
- Using an IDE?: Yes (VS Code)
- Power Supply: USB

## Problem Description

I want to send the node a new SSID and password using ESPTouch again after WiFi disconnection. But after disconnection when I enable it again and send it data using the android app, the node does not get the password!

### Steps to reproduce

1. Connect node to AP using ESPTouch.
2. Turn off AP.
3. Try to send the node a new SSID and password for another AP using ESPTouch.

### Code to reproduce this issue

```cpp/* Esptouch example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_wpa2.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_netif.h"
#include "esp_smartconfig.h"

/* FreeRTOS event group to signal when we are connected & ready to make a request */
static EventGroupHandle_t s_wifi_event_group;

/* The event group allows multiple bits for each event,
   but we only care about one event - are we connected
   to the AP with an IP? */
static const int CONNECTED_BIT = BIT0;
static const int ESPTOUCH_DONE_BIT = BIT1;
static const char *TAG = "smartconfig_example";

static void smartconfig_example_task(void * parm);

static void event_handler(void* arg, esp_event_base_t event_base,
                                int32_t event_id, void* event_data)
{
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
        xTaskCreate(smartconfig_example_task, "smartconfig_example_task", 4096, NULL, 3, NULL);
    } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
        xTaskCreate(smartconfig_example_task, "smartconfig_example_task", 4096, NULL, 3, NULL);
        xEventGroupClearBits(s_wifi_event_group, CONNECTED_BIT);
    } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
        xEventGroupSetBits(s_wifi_event_group, CONNECTED_BIT);
    } else if (event_base == SC_EVENT && event_id == SC_EVENT_SCAN_DONE) {
        ESP_LOGI(TAG, "Scan done");
    } else if (event_base == SC_EVENT && event_id == SC_EVENT_FOUND_CHANNEL) {
        ESP_LOGI(TAG, "Found channel");
    } else if (event_base == SC_EVENT && event_id == SC_EVENT_GOT_SSID_PSWD) {
        ESP_LOGI(TAG, "Got SSID and password");

        smartconfig_event_got_ssid_pswd_t *evt = (smartconfig_event_got_ssid_pswd_t *)event_data;
        wifi_config_t wifi_config;

        bzero(&wifi_config, sizeof(wifi_config_t));
        memcpy(wifi_config.sta.ssid, evt->ssid, sizeof(wifi_config.sta.ssid));
        memcpy(wifi_config.sta.password, evt->password, sizeof(wifi_config.sta.password));
        wifi_config.sta.bssid_set = evt->bssid_set;
        if (wifi_config.sta.bssid_set == true) {
            memcpy(wifi_config.sta.bssid, evt->bssid, sizeof(wifi_config.sta.bssid));
        }

        ESP_LOGI(TAG, "SSID:%s", wifi_config.sta.ssid);
        ESP_LOGI(TAG, "PASSWORD:%s", wifi_config.sta.password);

        ESP_ERROR_CHECK( esp_wifi_disconnect() );
        ESP_ERROR_CHECK( esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
        esp_wifi_connect();
    } else if (event_base == SC_EVENT && event_id == SC_EVENT_SEND_ACK_DONE) {
        xEventGroupSetBits(s_wifi_event_group, ESPTOUCH_DONE_BIT);
    }
}

static void initialise_wifi(void)
{
    ESP_ERROR_CHECK(esp_netif_init());
    s_wifi_event_group = xEventGroupCreate();
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_t *sta_netif = esp_netif_create_default_wifi_sta();
    assert(sta_netif);

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );

    ESP_ERROR_CHECK( esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL) );
    ESP_ERROR_CHECK( esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &event_handler, NULL) );
    ESP_ERROR_CHECK( esp_event_handler_register(SC_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL) );

    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK( esp_wifi_start() );
}

static void smartconfig_example_task(void * parm)
{
    EventBits_t uxBits;
    ESP_ERROR_CHECK( esp_smartconfig_set_type(SC_TYPE_ESPTOUCH_V2) );
    smartconfig_start_config_t cfg = SMARTCONFIG_START_CONFIG_DEFAULT();
    cfg.enable_log = true;
    cfg.esp_touch_v2_enable_crypt = true;
    cfg.esp_touch_v2_key = "0123456789101112";
    ESP_ERROR_CHECK( esp_smartconfig_start(&cfg) );
    while (1) {
        uxBits = xEventGroupWaitBits(s_wifi_event_group, CONNECTED_BIT | ESPTOUCH_DONE_BIT, true, false, portMAX_DELAY);
        if(uxBits & CONNECTED_BIT) {
            ESP_LOGI(TAG, "WiFi Connected to ap");
        }
        if(uxBits & ESPTOUCH_DONE_BIT) {
            ESP_LOGI(TAG, "smartconfig over");
            esp_smartconfig_stop();
            vTaskDelete(NULL);
        }
    }
}

void app_main(void)
{
    esp_log_level_set("*", ESP_LOG_VERBOSE);
    ESP_ERROR_CHECK( nvs_flash_init() );
    initialise_wifi();
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

I (10087) esp_netif_handlers: sta ip: 192.168.43.79, mask: 255.255.255.0, gw: 192.168.43.1
I (10087) smartconfig_example: WiFi Connected to ap
I (13107) smartconfig_example: smartconfig over
I (18207) wifi:state: run -> init (3c0)
I (18207) wifi:pm stop, total sleep time: 3989071 us / 8775195 us

I (18207) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (18267) smartconfig: SC version: V3.0
I (22357) wifi:ic_enable_sniffer
I (22367) smartconfig: Start to find channel...
I (22367) smartconfig_example: Scan done
I (39647) smartconfig: lock channel: 1
I (39647) smartconfig: TYPE: ESPTOUCH_V2
I (39647) smartconfig: offset:52 tods: 0
I (39647) smartconfig_example: Found channel
I (39657) smartconfig: total_seq: 7
.
.
.
I (43257) smartconfig: ssid: SSID, pwd:
I (43257) wifi:ic_disable_sniffer
I (43267) smartconfig: ACK port: 18266
I (43267) smartconfig_example: Got SSID and password
I (43277) smartconfig_example: SSID:SSID
I (43277) smartconfig_example: PASSWORD:

Alvin1Zhang, 2021-03-08:
Thanks for reporting and sorry for the inconvenience, we will look into.

mohammadjavadpirhadi, 2021-03-08:
> Thanks for reporting and sorry for the inconvenience, we will look into.

Thanks, @Alvin1Zhang
Waiting for it :)

mohammadjavadpirhadi, 2021-03-10:
> Hi @mohammadjavadpirhadi ， Thank you for your report.
> I have solved this issue and will merge the master version soon.

Hi @espxiehang, Thank you for the quick fix!
I would be glad you to notify me once it's merged.

closed 2021-03-12 by espressif-bot