espressif/esp-idf #6680: Storing of structure variable into NVS is not working

kalkimahesh96, 2021-03-08:
#include "pr_nvs.h"

struct qwer{
  int a;
  char b;
  char c[20];
  float f;
};

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  NVS.begin();
}

void loop() {
  // put your main code here, to run repeatedly:
  int a = 36, b= 0;

  qwer obj;
  obj.a = 134;
  obj.b = 3;
  strcpy(obj.c,"mahesh");
  obj.f = 33.456;

  qwer obj1;
  
  NVS.put(1000, obj);

  NVS.get(1000,obj1);
  Serial.println("----------------");
  Serial.println(obj1.a );
  Serial.println((int)obj1.b );
  Serial.println(obj1.c );
  Serial.println(obj1.f );
  Serial.println("----------------");
  delay(2000);

}

I want to store the structure variable into NVS and fetch the value from NVS.
I'm attaching the NVS ESP32 arduino Library.
Kindly help me in this regard.
[pr_nvs .docx](https://github.com/espressif/esp-idf/files/6101579/pr_nvs.docx)

Alvin1Zhang, 2021-03-09:
@kalkimahesh96 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

kalkimahesh96, 2021-03-09:
> @kalkimahesh96 Thanks for reporting. Would you please help provide more details as suggested in the [issue [...]

@Alvin1Zhang  
Expected output:
----------------
134
3
mahesh
33.456
----------------

Actual output :
----------------
134
0

0.00
----------------

I have attached the code and ESP32 nvs library also earlier. Kindly go through it and help me in this regard.

rrtandler, 2024-09-12:
@kalkimahesh96  - is the issue still actual or was you able to figure out the reason why you do not see the expected result ?

rrtandler, 2025-10-31:
@kalkimahesh96 - As this issue is quite obsolete, I am closing it now. Feel free to reopen in case you have some new information.

closed 2025-10-31 by rrtandler