espressif/esp-idf #6682: Wifi disconnect when calling "adc1_get_raw((adc1_channel_t)ADC1_CHANNEL_0);"

MS71, 2021-03-08:
## Environment

- Development Kit:        ESP32-S2-Wrover
- Kit version custom      PCB
- Module or chip used:  ESP32-S2-WROVER
- IDF version                 v4.3-beta1
- Build System:             idf.py
- Compiler version        xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux/Ubuntu
- Using an IDE?:          No
- Power Supply:           Battery

//Detailed problem description goes here.
My ESP32-S2 firmware is connecting to my wifi network.
I am able to ping the ESP32-S2 from my ubuntu machine.
In that moment, when i call the "adc1_get_raw((adc1_channel_t)ADC1_CHANNEL_0)"
function, the ESP32-S2 got disconnected from wifi. From that is, i get a ping timeout.

### Expected Behavior
ESP32-S2 should stay connected wifi, independent  of ADC1.0.

### Actual Behavior
Wifi disconnect when ADC1.0 is read.

### Steps to reproduce
* configure the code with your SSID and PSK
* compile the code with undef ENABLE_ADC
* check if you can ping the ESP32 from the developent pc
* define ENABLE_ADC
* recompile the code.
* ping is possible until adc1_get_raw((adc1_channel_t)ADC1_CHANNEL_0) is called.

### Code to reproduce this issue
derived from examples/wifi/getting_started/station **(my changes in bold)**
/* WiFi station Example
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
**#define ENABLE_ADC**

#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"
**#ifdef ENABLE_ADC
#include <driver/adc.h>
#include <driver/adc_common.h>
#endif**

#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use WiFi configuration that you can set via project configuration menu

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_ESP_MAXIMUM_RETRY  CONFIG_ESP_MAXIMUM_RETRY

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t s_wifi_event_group;

/* The event group allows multiple bits for each event, but we only care about two events:
 * - we are connected to the AP with an IP
 * - we failed to connect after the maximum amount of retries */
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

static const char *TAG = "wifi station";

static int s_retry_num = 0;

static void event_handler(void* arg, esp_event_base_t event_base,
                                int32_t event_id, void* event_data)
{
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
        esp_wifi_connect();
    } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
        if (s_retry_num < EXAMPLE_ESP_MAXIMUM_RETRY) {
            esp_wifi_connect();
            s_retry_num++;
            ESP_LOGI(TAG, "retry to connect to the AP");
        } else {
            xEventGroupSetBits(s_wifi_event_group, WIFI_FAIL_BIT);
        }
        ESP_LOGI(TAG,"connect to the AP fail");
    } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
        ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
        ESP_LOGI(TAG, "got ip:" IPSTR, IP2STR(&event->ip_info.ip));
        s_retry_num = 0;
        xEventGroupSetBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
    }
}

void wifi_init_sta(void)
{
    s_wifi_event_group = xEventGroupCreate();

    ESP_ERROR_CHECK(esp_netif_init());

    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    esp_event_handler_instance_t instance_any_id;
    esp_event_handler_instance_t instance_got_ip;
    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        ESP_EVENT_ANY_ID,
                                                        &event_handler,
                                                        NULL,
                                                        &instance_any_id));
    ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT,
                                                        IP_EVENT_STA_GOT_IP,
                                                        &event_handler,
                                                        NULL,
                                                        &instance_got_ip));

    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .password = EXAMPLE_ESP_WIFI_PASS,
            /* Setting a password implies station will connect to all security modes including WEP/WPA.
             * However these modes are deprecated and not advisable to be used. Incase your Access point
             * doesn't support WPA2, these mode can be enabled by commenting below line */
	     .threshold.authmode = WIFI_AUTH_WPA2_PSK,

            .pmf_cfg = {
                .capable = true,
                .required = false
            },
        },
    };
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
    ESP_ERROR_CHECK(esp_wifi_start() );

    ESP_LOGI(TAG, "wifi_init_sta finished.");

    /* Waiting until either the connection is established (WIFI_CONNECTED_BIT) or connection failed for the maximum
     * number of re-tries (WIFI_FAIL_BIT). The bits are set by event_handler() (see above) */
    EventBits_t bits = xEventGroupWaitBits(s_wifi_event_group,
            WIFI_CONNECTED_BIT | WIFI_FAIL_BIT,
            pdFALSE,
            pdFALSE,
            portMAX_DELAY);

**#ifdef ENABLE_ADC
    vTaskDelay(5000 / portTICK_PERIOD_MS); // allow some few pings
    adc1_get_raw((adc1_channel_t)ADC1_CHANNEL_0);
#endif**

    /* xEventGroupWaitBits() returns the bits before the call returned, hence we can test which event actually
     * happened. */
    if (bits & WIFI_CONNECTED_BIT) {
        ESP_LOGI(TAG, "connected to ap SSID:%s password:%s",
                 EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
    } else if (bits & WIFI_FAIL_BIT) {
        ESP_LOGI(TAG, "Failed to connect to SSID:%s, password:%s",
                 EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
    } else {
        ESP_LOGE(TAG, "UNEXPECTED EVENT");
    }

    /* The event will not be processed after unregister */
    ESP_ERROR_CHECK(esp_event_handler_instance_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, instance_got_ip));
    ESP_ERROR_CHECK(esp_event_handler_instance_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, instance_any_id));
    vEventGroupDelete(s_wifi_event_group);
}

void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

**#ifdef ENABLE_ADC
    adc1_config_width(ADC_WIDTH_BIT_13); 
    adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_DB_6);
#endif**

    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta();
}

**sdkconfig:**
CONFIG_ESP_WIFI_SSID="myssid"
CONFIG_ESP_WIFI_PASSWORD="mypsf"
CONFIG_LOG_DEFAULT_LEVEL_WARN=y

MS71, 2021-03-09:
Another observation, which might related to this.
If i compile the code with LOG_DEFAULT_LEVEL_INFO, then i get next exception while executing it.

I (51) boot: ESP-IDF v4.3-beta1 2nd stage bootloader
I (51) boot: compile time 17:31:32
I (51) boot: chip revision: 0
I (53) boot.esp32s2: SPI Speed      : 80MHz
I (58) boot.esp32s2: SPI Mode       : DIO
I (63) boot.esp32s2: SPI Flash Size : 2MB
I (67) boot: Enabling RNG early entropy source...
I (73) boot: Partition Table:
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (91) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (98) boot:  2 factory          factory app      00 00 00010000 00100000
I (106) boot: End of partition table
I (110) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=170c0h ( 94400) map
I (137) esp_image: segment 1: paddr=000270e8 vaddr=3ffc6450 size=03cf4h ( 15604) load
I (140) esp_image: segment 2: paddr=0002ade4 vaddr=40022000 size=00404h (  1028) load
I (143) esp_image: segment 3: paddr=0002b1f0 vaddr=40022404 size=04e28h ( 20008) load
I (156) esp_image: segment 4: paddr=00030020 vaddr=40080020 size=6d410h (447504) map
I (244) esp_image: segment 5: paddr=0009d438 vaddr=4002722c size=0f218h ( 61976) load
I (270) boot: Loaded app from partition at offset 0x10000
I (271) boot: Disabling RNG early entropy source...
I (282) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (282) cpu_start: Pro cpu up.
I (342) cpu_start: Pro cpu start user code
I (342) cpu_start: cpu freq: 160000000
I (343) cpu_start: Application information:
I (345) cpu_start: Project name:     wifi_station
I (351) cpu_start: App version:      66c5f11-dirty
I (356) cpu_start: Compile time:     Mar  9 2021 17:31:54
I (362) cpu_start: ELF file SHA256:  1d96bf36bae96e87...
I (368) cpu_start: ESP-IDF:          v4.3-beta1
I (373) heap_init: Initializing. RAM available for dynamic allocation:
I (381) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (387) heap_init: At 3FFCE8A8 len 0002D758 (181 KiB): DRAM
I (393) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (400) spi_flash: detected chip: gd
I (404) spi_flash: flash io: dio
W (408) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (425) cpu_start: Starting scheduler on PRO CPU.
I (435) wifi station: ESP_WIFI_MODE_STA
I (455) wifi:wifi driver task: 3ffd6240, prio:23, stack:6656, core=0
I (455) system_api: Base MAC address is not set
I (455) system_api: read default base MAC address from EFUSE
I (465) wifi:wifi firmware version: 6b2834e
I (465) wifi:wifi certification version: v7.0
I (465) wifi:config NVS flash: enabled
I (465) wifi:config nano formating: disabled
I (475) wifi:Init data frame dynamic rx buffer num: 32
I (475) wifi:Init management frame dynamic rx buffer num: 32
I (485) wifi:Init management short buffer num: 32
I (485) wifi:Init dynamic tx buffer num: 32
I (495) wifi:Init static rx buffer size: 1600
I (495) wifi:Init static rx buffer num: 10
I (495) wifi:Init dynamic rx buffer num: 32
I (505) wifi_init: rx ba win: 6
I (505) wifi_init: tcpip mbox: 32
I (515) wifi_init: udp mbox: 6
I (515) wifi_init: tcp mbox: 6
I (515) wifi_init: tcp tx win: 5744
I (525) wifi_init: tcp rx win: 5744
I (525) wifi_init: tcp mss: 1440
I (535) wifi_init: WiFi IRAM OP enabled
I (535) wifi_init: WiFi RX IRAM OP enabled
I (545) phy_init: phy_version 1300,2887b9c,Dec 16 2020
I (675) wifi:mode : sta (7c:df:a1:0b:e4:4e)
I (675) wifi:enable tsf
I (675) wifi station: wifi_init_sta finished.
[... 29 lines cut ...]
0x400e5465: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)

0x400ed425: esp_log_writev at /home/maik/workspace/esp32/latest/components/log/log.c:189

0x4002f975: esp_log_write at /home/maik/workspace/esp32/latest/components/log/log.c:199

0x400d6785: lib_printf at /home/maik/workspace/esp32/latest/components/esp_wifi/src/lib_printf.c:47 (discriminator 13)

0x400d67af: phy_printf at /home/maik/workspace/esp32/latest/components/esp_wifi/src/lib_printf.c:57

0x4002695b: pll_vol_cal at ??:?

0x40026a35: wifi_track_pll_cap at ??:?

0x400284a4: tx_pwctrl_background at ??:?

0x400c2d9f: lmacProcessTxComplete at ??:?

0x400328d1: ppTask at ??:?

0x4002cefd: vPortTaskWrapper at /home/maik/workspace/esp32/latest/components/freertos/port/xtensa/port.c:168

ELF file SHA256: 1d96bf36bae96e87

Rebooting...

lienbacher, 2021-03-22:
I am also running into this problem.

Alvin1Zhang, 2021-03-23:
Thanks for reporting, we have a fix which is now under internal reviewing, the issue will close automatically once the fix is available on GitHub. Thanks.

MS71, 2021-03-30:
Hi,
i can't see the fix on branch:latest. Is this already pushed to github?
Maik

lienbacher, 2021-03-30:
@Alvin1Zhang this issue is quite a serious breaker. It would be great if this could be pushed real soon. thank you!

closed 2021-04-01 by espressif-bot

william-ferguson-au, 2021-04-09:
@Alvin1Zhang I'm still seeing this fault on master v4.3-dev-2136-gb0150615d
Has it been merged?

If not, can you give an ETA?

espxiehang, 2021-04-09:
Hi @william-ferguson-au ,The fix version of release/v4.3 has not been synchronized to GitHub.

You can test it with cc6bfcd9ac17036824e005f1569df4b66c6415d3 version.

MS71, 2021-04-09:
Hi,
the issue is NOT solved with next commit id.
Same with master and v4.3 release branch.

commit cc6bfcd9ac17036824e005f1569df4b66c6415d3 (HEAD)
Author: Armando <douyiwen@espressif.com>
Date:   Fri Mar 26 12:18:31 2021 +0800

    adc: apply adc power API to adc driver
    
    Closes https://github.com/espressif/esp-idf/issues/6269
    Closes https://github.com/espressif/esp-idf/issues/6682

Tested with modified station sample code. (see first comment of this issue)

MS71, 2021-04-09:
somehow, the wifi connection is not stable when adc is called while connected.

E (25442) wifi station: calling **adc1_get_raw**
I (25742) esp_netif_handlers: sta ip: 192.168.0.165, mask: 255.255.255.0, gw: 192.168.0.1
I (25742) wifi station: got ip:192.168.0.165
E (26442) wifi station: calling **adc1_get_raw**
I (27432) wifi:state: run -> init (22c0)
I (27432) wifi:pm stop, total sleep time: 1400222 us / 2776383 us

W (27442) wifi:<ba-del>idx
I (27442) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (27442) wifi station: retry to connect to the AP
I (27452) wifi station: connect to the AP fail
I (27452) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (27462) wifi:state: init -> auth (b0)
E (27462) wifi station: calling **adc1_get_raw**
I (27482) wifi:state: auth -> assoc (0)
I (27492) wifi:state: assoc -> run (10)
I (27522) wifi:connected with mySSID, aid = 2, channel 6, BW20, bssid = 2c:91:ab:38:1d:7f
I (27522) wifi:security: WPA3-SAE, phy: bgn, rssi: -60
I (27522) wifi:pm start, type: 1

I (27522) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (28242) esp_netif_handlers: sta ip: 192.168.0.165, mask: 255.255.255.0, gw: 192.168.0.1
I (28242) wifi station: got ip:192.168.0.165
E (28472) wifi station: calling **adc1_get_raw**

void wifi_init_sta(void)
{
    s_wifi_event_group = xEventGroupCreate();

    ESP_ERROR_CHECK(esp_netif_init());

    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    esp_event_handler_instance_t instance_any_id;
    esp_event_handler_instance_t instance_got_ip;
    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        ESP_EVENT_ANY_ID,
                                                        &event_handler,
                                                        NULL,
                                                        &instance_any_id));
    ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT,
                                                        IP_EVENT_STA_GOT_IP,
                                                        &event_handler,
                                                        NULL,
                                                        &instance_got_ip));

    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .password = EXAMPLE_ESP_WIFI_PASS,
            /* Setting a password implies station will connect to all security modes including WEP/WPA.
             * However these modes are deprecated and not advisable to be used. Incase your Access point
             * doesn't support WPA2, these mode can be enabled by commenting below line */
	     .threshold.authmode = WIFI_AUTH_WPA2_PSK,

            .pmf_cfg = {
                .capable = true,
                .required = false
            },
        },
    };
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
    ESP_ERROR_CHECK(esp_wifi_start() );

    ESP_LOGI(TAG, "wifi_init_sta finished.");

    /* Waiting until either the connection is established (WIFI_CONNECTED_BIT) or connection failed for the maximum
     * number of re-tries (WIFI_FAIL_BIT). The bits are set by event_handler() (see above) */
    EventBits_t bits = xEventGroupWaitBits(s_wifi_event_group,
            WIFI_CONNECTED_BIT | WIFI_FAIL_BIT,
            pdFALSE,
            pdFALSE,
            portMAX_DELAY);

    /* xEventGroupWaitBits() returns the bits before the call returned, hence we can test which event actually
     * happened. */
    if (bits & WIFI_CONNECTED_BIT) {
        ESP_LOGI(TAG, "connected to ap SSID:%s password:%s",
                 EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
    } else if (bits & WIFI_FAIL_BIT) {
        ESP_LOGI(TAG, "Failed to connect to SSID:%s, password:%s",
                 EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
    } else {
        ESP_LOGE(TAG, "UNEXPECTED EVENT");
    }

**#ifdef ENABLE_ADC
    vTaskDelay(10000 / portTICK_PERIOD_MS); // allow some few pings
    while(1)
    {
        ESP_LOGE(TAG, "calling adc1_get_raw");
        adc1_get_raw((adc1_channel_t)ADC1_CHANNEL_0);
        vTaskDelay(1000 / portTICK_PERIOD_MS); // allow some few pings
    }
#endif**

    /* The event will not be processed after unregister */
    ESP_ERROR_CHECK(esp_event_handler_instance_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, instance_got_ip));
    ESP_ERROR_CHECK(esp_event_handler_instance_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, instance_any_id));
    vEventGroupDelete(s_wifi_event_group);
}

lienbacher, 2021-06-06:
Is there any update to this issue?

briantanabe, 2021-07-20:
Bump, this should not be closed!

nopnop2002, 2021-11-22:
@MS71 

I have same issues.
Can you reopen this issues?

nopnop2002, 2021-11-22:
@MS71 

ESP-IDF V4.4 has the same problem.

Can you re-open this issues?

```
I (0) cpu_start: App cpu up.
I (432) cpu_start: Pro cpu start user code
I (432) cpu_start: cpu freq: 240000000
I (432) cpu_start: Application information:
I (437) cpu_start: Project name:     canvas-gauges
I (442) cpu_start: App version:      ed733ce-dirty
I (448) cpu_start: Compile time:     Nov 21 2021 17:26:50
I (454) cpu_start: ELF file SHA256:  5fb2d61424ac76c6...
I (460) cpu_start: ESP-IDF:          v4.4-dev-3573-gfb24a2941c-dirty
I (467) heap_init: Initializing. RAM available for dynamic allocation:
I (474) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (480) heap_init: At 3FFB8280 len 00027D80 (159 KiB): DRAM
I (487) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (493) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (499) heap_init: At 400953D8 len 0000AC28 (43 KiB): IRAM
I (506) spi_flash: detected chip: winbond
I (510) spi_flash: flash io: dio
W (514) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (528) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (575) main: ESP_WIFI_MODE_STA
I (585) wifi:wifi driver task: 3ffc1920, prio:23, stack:6656, core=0
I (585) system_api: Base MAC address is not set
I (585) system_api: read default base MAC address from EFUSE
I (595) wifi:wifi firmware version: f84e709
I (595) wifi:wifi certification version: v7.0
I (595) wifi:config NVS flash: enabled
I (595) wifi:config nano formating: disabled
I (605) wifi:Init data frame dynamic rx buffer num: 32
I (605) wifi:Init management frame dynamic rx buffer num: 32
I (615) wifi:Init management short buffer num: 32
I (615) wifi:Init dynamic tx buffer num: 32
I (625) wifi:Init static rx buffer size: 1600
I (625) wifi:Init static rx buffer num: 10
I (625) wifi:Init dynamic rx buffer num: 32
I (635) wifi_init: rx ba win: 6
I (635) wifi_init: tcpip mbox: 32
I (645) wifi_init: udp mbox: 6
I (645) wifi_init: tcp mbox: 6
I (645) wifi_init: tcp tx win: 5744
I (655) wifi_init: tcp rx win: 5744
I (655) wifi_init: tcp mss: 1440
I (665) wifi_init: WiFi IRAM OP enabled
I (665) wifi_init: WiFi RX IRAM OP enabled
I (675) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (765) wifi:mode : sta (3c:71:bf:4f:c1:a0)
I (765) wifi:enable tsf
I (775) main: wifi_init_sta finished.
I (795) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (795) wifi:state: init -> auth (b0)
I (805) wifi:state: auth -> assoc (0)
I (825) wifi:state: assoc -> run (10)
I (935) wifi:connected with aterm-d5a4ee-g, aid = 6, channel 1, BW20, bssid = f8:b7:97:36:de:52
I (935) wifi:security: WPA2-PSK, phy: bgn, rssi: -52
I (935) wifi:pm start, type: 1

I (1005) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2075) esp_netif_handlers: sta ip: 192.168.10.161, mask: 255.255.255.0, gw: 192.168.10.1
I (2075) main: got ip:192.168.10.161
I (2075) main: connected to ap SSID:aterm-d5a4ee-g
I (2085) main: mdns hostname set to: [esp32-server]

Execute adc1_get_raw() at 1 second intervals:

I (5553205) wifi:bcn_timout,ap_probe_send_start
I (5555705) wifi:ap_probe_send over, resett wifi status to disassoc
I (5555705) wifi:state: run -> init (c800)
I (5555715) wifi:pm stop, total sleep time: 4308259922 us / 5554770201 us

W (5555715) wifi:<ba-del>idx
W (5555715) wifi:<ba-del>idx
I (5555715) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
```

clemrdh, 2024-12-05:
Please re-open this issue as this is not solved ?