espressif/esp-idf #6683: FTM support with WIFI_MODE_APSTA

roger-, 2021-03-08:
Not sure if it is supported today, but I would like to have the following setup:

`[non-ESP AP] --- [ESP32 in APSTA mode] --- FTM --- [ESP32 in STA mode]`

Is this possible?

sagb2015, 2021-03-10:
@roger-  We do not support FTM in concurrent mode yet because it needs additional handling for off-channel operation. ie when ESP station performs FTM on different channel than the one on which ESP softAP is active.

roger-, 2021-03-10:
Thanks @sagb2015   

So it still won't work if they're on the same channel (isn't that necessary either way)?

And will it not work if either the APSTA or the STA initiate the FTM?

oursland, 2023-04-26:
I am running two ESP32-S3 nodes, each running in APSTA mode, and they are able to perform FTM ranging with each other.

nachiketkukade, 2024-03-04:
Above comment is right. A device in AP-STA mode can initiate FTM with ex-AP but only in same channel. In case of different channel, it will likely end up in failure due to contention with SoftAP mode. We're adding changes that will simply return error in this case

closed 2024-03-04 by nachiketkukade