espressif/esp-idf #6684: Bluetooth connect timeout after Android phone app disconnect and reconnect

cattledogGH, 2021-03-09:
ESP32 Dev Module, Arduino Core 1.0.5, 

BluetoothSerial.h included with core library properties shows
`name=BluetoothSerial
version=1.0
author=Evandro Copercini
maintainer=Evandro Copercini
sentence=Simple UART to Classical Bluetooth bridge for ESP32
paragraph=
category=Communication
url=
architectures=esp32`

When using BluetoothSerial.h and library example SerialToSerialBT with an Android phone (Kai Morich[ SerialBluetoothTerminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en_US&gl=US)) the phone app can not reconnect after disconnect without restarting the ESP32.

Initial connection and example code work as expected. Disconnect phone app. Phone will not reconnect without ESP32 reset.

I believe that this issue is related to this issue which was repaired in the esp-idf
[https://github.com/espressif/esp-idf/issues/1537](url)

Request that the reconnection error be corrected in BluetoothSerial.h for ESP32.

cattledogGH, 2021-03-09:
Posted to esp-idf in error. Should have been posted to
[https://github.com/espressif/arduino-esp32](url)

Can this be moved?

TianaESP, 2021-03-09:
Hi @cattledogGH,

It's ok to leave it here. For the issue you mentioned, can you please provide more information including IDF version you use and any necessary log you have? Thanks.