espressif/esp-idf #6686: Failed to initialize 64 GB microsd card

petercipov, 2021-03-09:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    platformio/framework-espidf          3.40200.210118 @ ~3.40200.0  
- Build System:         [platformio]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [macOS]
- Using an IDE?: [Yes (vscode + platfomio)]
- Power Supply:         [USB]

## Problem Description
Using 64 GB micro sd card (Kingston MicroSDXC 64GB Canvas Go! Plus)does not work with example code https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card This sd card works well with arduino uno and is well formatted with fat32

I am using 3.3 V micro sd card module to connect to esp32 board.

I have tried 32 GB (Samsung MicroSDHC 32GB EVO Plus) and 16GB (SanDisk MicroSDHC 16GB Ultra Android) micro sd card. Both works as expected

I have spotted in logs that spi commands (52 and 5) are not supported by particular 64 GB card.

### Expected Behavior
example code works with 64 Gb micro sd card or update documentation that such cards are not supported.

### Actual Behavior

Failed to initialize the card

### Steps to reproduce

1. build example code https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card
2. upload into esp32 board 
3. connect micro sd module 3.3V GND, PIN_NUM_MISO 2
PIN_NUM_MOSI 15
PIN_NUM_CLK  14
PIN_NUM_CS   13
4. insert 64 GB microSD card
5. connect to serial
6. press reset button

## Debug Logs

```
[0;32mI (29) boot: ESP-IDF 3.40200.210118 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 20:12:10␛[0m
␛[0;32mI (29) boot: chip revision: 1␛[0m
␛[0;32mI (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0␛[0m
␛[0;32mI (39) boot.esp32: SPI Speed      : 40MHz␛[0m
␛[0;32mI (44) boot.esp32: SPI Mode       : DIO␛[0m
␛[0;32mI (48) boot.esp32: SPI Flash Size : 4MB␛[0m
␛[0;32mI (53) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (58) boot: Partition Table:␛[0m
␛[0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (84) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (92) boot: End of partition table␛[0m
␛[0;32mI (96) boot_comm: chip revision: 1, min. application chip revision: 0␛[0m
␛[0;32mI (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0b2ec ( 45804) map␛[0m
␛[0;32mI (131) esp_image: segment 1: paddr=0x0001b314 vaddr=0x3ffb0000 size=0x021e0 (  8672) load␛[0m
␛[0;32mI (135) esp_image: segment 2: paddr=0x0001d4fc vaddr=0x40080000 size=0x00404 (  1028) load␛[0m
␛[0;32mI (138) esp_image: segment 3: paddr=0x0001d908 vaddr=0x40080404 size=0x02710 ( 10000) load␛[0m
␛[0;32mI (152) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x1fe74 (130676) map␛[0m
␛[0;32mI (210) esp_image: segment 5: paddr=0x0003fe9c vaddr=0x40082b14 size=0x09598 ( 38296) load␛[0m
␛[0;32mI (236) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (236) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (236) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (240) cpu_start: Application information:␛[0m
␛[0;32mI (245) cpu_start: Project name:     dinmeter-cam-idf␛[0m
␛[0;32mI (250) cpu_start: App version:      9113913-dirty␛[0m
␛[0;32mI (256) cpu_start: Compile time:     Mar  2 2021 20:09:47␛[0m
␛[0;32mI (262) cpu_start: ELF file SHA256:  b1c0634d45acdd02...␛[0m
␛[0;32mI (268) cpu_start: ESP-IDF:          3.40200.210118␛[0m
␛[0;32mI (274) cpu_start: Starting app cpu, entry point is 0x40081fd0␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (284) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (291) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM␛[0m
␛[0;32mI (297) heap_init: At 3FFB2AA0 len 0002D560 (181 KiB): DRAM␛[0m
␛[0;32mI (303) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (310) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (316) heap_init: At 4008C0AC len 00013F54 (79 KiB): IRAM␛[0m
␛[0;32mI (322) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (341) spi_flash: detected chip: generic␛[0m
␛[0;32mI (341) spi_flash: flash io: dio␛[0m
␛[0;32mI (342) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (350) example: Initializing SD card␛[0m
␛[0;32mI (350) example: Using SPI peripheral␛[0m
␛[0;32mI (360) example: Attaching FS␛[0m
␛[0;32mI (360) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (370) sdspi_transaction: cmd=52, R1 response: command not supported␛[0m
␛[0;32mI (420) sdspi_transaction: cmd=5, R1 response: command not supported␛[0m
␛[0;31mE (1420) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107␛[0m
␛[0;31mE (1420) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).␛[0m
␛[0;32mI (1420) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;31mE (1430) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.␛[0m
```

chegewara, 2021-03-09:
https://github.com/espressif/esp-iot-solution/tree/master/examples/audio/wav_player#card-fails-to-initialize-with-sdmmc_init_sd_scr-send_scr-1-returned-0x107-error

I would also suggest to decrease frequency to 8-10MHz for testing.

Can you post link to sd card module, or picture of it, please.

petercipov, 2021-03-09:
I am using microSD Card modul SPI 3.3V, datasheet [SD-MINI.pdf](https://github.com/espressif/esp-idf/files/6111081/SD-MINI.pdf)

I have tried `host.max_freq_khz = SDMMC_FREQ_PROBING;`  no working 
I have connected logic analyser and decoded spi, and found same as in debug logs, those commands are unssuported by microsd card
[image]

I have tied to create my own module https://www.mischianti.org/2019/12/15/how-to-use-sd-card-with-esp8266-and-arduino/ , got same issue with unssuported command 52 and 5

I have created bug here as then I tried my last option - try different sd cards (mentioned above) and they work flawlesly. I see this as an issue in a driver.

igrr, 2021-03-09:
@petercipov Could  you please save and attach here the data captured by the logic analyzer?

CMD5 and CMD52 are indeed unsupported by memory cards (these are SDIO commands) but this shouldn't influence subsequent initialization (per standard, at least...) I'd like to see what happens later in the initialization process, maybe there is some other issue.

igrr, 2021-03-09:
By the way, you can also check if indeed sending the IO commands causes this card to fail during initialization. To do this, delete two lines:

https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/components/sdmmc/sdmmc_init.c#L45-L46

and

https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/components/sdmmc/sdmmc_init.c#L54-L55

then try to run the example again.

petercipov, 2021-03-10:
Hello @igrr ,

here is dslogic file with full communication (using dsview 1.1.2)

[64gb-dslogic.dsl.zip](https://github.com/espressif/esp-idf/files/6118035/64gb-dslogic.dsl.zip)

I have recompiled firmware with commenting those two commands, got different error

`
␛[0;32mI (29) boot: ESP-IDF 3.40200.210118 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 19:16:16␛[0m
␛[0;32mI (29) boot: chip revision: 1␛[0m
␛[0;32mI (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0␛[0m
␛[0;32mI (39) boot.esp32: SPI Speed      : 40MHz␛[0m
␛[0;32mI (44) boot.esp32: SPI Mode       : DIO␛[0m
␛[0;32mI (48) boot.esp32: SPI Flash Size : 4MB␛[0m
␛[0;32mI (53) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (58) boot: Partition Table:␛[0m
␛[0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (84) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (92) boot: End of partition table␛[0m
␛[0;32mI (96) boot_comm: chip revision: 1, min. application chip revision: 0␛[0m
␛[0;32mI (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0a98c ( 43404) map␛[0m
␛[0;32mI (130) esp_image: segment 1: paddr=0x0001a9b4 vaddr=0x3ffb0000 size=0x021c8 (  8648) load␛[0m
␛[0;32mI (134) esp_image: segment 2: paddr=0x0001cb84 vaddr=0x40080000 size=0x00404 (  1028) load␛[0m
␛[0;32mI (137) esp_image: segment 3: paddr=0x0001cf90 vaddr=0x40080404 size=0x03088 ( 12424) load␛[0m
␛[0;32mI (152) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x1de64 (122468) map␛[0m
␛[0;32mI (206) esp_image: segment 5: paddr=0x0003de8c vaddr=0x4008348c size=0x08a44 ( 35396) load␛[0m
␛[0;32mI (230) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (230) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (230) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (234) cpu_start: Application information:␛[0m
␛[0;32mI (239) cpu_start: Project name:     sdcard-demo␛[0m
␛[0;32mI (244) cpu_start: App version:      0771137␛[0m
␛[0;32mI (249) cpu_start: Compile time:     Mar 10 2021 19:13:22␛[0m
␛[0;32mI (255) cpu_start: ELF file SHA256:  9bfe1305702e822c...␛[0m
␛[0;32mI (261) cpu_start: ESP-IDF:          3.40200.210118␛[0m
␛[0;32mI (267) cpu_start: Starting app cpu, entry point is 0x40081fb0␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (277) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (284) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM␛[0m
␛[0;32mI (290) heap_init: At 3FFB2A68 len 0002D598 (181 KiB): DRAM␛[0m
␛[0;32mI (296) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (303) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (309) heap_init: At 4008BED0 len 00014130 (80 KiB): IRAM␛[0m
␛[0;32mI (315) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (334) spi_flash: detected chip: generic␛[0m
␛[0;32mI (334) spi_flash: flash io: dio␛[0m
␛[0;32mI (334) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (342) example: Initializing SD card␛[0m
␛[0;32mI (342) example: Using SPI peripheral␛[0m
␛[0;32mI (352) example: Attaching FS␛[0m
␛[0;32mI (352) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;33mW (402) vfs_fat_sdmmc: failed to mount card (1)␛[0m
␛[0;31mE (402) vfs_fat_sdmmc: mount_to_vfs failed (0xffffffff).␛[0m
␛[0;32mI (402) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;31mE (412) example: Failed to mount filesystem. If you want the card to be formatted, set the EXAMPLE_FORMAT_IF_MOUNT_FAILED menuconfig option.␛[0m
`

I have set EXAMPLE_FORMAT_IF_MOUNT_FAILED to also to true, but got same error.

igrr, 2021-03-28:
Hi @petercipov, sorry for not replying sooner. So far I have managed to get the same model of the card and to reproduce the issue. Unfortunately, I haven't been able to figure out what makes this specific card model to time out replying to SEND_OP_COND command, yet. Will let you know when there is some progress.

chegewara, 2021-03-30:
I can confirm regression in master branch.
Tested the same example and exactly the same hardware, including sd card, with master branch 1067b28707e527f177752741e3aa08b5dc64a4d7 and b0150615dff529662772a60dcb57d5b559f480e2
This commit works:
b0150615dff529662772a60dcb57d5b559f480e2
this commit does not works
1067b28707e527f177752741e3aa08b5dc64a4d7

I am reporting here due to cmd 52
```
I (333) example: Using SPI peripheral
I (333) gpio: GPIO[40]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (343) sdspi_transaction: cmd=52, R1 response: command CRC error
E (353) sdmmc_io: sdmmc_io_reset: unexpected return: 0x109
E (363) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
I (363) gpio: GPIO[40]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (373) example: Failed to initialize the card (ESP_ERR_INVALID_CRC). Make sure SD card lines have pull-up resistors in place.
```

board: S2 saola
esp-idf commit: see above 
code: sd_card example

gracianomor, 2021-06-29:
Hello. I'm new in this discouus.

I'am using the driver SDSPI by esp-tool and when I off  my device and I up I have this issue:

I (315) spi_flash: detected chip: generic
I (318) spi_flash: flash io: dio
W (322) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (336) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (346) example: Initializing SD card
I (346) example: Using SPI peripheral
I (356) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (366) sdspi_transaction: cmd=52, R1 response: command not supported
I (416) sdspi_transaction: cmd=5, R1 response: command not supported
E (1416) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (1416) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
I (1416) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1426) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.

can some one help me, please?

and I have another question.

in this driver, How can I to open a file?

I'm using an task to open the file but I have this issue;

E (1126) example: Failed to open file for writing
E (1136) FreeRTOS: FreeRTOS Task "Prueba_sdcard_t" should not return, Aborting now!

Thank you

closed 2024-03-04 by espressif-bot