espressif/esp-idf #6687: Inaccurate timing

helittr, 2021-03-10:
## Environment
- Development Kit: [none]
- Module or chip used: [ESP32-WROOM-32]
- IDF version: v4.2
- Build System: [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: [Windows]
- Using an IDE?: [No]
- Power Supply: [USB]
## Problem Description
The ESP32's time is not accurate.After five minutes of deepsleep, the time was more than some seconds behind the actual time.

Do you have any idea about this? how should i do to fix this problem?

## Debug Logs
I recorded the print time of the log.
- Before deepsleep
```
[2021-03-10 11:19:48]                I (547) Time: 2021-03-10 11:19:47
[2021-03-10 11:19:53]               W (4847) deepsleep: Entering deep sleep, 300s
```
- After deepsleep
```
[2021-03-10 11:04:45]                I (547)  Time: 2021-03-10 11:04:19
[2021-03-10 11:04:49]               W (4927) deepsleep: Entering deep sleep, 300s
```

negativekelvin, 2021-03-10:
https://github.com/espressif/esp-idf/blob/39cbf2f7ed4810913031055065da08227020e8b4/components/esp32/Kconfig#L539-L571

helittr, 2021-03-10:
```shell
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
# CONFIG_ESP32_RTC_CLK_SRC_INT_RC is not set
# CONFIG_ESP32_RTC_CLK_SRC_EXT_CRYS is not set
# CONFIG_ESP32_RTC_CLK_SRC_EXT_OSC is not set
CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y
CONFIG_ESP32_RTC_CLK_CAL_CYCLES=1024
```

parapilot, 2021-12-17:
[sntp_deep_sleep_drift.zip](https://github.com/espressif/esp-idf/files/7735450/sntp_deep_sleep_drift.zip)
observations done with IDFv4.3.1
- switching from 150kHz RTC to 8.5MHz derived RTC makes time deviation worse
- observations are independent of using dynmaic power management or fixed frequency for xtensa cores
- recalibration of RTC slow clock is done on every wakeup
the basic idea was to have a minimal sample : on power on reset once wifi connection is established and ntp_sync performed.
data collected to host in every cycle time before sleep, time on wakeup, slow clock cal
The python monitor compares the timestamps to each other and to the host clock as external time reference.
What I don't understand is why the time offset observed can be split in a runtime compensation and rtc_jitter.
The runtime component in principle is clear: ESP32 startup and processing takes time so if I want a fixed interval sleep time should be INTERVAL - runtime. But what I don't understand is why I observe a quite strong deviation that makes the RTC clock go ahead of reference time. Thus increasing the sleep and tuning the RTC clock with this JITTER value improves. And this JITTER coefficient seems to scale linearly with the INTERVAL length. So I don't understand why this is better reflected in the RTC calib value. Unfortunately I was not able to lookup the complete time keeping code since it seems to go down into the toolchain.

tomocchi3411, 2022-03-26:
Hello, I will post for the first time.
Recently, I found a way to get a relatively accurate time with the built-in RTC's 150kHz clock.
If you write the time for one cycle of the 150kHz clock (msec in Q13.19 format) to RTC_CNTL_STORE1_REG of ESP32, RTC seems to calculate the time based on this cycle.
The RTC clock cycle (Q13.19 format) when the ESP32 main clock is running can be obtained with the rtc_clk_cal () function, but in the experiment, the clock cycle during deep sleep seems to be about 0.7% shorter than this, so it is corrected to RTC_CNTL_STORE1_REG. Write the value you have set.
It's also a good idea to wake up and update RTC_CNTL_STORE1_REG every few minutes to reduce the effects of temperature drift.
I have uploaded a test sketch for the arduino IDE instead of ESP-IDF.
In my experience, the correction coefficient rtcClkRate should be around 0.993, but there are individual differences, so please change it in various ways.
If set properly, it can operate with an error of less than 1 minute a day without synchronizing with NTP.
thank you
[testSlowClock.zip](https://github.com/espressif/esp-idf/files/8354710/testSlowClock.zip)
.

parapilot, 2022-03-27:
Interesting, tomocchi3411 ended up using the same compensation mechanism (modifying slow clk cal register). If his conclusion is correct, that the 150kHz slow clock speed changes between power modes, this explaines why it's possible to actually compensate. It would be interesting to hear from espressif what could cause this and if there is a state of the chip e.g. wake up stub or something to measure the actual deep sleep slow clock frequency. Otherwise it might be tedious to measure the compensation factor for each chip.

tomocchi3411, 2022-03-28:
Thank you for your prompt reply, parapilot.

I think the reason why the oscillator frequency of the 150kHz Slow Clock fluctuates is the temperature drift and voltage change.
Since the current consumption of the ESP32 is different between Wake up and Deep Sleep, I expect that the oscillator voltage will be higher in Deep Sleep and therefore the clock frequency will be slightly higher.
The method of obtaining the compensation value for each individual can be easily calculated from the time difference between the time obtained from NTP and the time of the built-in RTC.
  rtcClkRate = (timePassed - timeError) / timePassed
For example, if the built-in RTC is ahead of NTP for 1 minute (60 sec) in 3 hours (10,800 sec).
  rtcClkRate = 0.9944 = (10,800 - 60) / 10,800
Since rtc_clk_cal () is the cycle at the time of Wakeup, if you multiply this by rtcClkRate, the clock cycle during DeepSleep can be obtained.
I would like comments from espressif, too.
thank you.

parapilot, 2022-03-28:
Hi tomocchi3411. Your explanation sounds reasonable. I measured temperature drift against ambient over days without thinking about stronger short term changes of substrate temperature. Hower you're right that changes in power consumption and voltage level have a high short term impact on substrate temperature and RC frequency. Thus a first speculation could be that part of the problem is the disfunctional internal temperature point which was removed over time.
Besides the compensation of the internal RC clock I also played with accurate timing using an external clock quartz. There my second problem was, that I could not figure out how exactly the handover between RTC clock and precision timing during wakeup is done. Thus I was not able to measure the exact wakeup time since parts of bootloader are processed before I could obtain the wakeup timestamp. Assuming timetamp before sleep + sleep period with clock quarz was good enough.

tomocchi3411, 2022-04-01:
Hi parapilot, thank you for your reply.
As you say, it's nice to have an external crystal, but I think helittr wants the built-in RTC to be as accurate as possible.
The attached Arduino sketch automatically calibrates the built-in RTC so that you can find the appropriate coefficient rtcClkRate.
After the first NTP synchronization, it wakes up from deepsleep every 5 minutes and outputs the built-in RTC time to the serial port.
Also, when the SW connected to GPIO34 is pressed, the appropriate rtcClkRate will be calculated and changed.
The following is an example of operation after rtcClkRate caliblation.
The left side is the actual time of the PC, and the right side is the built-in RTC time of ESP32.

10:50:19.655 -> 2022/04/01(Fri)_10:50:19.750 from WiFi
10:50:19.655 -> RTC Clk Period : 6.035938 usec, RTC Clk Freq : 165.674 kHz
10:50:19.702 -> RTC Clk Rate   : 0.993479

10:55:17.104 -> 2022/04/01(Fri)_10:55:19.753 from internal rtc
10:55:17.104 -> RTC Clk Period : 5.983398 usec, RTC Clk Freq : 167.129 kHz

11:00:16.928 -> 2022/04/01(Fri)_11:00:19.752 from internal rtc
11:00:16.928 -> RTC Clk Period : 5.979930 usec, RTC Clk Freq : 167.226 kHz

11:05:16.946 -> 2022/04/01(Fri)_11:05:19.751 from internal rtc
11:10:17.019 -> 2022/04/01(Fri)_11:10:19.750 from internal rtc
11:15:16.909 -> 2022/04/01(Fri)_11:15:19.750 from internal rtc
11:20:17.010 -> 2022/04/01(Fri)_11:20:19.749 from internal rtc

11:50:17.093 -> 2022/04/01(Fri)_11:50:19.744 from internal rtc
11:50:17.093 -> RTC Clk Period : 5.983373 usec, RTC Clk Freq : 167.129 kHz

12:30:16.750 -> 2022/04/01(Fri)_12:30:19.739 from internal rtc
12:35:16.810 -> 2022/04/01(Fri)_12:35:19.738 from internal rtc
12:40:16.759 -> 2022/04/01(Fri)_12:40:19.737 from internal rtc
12:45:16.789 -> 2022/04/01(Fri)_12:45:19.736 from internal rtc

12:50:16.635 -> 2022/04/01(Fri)_12:50:19.736 from internal rtc
12:50:16.635 -> RTC Clk Period : 5.977148 usec, RTC Clk Freq : 167.303 kHz

15:50:16.907 -> 2022/04/01(Fri)_15:50:19.712 from internal rtc
15:50:16.907 -> RTC Clk Period : 5.994775 usec, RTC Clk Freq : 166.811 kHz

At first, the RTC time synchronized with NTP and the PC time are almost the same, but after 10 minutes, the RTC will advance by about 3 seconds.
After that, the difference of 3 seconds is maintained for 5 hours, so it can be said that it is reasonably accurate.
The first 3 seconds difference is probably due to the high power consumption of WiFi, so I think the temperature rise had an effect.
In actual use, it is necessary to consider the current and temperature rise during awake state.
Thank you
[testSlowClockAutocalib.zip](https://github.com/espressif/esp-idf/files/8395698/testSlowClockAutocalib.zip)

helittr, 2022-04-01:
**Thank your suggestions very much.**

I has konwn that the time is  more accurate when  RTC clk source is **150KHz** RC by trying all the internal clock sources. This deviation is within my tolerance. Your discussions wil help me further reduce the time deviation.
But I'm a little confused，as [parapilot](https://github.com/parapilot) said, why do switch from 150kHz RTC to 8.5MHz derived RTC makes time deviation worse? The 8.5MHZ clock should has Better stability than 150KHz RTC.

moose4lord, 2022-08-21:
@tomocchi3411, I wanted to thank you for your comments and code. I have an ESP32 project that gets the current time from an NTP server, runs for about 2 hours, then goes into deep sleep mode for 22 hours (it wakes up the next day and repeats the process). While it is deep sleeping the rtc time keeping is not very accurate, and the app would consequently wake up several minutes early every day.

I ran your rtc calibration sketch to determine the proper calibration offset, and incorporated that into my own sketch and now the ESP32 keeps much better time when in deep sleep mode. I thought it was a clever solution, although I can find very little documentation about using the RTC_CNTL_STORE1_REG register to fine tune the rtc clock, so kudos to you for figuring that one out.

Also, your code to blink an LED while in deep sleep is genius. I had never seen that done before. I learned a lot looking over your sketches.

tomocchi3411, 2022-08-29:
@moose4lord Thank you for comments, helittr and moose4lord.
I agree with helittr that an 8.5MHz clock source should be more stable than a 150kHz RC oscillator.
Therefore, as a result of various experiments, I found that the RTC with 8.5MHz clock is very stable if ULP is kept running during deep sleep.
In the previously uploaded test code (testSlowClockAutocalib.zip), the ULP was repeatedly running for 10msec LED lighting and 990msec stopping, but in this case, the 150kHz RC oscillator is more stable due to the effect of temperature correction. 
In this test code (testSlowClockAutocalib3.zip), the RTC is stabilized by keeping ULP running during deep sleep, and by adding further correction, the time error of the day can be reduced to a few seconds.

PC Time         RTC Time      T(us)     Freq(kHz) Rate     temp(C)   
20:01:05.468 -> 20:01:05.705, 29.877513, 33.469, 0.999890, 32.69
22:01:05.501 -> 22:01:05.731, 29.890258, 33.455, 0.999890, 29.86
00:01:05.432 -> 00:01:05.756, 29.872705, 33.475, 0.999890, 30.98
02:01:05.620 -> 02:01:05.786, 29.872533, 33.475, 0.999890, 30.77
04:01:05.680 -> 04:01:05.820, 29.852294, 33.498, 0.999890, 29.97
06:01:06.071 -> 06:01:05.841, 29.855932, 33.494, 0.999890, 30.11
08:01:06.322 -> 08:01:05.863, 29.872436, 33.475, 0.999890, 29.13
10:01:06.709 -> 10:01:05.886, 29.865428, 33.483, 0.999890, 30.77
12:01:06.635 -> 12:01:05.910, 29.858642, 33.491, 0.999890, 30.25
14:01:06.501 -> 14:01:05.936, 29.847534, 33.503, 0.999890, 31.44
16:01:06.433 -> 16:01:05.959, 29.852172, 33.498, 0.999890, 33.08
18:01:06.662 -> 18:01:05.978, 29.844776, 33.506, 0.999890, 31.05
20:01:06.696 -> 20:01:05.999, 29.842601, 33.509, 0.999890, 31.44

The above is the actual measurement, Freq(kHz) is the RTC clock frequency of 8.5MHz/256, and temp(C) is the temperature(Celsius) of the ESP32 measured with a thermistor .

Looking at this, the clock frequency doesn't seem to be affected by temperature.
I found that the 8.5MHz clock can be used for highly accurate time measurement, so I used to attach an external RTC, but now I don't need it.
However, the current consumption will increase by a few microamperes.
(https://lastminuteengineers.com/esp32-sleep-modes-power-consumption/)
I'm happy to help moose4lord's project.
[testSlowClockAutocalib3.zip](https://github.com/espressif/esp-idf/files/9441338/testSlowClockAutocalib3.zip)

moose4lord, 2022-09-08:
Thanks for the update @tomocchi3411. I was initially pretty happy with your calibration code using the internal RC oscillator, but the longer I ran it, the more variation I saw in the deep sleep timing. It was much better than my old uncalibrated setup, but over the course of a week or two, I would see the deep sleep times start to drift. I suspected the drift was due to temperature variation throughout the week, so I ran some experiments putting the ESP32 in my refrigerator and I did see quite a bit of temperature instability, especially when the ESP32 is undergoing temperature transitions.

I live in Vermont and my project is outside, so the temperature varies considerable throughout the year. I was not looking forward to recalibrating every season. :) So when I saw your new code using the 8.5MHz/256 oscillator, it was good news. I've done a little testing and I agree, that oscillator seems much more stable. And it only adds about 20uA to the deep sleep current draw, which is OK for me. I'll integrate your new code into my project and let you know how it goes.

moose4lord, 2022-09-28:
Hello @tomocchi3411, sorry it's taken so long to reply. I got distracted by other projects, but now I'm back to investigating the ESP32 sleep timing issue. Using your latest calibration code and 8.5MHz/256 clock source, I was able to get sleep timing results similar to yours for a two hour sleep time:
```
2 hours sleep time
19:38:04.045 -> 19:38:03.996, 29.018238, 34.461, 1.000080, 151.59
21:38:07.247 -> 21:38:03.842, 29.033763, 34.442, 1.000080, 175.04
23:38:05.875 -> 23:38:03.688, 29.034399, 34.441, 1.000080, 135.24
01:38:03.494 -> 01:38:03.533, 29.021753, 34.456, 1.000080, 136.95
03:38:04.433 -> 03:38:03.379, 29.025634, 34.452, 1.000080, 135.06
05:38:03.809 -> 05:38:03.225, 29.012964, 34.467, 1.000080, 132.71
07:38:05.718 -> 07:38:03.071, 29.017333, 34.462, 1.000080, 135.24
```

Unfortunately, if I extend the sleep time to four hours, I start to see some timing instability:
```
4 hours sleep time
15:01:45.944 -> 15:01:45.894, 29.025318, 34.452, 1.000080, 157.80
19:01:46.013 -> 19:01:45.746, 29.003173, 34.478, 1.000080, 175.04
23:02:12.771 -> 23:01:45.598, 29.022974, 34.455, 1.000080, 137.73 // +27 seconds??
03:02:10.144 -> 03:01:45.449, 29.024414, 34.453, 1.000080, 137.73 // -2 seconds
07:02:16.333 -> 07:01:45.301, 29.013330, 34.466, 1.000080, 135.06 // +6 seconds
11:02:04.125 -> 11:01:45.153, 29.029491, 34.447, 1.000080, 138.12 // -12 seconds
15:02:12.755 -> 15:01:45.005, 29.047435, 34.426, 1.000080, 138.92 // +8 seconds
```
I modified your testSlowClockAutocalib3 sketch (see attached) to ramp the sleep time from 3 minutes to 8 hours, periodically resyncing with the SMTP server, hoping it would converge on a very accurate value for RTC_CLOCK_RATE. Unfortunately, it didn't converge on a value:
```
15:37:40.884 -> *** With rtcClkRate=1.000080, RTC gained    28 msec over 0.06 hours, rtcClkRate changed : 0.999980
15:45:11.370 -> *** With rtcClkRate=0.999980, RTC lost     119 msec over 0.13 hours, rtcClkRate changed : 1.000191
16:00:15.281 -> *** With rtcClkRate=1.000191, RTC gained   467 msec over 0.25 hours, rtcClkRate changed : 0.999776
16:30:30.089 -> *** With rtcClkRate=0.999776, RTC gained   465 msec over 0.50 hours, rtcClkRate changed : 0.999570
17:30:37.085 -> *** With rtcClkRate=0.999570, RTC lost    3017 msec over 1.00 hours, rtcClkRate changed : 1.000239
19:30:39.743 -> *** With rtcClkRate=1.000239, RTC gained  1298 msec over 2.00 hours, rtcClkRate changed : 1.000095
23:30:39.001 -> *** With rtcClkRate=1.000095, RTC lost    4644 msec over 4.00 hours, rtcClkRate changed : 1.000353
07:30:32.684 -> *** With rtcClkRate=1.000353, RTC gained  6253 msec over 8.00 hours, rtcClkRate changed : 1.000179
15:30:49.539 -> *** With rtcClkRate=1.000179, RTC lost    3301 msec over 8.00 hours, rtcClkRate changed : 1.000271
23:30:28.998 -> *** With rtcClkRate=1.000271, RTC gained  9494 msec over 8.00 hours, rtcClkRate changed : 1.000007
07:30:13.494 -> *** With rtcClkRate=1.000007, RTC gained 18979 msec over 8.00 hours, rtcClkRate changed : 0.999479
```
I don't know, it seems the longer the ESP is in deep sleep mode, the more unstable the timing gets. The 8.5MHz/256 clock does help a lot, so I appreciate your help with that. My app sleeps for about 22 hours every day and tries to wake up at a specific time. I'm content that it usually wakes up within a couple minutes of the correct time, but it bugs me that I can't get it any better than that.

You mentioned you had some experience with external crystal oscillators. Does an external crystal provide better stability than the 8.5MHz/256 oscillator?

[autocal4.zip](https://github.com/espressif/esp-idf/files/9664903/autocal4.zip)

KaeLL, 2022-09-28:
@igrr Any thoughts/tips on this?

tomocchi3411, 2022-10-01:
Thanks for reporting an interesting experiment, @moose4lord

In your code autocal4 and in my testSlowClockAutocalib3,
 (1) calling getRtcClkPeriod() in setup() after wakeup
 (2) calling setRtcClkPeriod() in goPowerMode2_Sleep() before sleep
are repeated.
They play an important role in calibrating the RTC timer.
 (1) measuring the RTC clock cycle*1 compared to the CPU clock
 (2) writing the clock cycle calibration value*2 to the RTC_CNTL_STORE1_REG register
   *1 rtcClkPeriod
   *2 rtcClkPeriodCalib = rtcClkPeriod * rtcClkRate

The calibration value rtcClkPeriodCalib is the product of two parameters:
 rtcClkPeriod: in order to correct changes in the RTC clock comparing to the CPU clock
 rtcClkRate  : in order to compensate crystal CPU clock error comparing to SNTP

I don't know if the source of the RTC clock (8.5MHz/256) is a crystal or an oscillator, but the RTC frequency changes depending on the environment.
For this reason, rtcClkPeriod must be measured and calibrated frequently to follow the fluctuations of the RTC clock.
If the ESP32 sleeps for 8 hours, I expect that the RTC clock fluctuates during that time and causes a large error in the timer.
I think it's better that you shoud calibrate the RTC clock at intervals about ten minutes without SNTP in order to keep the daily error to a few seconds .
On the other hand, rtcClkRate only corrects the individual differences of crystals on each ESP32, so I think that you can fix the value after repeating comparative calibration with SNTP several times.

By the way, regarding the external crystal, what I used before was not a crystal, but an external RTC called DS3234s. Although it is highly accurate and can be battery backuped, it is not recommended because it is more expensive than ESP32.
On page 12 of the datasheet*3, I think the RTC clock will be stable if you use a 32.786kHz crystal.
   *3 https://www.espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_en.pdf
There are other reasonable external RTCs that can be used with I2C, but both the crystal and the external RTC will consume the ESP32 IO terminals.

I wish moose4lord the best of luck with the project and will help if I can.

KonstantinKondrashov, 2022-11-16:
Hi!
The initial issue here is that  `8md256 as rtc slow clk does not work properly during deep sleep`.  It's related to  https://github.com/espressif/esp-idf/issues/9455.

The fix below that we have right now (not merged yet) helps to fix it but it rises consumption in the deep sleep from 4-5uA to 13uA.
Fix for esp32 when RTC_CLK_SRC_INT_8MD256=y is used, time becomes more accurate.

```diff
diff --git a/components/esp_hw_support/port/esp32/rtc_sleep.c b/components/esp_hw_support/port/esp32/rtc_sleep.c
index de16b48d38bbbf1bef1cb06d204a76c1472594b2..5321a1e888eec5f84b6a4a80e526b4f576035ece 100644
--- a/components/esp_hw_support/port/esp32/rtc_sleep.c
+++ b/components/esp_hw_support/port/esp32/rtc_sleep.c
@@ -108,11 +108,13 @@ void rtc_sleep_get_default_config(uint32_t sleep_flags, rtc_sleep_config_t *out_
         out_config->dig_dbias_slp = RTC_CNTL_DBIAS_0V90;
         out_config->rtc_dbias_wak = RTC_CNTL_DBIAS_1V10;
         out_config->rtc_dbias_slp = RTC_CNTL_DBIAS_0V90;
+        out_config->dbg_atten     = !(sleep_flags & RTC_SLEEP_PD_INT_8M) ? RTC_CNTL_DBG_ATTEN_NODROP : RTC_CNTL_DBG_ATTEN_DEFAULT;
     } else {
         out_config->dig_dbias_wak = RTC_CNTL_DBIAS_1V10;
         out_config->dig_dbias_slp = !((sleep_flags) & RTC_SLEEP_PD_INT_8M) ? RTC_CNTL_DBIAS_1V10 : RTC_CNTL_DBIAS_0V90;
         out_config->rtc_dbias_wak = RTC_CNTL_DBIAS_1V10;
         out_config->rtc_dbias_slp = !((sleep_flags) & RTC_SLEEP_PD_INT_8M) ? RTC_CNTL_DBIAS_1V10 : RTC_CNTL_DBIAS_0V90;
+        out_config->dbg_atten     = RTC_CNTL_DBG_ATTEN_NODROP;
     }
 }
 
@@ -207,7 +209,6 @@ void rtc_sleep_init(rtc_sleep_config_t cfg)
         CLEAR_PERI_REG_MASK(RTC_CNTL_OPTIONS0_REG, RTC_CNTL_BB_I2C_FORCE_PU);
     } else {
         CLEAR_PERI_REG_MASK(RTC_CNTL_DIG_PWC_REG, RTC_CNTL_DG_WRAP_PD_EN);
-        REG_SET_FIELD(RTC_CNTL_BIAS_CONF_REG, RTC_CNTL_DBG_ATTEN, 0);
     }
 
     REG_SET_FIELD(RTC_CNTL_OPTIONS0_REG, RTC_CNTL_XTL_FORCE_PU, cfg.xtal_fpu);
@@ -226,6 +227,7 @@ void rtc_sleep_init(rtc_sleep_config_t cfg)
     REG_SET_FIELD(RTC_CNTL_REG, RTC_CNTL_DBIAS_WAK, cfg.rtc_dbias_wak);
     REG_SET_FIELD(RTC_CNTL_REG, RTC_CNTL_DIG_DBIAS_WAK, cfg.dig_dbias_wak);
     REG_SET_FIELD(RTC_CNTL_REG, RTC_CNTL_DIG_DBIAS_SLP, cfg.dig_dbias_slp);
+    REG_SET_FIELD(RTC_CNTL_BIAS_CONF_REG, RTC_CNTL_DBG_ATTEN, cfg.dbg_atten);
 
     REG_SET_FIELD(RTC_CNTL_SLP_REJECT_CONF_REG, RTC_CNTL_DEEP_SLP_REJECT_EN, cfg.deep_slp_reject);
     REG_SET_FIELD(RTC_CNTL_SLP_REJECT_CONF_REG, RTC_CNTL_LIGHT_SLP_REJECT_EN, cfg.light_slp_reject);
diff --git a/components/soc/esp32/include/soc/rtc.h b/components/soc/esp32/include/soc/rtc.h
index aa2b6f7fc99a30ab99dc7ef94c27c424204b7f59..338c464ff15eda50b9e419386312785b1ca7f650 100644
--- a/components/soc/esp32/include/soc/rtc.h
+++ b/components/soc/esp32/include/soc/rtc.h
@@ -502,6 +502,7 @@ typedef struct rtc_sleep_config_s {
     uint32_t xtal_fpu : 1;              //!< keep main XTAL powered up in sleep
     uint32_t deep_slp_reject : 1;       //!< enable deep sleep reject
     uint32_t light_slp_reject : 1;      //!< enable light sleep reject
+    uint32_t dbg_atten : 2;             //!< voltage parameter
 } rtc_sleep_config_t;
 
 #define RTC_SLEEP_PD_DIG                BIT(0)  //!< Deep sleep (power down digital domain)
diff --git a/components/soc/esp32/include/soc/rtc_cntl_reg.h b/components/soc/esp32/include/soc/rtc_cntl_reg.h
index aec51adf6a4bea61b8b6dc14305631391a06a346..edb7553f20120b88ef58ea78945ea5a7ce4ea6ae 100644
--- a/components/soc/esp32/include/soc/rtc_cntl_reg.h
+++ b/components/soc/esp32/include/soc/rtc_cntl_reg.h
@@ -1067,6 +1067,8 @@
 #define RTC_CNTL_DBG_ATTEN_V  0x3
 #define RTC_CNTL_DBG_ATTEN_S  24
 #define RTC_CNTL_DBG_ATTEN_DEFAULT  3
+#define RTC_CNTL_DBG_ATTEN_NODROP   0
+
 #define RTC_CNTL_REG          (DR_REG_RTCCNTL_BASE + 0x7c)
 /* RTC_CNTL_FORCE_PU : R/W ;bitpos:[31] ;default: 1'd1 ; */
 /*description: RTC_REG force power up*/
```
cc @lennyJL 

Note. This is not the final version of this fix.

AxelLin, 2022-12-08:
@KonstantinKondrashov Any update? (It's in reviewing state for 2 weeks).

AxelLin, 2023-02-06:
> Hi! The initial issue here is that `8md256 as rtc slow clk does not work properly during deep sleep`. It's related to #9455.
>
> The fix below that we have right now (not merged yet) helps to fix it [...]

Is this power consumption implact only  when RTC_CLK_SRC_INT_8MD256=y ?

AxelLin, 2023-02-06:
@KonstantinKondrashov I cannot find this fix in any stable branches.

lennyJL, 2023-02-07:
> Is this power consumption implact only when RTC_CLK_SRC_INT_8MD256=y ?

@AxelLin For this fix, Yes.

lennyJL, 2023-02-07:
> @KonstantinKondrashov I cannot find this fix in any stable branches.

This fix has been merged into the master branch, but not yet backported. Still need to wait.

Alvin1Zhang, 2023-04-24:
Thanks for reporting, feel free to reopen.

closed 2023-04-24 by Alvin1Zhang

rini711, 2023-09-12:
I am trying to modify below lines in esp32/build-GENERIC_C3/sdkconfig file, but after rebuilding MicroPython below changes are becoming default, Is there any way to modify below line ?
#CONFIG_ESP32C3_RTC_CLK_SRC_INT_RC is not set
# CONFIG_ESP32C3_RTC_CLK_SRC_EXT_CRYS is not set
# CONFIG_ESP32C3_RTC_CLK_SRC_EXT_OSC is not set
# CONFIG_ESP32C3_RTC_CLK_SRC_INT_8MD256=y
CONFIG_ESP32C3_RTC_CLK_CAL_CYCLES=3000