espressif/esp-idf #6688: Start scan error while a connection is being established.

alutov, 2021-03-10:
Maybe someone will come in handy. Recently I came across one feature of the esp32 BLE library. If I try to start a scan (esp_ble_gap_start_scanning (...)) while a connection is being established with some ble device (already called esp_ble_gattc_open (...) but even before ESP_GATTC_OPEN_EVT), the command is executed as expected with the error "scan start failed ". But after that it will be impossible to start or stop the scan (I suspect the connection command does not work either) until the esp32 is restarted. That is, the existing connections are still working, but it is impossible to establish new ones. In reality, this situation occurs when two connections are lost with a small time difference. The scan has already started and stopped, the first device has been found, a connection command has been given (here I note that the default connection timeout is 30 seconds). And at this time ESP_GATTC_DISCONNECT_EVT comes from the second device, which starts the scan again. In order not to get into this situation, you need to check two conditions in the procedure for starting a scan of devices and not start a scan if any of them is executed:
1 - whether a scan has already been started or whether there has already been a command to start a scan.
2- is there a connection with any device at the moment.
In the [example](https://www.esp32.com/download/file.php?id=1066&sid=3f087fc6ba04e5c710e5c8292c60f023) I found, the second check is not performed.
After the correction, my program works much more reliably.

xulongzhuang, 2021-03-19:
Hi，alutov，According to the problem you described above, my recurrence is as follows: Use [gattc_multi_connect ](https://github.com/espressif/esp-idf/tree/release/v4.3/examples/bluetooth/bluedroid/ble/gattc_multi_connect)to connect gatts_a and gatts_b, and make some small changes to gattc_multi_connect: After calling esp_ble_gattc_open() to connect to gatts_a, immediately call esp_ble_gap_start_scanning() to start scanning, and also encountered "Scan start failed" error, and no longer scan gatts_b, is my above test the problem you described?

alutov, 2021-03-19:
Yes, that's what I wrote about. But after that I could not run any scan (scan gatts_a, scan gatts_b, etc) at all before restarting esp. All the time the "scan start error" with code 1.  I don't know if it's worth editing the library, you can just do an extra check in your code. But you just need to know about it.

xulongzhuang, 2021-03-19:
When gatts_a is actively disconnected from it, ESP_GATTC_DISCONNECT_EVT will be triggered at this time. If the start scan is performed at this time, I think it can scan gatts_a or gatts_b again without waiting for esp to restart, right? I am using IDF releasev/4.3 gattc_multi_connect example like this

alutov, 2021-03-19:
If the no any active esp_ble_gattc_open (...)  tasks at that moment, then yes, the scan will start. If at this moment there was an establishment of a connection with gatts_b the scan will not start. It's normal because ble library is busy. The connection with gatts_b most likely will not happen (I don’t remember exactly here), and after that  ESP_GATTC_OPEN_EVT returns an error, the moment of the start of the scan comes again. The problem is that the scan will not start here with error 1.

xulongzhuang, 2021-03-22:
Hi, alutov, I think the mechanism of this library is set up like this: when esp_ble_gattc_open (...) triggers to return the result of ESP_GATTC_OPEN_EVT, scanning or other actions can be performed. It does not support concurrent operations. Of course, adding some detection at the application layer Better, it is better to avoid this concurrency in actual use. For example, in [gattc_multi_connect](https://github.com/espressif/esp-idf/blob/release/v4.3/examples/bluetooth/bluedroid/ble/gattc_multi_connect/main/gattc_multi_connect.c), the scan is placed after the ESP_GATTC_WRITE_CHAR_EVT event

alutov, 2021-03-22:
Totally agree with you but the fact of the matter is that in the given example there is a call to the scan procedure in the ESP_GATTC_DISCONNECT_EVT. But for some reason, only for the gatts_a device, the scan is not called in the gatts_b and gatts_c. I came across a description somewhere that explained the principle of multiconnect. First, device A, then B, and only then C. In [my code](https://github.com/alutov/ESP32-R4sGate-for-Redmond) the devices must be connected in any order and independently of each other, and therefore I added the scan call to all ESP_GATTC_DISCONNECT_EVT procedures. Perhaps it is possible to call a scan only in a ESP_GATTC_WRITE_CHAR_EVT, but in this case this situation is also possible. I also agree that you can just check the additional condition and the problem is solved. You just need to pay attention to this, especially since the problem can appear after hours and even days of work and the loss of all connections without restart or any other indication besides the scan start error. I read at least 2 issues where there are similar problems.

xulongzhuang, 2021-03-22:
In addition to checking before use, another method can also be to ensure that the two processes are not disturbed, such as starting a user task and using semaphores to ensure that each process is fully executed. Thank you for your feedback. This will help many people.

closed 2021-08-04 by Alvin1Zhang