espressif/esp-idf #6689: Cannot change port of led to GPIO 8 in ledc example (URGENT for me)

TobiasBoeschMiele, 2021-03-10:
----------------------------- Delete below -----------------------------

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [HUZZAH32]
- Kit version (HUZZAH32): []
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
     v4.4-dev-4-g73db14240
- Build System:         [idf.py (eclipse plugin)]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [eclipse plugin].
- Using an IDE?: [eclipse plugin]
- Power Supply:         [USB]

## Problem Description

Using the ledc example works. Changing GPIO 5 to GPIO8:
https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/examples/peripherals/ledc/main/ledc_example_main.c#L54
 makes the example run into a user exception:
https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/components/freertos/port/xtensa/xtensa_vectors.S#L513
 when attempting to set the fade function: 
https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/examples/peripherals/ledc/main/ledc_example_main.c#L157

There is really nothing that comes out of the UART or JTAG that I can use for determine what is going on and that is the main problem. I have no idea why the user exception is activated.

### Expected Behavior

LED on GPIO 8 can be used the same way as LED on GPIO 5.

### Actual Behavior

Changing GPIO 5 to GPIO8:
https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/examples/peripherals/ledc/main/ledc_example_main.c#L54
 makes the example run into a user exception:
https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/components/freertos/port/xtensa/xtensa_vectors.S#L513
 when attempting to set the fade function: 
https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/examples/peripherals/ledc/main/ledc_example_main.c#L157

### Steps to reproduce

1. Take the ledc example
2. Test it
3. Changing GPIO 5 to GPIO8:
https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/examples/peripherals/ledc/main/ledc_example_main.c#L54
4. Check if the  the example run into a user exception:
https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/components/freertos/port/xtensa/xtensa_vectors.S#L513
 when attempting to set the fade function: 
https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/examples/peripherals/ledc/main/ledc_example_main.c#L157

// If possible, attach a picture of your setup/wiring here.

No special wiring - the board doesn't has an led connected. I use a custom PCB and I cannot share that so I reduced the problem to an example on a publicly available hardware.

### Code to reproduce this issue

[ledc example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/ledc)
where GPIO 5 is changed to GPIO8:
https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/examples/peripherals/ledc/main/ledc_example_main.c#L54

## Debug Logs

There is really nothing that comes out of the UART or JTAG that I can use for determine what is going on and that is the main problem. I have no idea why the user exception is activated.

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

chegewara, 2021-03-10:
This board is a standard wroom board, which means pin 8 (GPIO8) is used by flash and you cant use it in app.

TobiasBoeschMiele, 2021-03-10:
@chegewara , thanks for your quick help.

> This board is a standard wroom board, which means pin 8 (GPIO8) is used by flash and you cant use it in app.

That'll explain everything.
Can you please tell me where that is statet in the docs?

suda-morris, 2021-03-10:
[image]

TobiasBoeschMiele, 2021-03-10:
@suda-morris is this an official doc? Is there a reference or a link to that table?

TobiasBoeschMiele, 2021-03-10:
I think the official answer to this is simply that the WROOM module does not offer GPIO8 as a usable GPIO. That can be seen in https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32_datasheet_en.pdf under "2.2 Pin Description"

TobiasBoeschMiele, 2021-03-10:
Thanks for your quick help so far.

closed 2021-03-10 by TobiasBoeschMiele

TobiasBoeschMiele, 2021-03-10:
Chapter "3.2 External Flash and SRAM" of the WROOM32 datasheet states it more clearly:

> ESP32-WROOM-32SE integrates a 4 MB of external SPI flash. The integrated SPI flash is connected to GPIO6,
GPIO7, GPIO8, GPIO9, GPIO10 and GPIO11. These six pins cannot be used as regular GPIOs.