espressif/esp-idf #6694: ESP32 doesn't wake-up from deep sleep

RafigRzayev, 2021-03-11:
We have battery powered ESP32. 

In the program start, a custom button is configured as an external wake-up source (through RTC GPIO). Afterwards, once application is done, device goes to deep sleep. Normally, I am able to wake-up the device with the button afterwards. 

I only encounter problem in following scenario:

1) leave the device active for long time, battery discharges. 
2) charge the battery so that device can get enough power.
3) press the custom button to wake the system - **nothing happens at all**. 
4) reset the board - the system works normally again. Button for waking from sleep works.

So, the issue is that after some time device forgets that it can be awaken with the RTC GPIO. Only way to restore it is with manual reset.

Here https://github.com/espressif/arduino-esp32/issues/796 it was mentioned that it could be a power supply issue. I added 1000uF capacitor as suggested, but same thing happened again. 

Next, what I did is I left the device connected to PC with UART overnight, so that I can see the logs in the morning. In the morning, I caught the same scenario. Battery was discharged. I charged it. Pressed the button, and device didn't wake up. In the logs I saw that there were no errors/brownouts prior to execution of `esp_sleep_enable_ext0_wakeup` and device went to sleep normally. I have several suspicions: 

1) In the code, I made it so that when battery is low, e.g. 20%, device should automatically go to sleep to stay responsive and avoid brownout errors. But batteries that I use are old, so perhaps it is possible that battery gets completely depleted at that point. **So, my first question: when power supply becomes less than minimum required for ESP, is it true that sleep configuration gets erased?**  If yes, probably I need to raise the threshold for automatic deep sleep. 

2) Second thing that can be the reason is following: In my application, there are two separate tasks. In the logs, I saw that one of them triggered **restart**, while other triggered **deep sleep** simultaneously. From the logs it is visible that device didn't wake-up again afterwards, so possibly **deep sleep** got executed before the **restart**. **Could it be that **restart** was partially executed on one thread, cleared the wake-up config, and then deep-sleep got executed?** This would explain why device is in deep sleep but doesn't have configuration to wake-up from it.

RafigRzayev, 2021-03-11:
I did one more test. This time I used external power supply instead of battery. I waited until the device goes to sleep. Once in sleep, I decreased the power supply voltage to 0V. Then, increased it back to nominal voltage. Same thing happened. Button didn't work as wake-up resource. So, apparently wake-up configuration gets erased when power is lost (please correct me if I am wrong). 

But the interesting thing is following:

After increasing the power supply from 0V to nominal voltage, device didn't reset too. So, I noticed following behavior:
1) When power is lost during program execution, once power is back, device restarts automatically.
2) When power is lost during deep sleep, once power is back device doesn't restart which is big problem, because it needs manual reset. Also, when I looked into power consumption during this state (restored power supply and not restarting automatically), I was wondering if device was again in deep sleep. The current consumption was higher than in deep sleep, actually it coincided with bootmode for my PCB. Normally, for my PCB, power consumption is 17mA during deep sleep, and 33mA prior to flashing new program. So, to summarize, during deep sleep, I cut the power, then restored it - measured current was 33mA(not deep sleep), plus device didn't restart. 

**What is this state?

Why doesn't the device restart when power is back?

What would you suggest as a solution for this problem?** 

I think this scenario is problematic for many battery power applications (as seen from https://github.com/espressif/arduino-esp32/issues/796). As I have understood, some people approached the problem by adding external hardware(watchdog) responsible for resetting the ESP32 so that it works without manual reset.

igrr, 2021-03-11:
@RafigRzayev Please consider this note from the ESP32 datasheet:

> When a battery is used as the power supply for the ESP32 series of chips and modules, a supply voltage
supervisor is recommended, so that a boot failure due to low voltage is avoided. Users are recommended
to pull CHIP_PU low if the power supply for ESP32 is below 2.3 V. For the reset circuit, please refer to
Figure ESP32­-WROOM­-32 Peripheral Schematics

It seems that in your design, the supply voltage goes below 2.3V at some point, and the chip is unable to restart when the voltage is re-applied.

RafigRzayev, 2021-03-11:
Thanks a lot @igrr 

You described the situation correctly, when voltage drops and voltage is then re-applied, device is unable to restart. There are some questions that I would like to ask, I would appreciate your assistance. 

1) When voltage drops during **program execution**, voltage re-apply **causes restart**. On the other hand, when voltage drops during deep sleep, voltage re-apply doesn't cause restart. But for both scenarios I have the same hardware setup. Is it expected behavior? Why? 

2) > boot failure due to low voltage is avoided

   What is implied here by boot failure? Inability of ESP32 for future power-on reset? 

3) In our hardware setup, CHIP_PU is pulled-up. When reset button is pressed, CHIP_PU gets grounded and reset occurs. From your suggestion, I understood that I need to monitor the supply voltage, once it drops below 2.3V, I have to disable pull-up on CHIP_PU, make it pull-down. Once the power will come back, the grounded CHIP_PU will reset the device once, and afterwards I have to restore back the pull-up configuration. Do I understand it correctly? 

4) I am using SD card in the project, that's why MTDI pin is pulled-up. I saw following note in ESP32­-WROOM­-32 Peripheral Schematics: 

> MTDI should be kept at a low electric level when powering up the module.

After seeing this note, I tried following: made MTDI grounded during sleep, repeated voltage drop and reapply, but device didn't restart again. Just wanted to share this information for any case. 

5) 

> For the reset circuit, please refer to Figure ESP32­-WROOM­-32 Peripheral Schematics

Is it the one highlighted below? 

[image]

igrr, 2021-03-11:
> On the other hand, when voltage drops during deep sleep, voltage re-apply doesn't cause restart. But for both scenarios I have the same hardware setup. Is it expected behavior?

I think this is exactly what the note in the TRM implies — slowly reducing the voltage below the minimum operating voltage while the device is in sleep mode will put the circuit into some state from which it cannot boot up. I don't know the cause of this.

> What is implied here by boot failure? Inability of ESP32 for future power-on reset?

Yes, unless you toggle reset (CHIP_PU) it will stay stuck.

On the point 3 — yes, i think this should work. However please keep in mind that this is a software (ESP-IDF) issue tracker. If you would like to get an authoritative answer, I recommend you request your design review on Espressif website: https://www.espressif.com/en/products/services. It is free of charge.

Regarding MTDI — if you are using an SD card, you can remove the requirement for the reset level of this pin by burning the respective efuse: https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#note-about-gpio12-esp32-only.

On point 5 — yes, i believe this is a reset circuit. The recommended values can be found in https://www.espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_en.pdf, "2.2.1 Power-on Sequence".

RafigRzayev, 2021-03-11:
Thank you very much for clarifications, it helped a lot. 

As for MTDI, I have already burned the efuse. I believe then there shouldn't be any problems originating from it.  

Since the problem seems to be completely hardware related, I am closing the issue. 

I will try to share results of above mentioned suggestions.

closed 2021-03-11 by RafigRzayev

RafigRzayev, 2023-04-13:
Hi,

I used RC delay circuit and also had to use a Reset IC (Low Voltage Detector). It will turn on the system only once sufficient voltage is accumulated. 

According to (p20) https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32_datasheet_en.pdf :

“To ensure the power supply to the ESP32 chip during power-up, it is advised to add an RC delay circuit at the EN pin. 

The recommended setting for the RC delay circuit is usually R = 10 kΩ and C = 1 µF. However, specific parameters 
should be adjusted based on the power-up timing of the module and the power-up and reset sequence timing 
of the chip. For ESP32’s power-up and reset sequence timing diagram, please refer to Section Power Scheme in “

Below is the schematics of my project:
[image]

simonccnomis, 2023-04-14:
Thank you for your help. This is a great solution.