espressif/esp-idf #6696: esp32-s2 cann't modify the adc sampling frequency

Moliam97, 2021-03-12:
请问一下esp32-s2是否支持ADC高速采样，当前adc的采样频率有点低，支撑不了当前的应用场景，目前我没有找到相应的api、在技术参考手册中，也并未找到相关资料。感谢！！
<i>I would like to ask whether the esp32-s2 supports ADC high-speed sampling. The current sampling frequency of adc is a bit low and cannot support the current application scenarios. At present, I have not found the corresponding api, nor have I found the relevant information in the technical reference manual.Thanks</i>

Moliam97, 2021-03-12:
当前使用的是默认的RTC控制器，需要使用DIG控制器才可。

Moliam97, 2021-03-15:
目前我需要高速率采样ADC1通道8，2M速率，但是目前并没有实现，我有找到ESP32中采集速率的转换，但是由于4.2版本和3.x版本目前有些互不兼容，无法移植成功。详见附件。可否提供ESP32-S2的高速ADC采样例程？
[i2s_adc_scale.tar.gz](https://github.com/espressif/esp-idf/files/6139852/i2s_adc_scale.tar.gz)

closed 2021-03-22 by Moliam97

xuweilai9, 2022-05-23:
> 请问一下esp32-s2是否支持ADC高速采样，当前adc的采样频率有点低，支撑不了当前的应用场景，目前我没有找到相应的api、在技术参考手册中，也并未找到相关资料。感谢！！

你好，我想问下你最终解决了这个ESP32-S2的高速ADC采样的问题了吗？