espressif/esp-idf #6697: PLL lock lost at high temperature if not using Wifi

teekaytk, 2021-03-12:
## Environment

- Development Kit:      none

- Module or chip used:  ESP32-WROOM-32E
- IDF version (run ``git describe --tags`` to find it):    v4.3-dev-472-gcf056a7d0
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:    Linux (docker)
- Using an IDE?: No
- Power Supply:         [Battery (12V->3.3V)]

## Problem Description
Our application is industrial where the device will see temperatures in excess of 60 degrees Centigrade. 
We are using ESP32 as a microcontroller only. WiFi and Bluetooth have been turned off and we don't intend to use them in the current design.

Several (but not all) of our devices under production are failing thermal testing. When they go above 50 degrees Centigrade,  it seems the PLL Locks is lost, all the serial baud rates and I2C clock rates go out of of range and device stops communicating/communicates garbage. A hard or soft reset seems to fix the PLL. We did a 50Khz PWM out on one GPIO and made a video of it going out of spec as temperature was raised which is attached here as proof.

When we enable WIFI, we do not see this issue appear. My understanding is that the WIFI driver calls some kind of PLL -recalibration on periodic basis which is not happening because we don't use WIFI.

### Expected Behavior

Device to continue to work as expected over all temperature ranges

### Actual Behavior

PLL Looses sync and clock drift massively when temperature goes above 55 degrees. 

### Steps to reproduce

Disable wifi and bluetooth in code. 
Create PWN at 50Khz on a GPIO. Measure frequency on frequency counter or oscilloscope and increase the temperature to 60 degrees C

// If possible, attach a picture of your setup/wiring here.

Video attached
https://user-images.githubusercontent.com/21081546/110929725-2cdff900-8320-11eb-9ccb-02bcda33c6bd.mp4

teekaytk, 2021-04-07:
Hi 
Do you have any update on this. We received this from support:

__________________________
Dear Sir,
Heard from our RD team, frequency changing with temperature is normal since the frequency of some crystals may deviate with the temperature.
Adding Wifi to the code could solve the problem because it can improve the stability of the crystal.

Best regards,
__________________________

It looks now that this is definitely a firmware issue. We cannot add WiFi or Bluetooth to our code as our device is not certified to sue WiFi and does not have RF approvals for WiFi. Also our application code is really big and adding WiFi or Bluetooth will eat into valuable RAM resources.

Please can you expedite this as it is causing problems for us.

Thanks

Spritetm, 2021-04-12:
Just so we can reproduce this exactly, can you attach the project you use to generate the 50KHz PWM signal?

teekaytk, 2021-04-12:
@Spritetm 
We have sent a board to [email ref:_00D7FyCNM._5007F14crOf:ref] to hardware team so that they can see for themselves.

As for the 50Khz I am using amended LEDC code in our product code which I cannot share but please PM me and I am happy to share the binary with you.

Spritetm, 2021-04-13:
Ah, gotcha, from what you said earlier it sounded like solving the issue got stuck somewhere. Seeing my colleagues are getting the physical hardware, I'll let them figure out what the issue is.

teekaytk, 2021-04-14:
unfortunately it is stuck a bit as we have had a lot of to-and-fro with hardware team but I have suspected it to be firmware related from day one but not getting any firmware support on this.

felixcollins, 2023-01-30:
I'm seeing a similar issue with WROOM32E, but temperature is not part of the issue, I'm operating at room temperature. I made a forum post here -> https://www.esp32.com/viewtopic.php?f=13&t=31849&p=109074#p109074

Did you find any solution to this or clues as to the underlying cause?

felixcollins, 2023-01-30:
I changed the LEDC config ledc_timer_config_t.clk_cfg = LEDC_USE_RTC8M_CLK,  And it fixed the problem.