espressif/esp-idf #6699: Consecutive apptrace statements not output to the JTAG console without stepping through

TobiasBoeschMiele, 2021-03-12:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    v4.4-dev-4-g73db14240
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [].
- Using an IDE?: [Yes eclipse-plugin]
- Power Supply:         [USB]

## Problem Description

From two consecutive apptrace instruction only the last one gets output to the JTAG console unless one steps through each instruction manually.
I also tried to [set the openocd command for tracing to blocking ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/app_trace.html#openocd-application-level-tracing-commands)without success.

### Expected Behavior

Running the program puts out all trace data each after another without halting the program.

### Actual Behavior

Running the program only puts out the trace data of the last apptrace instruction

### Steps to reproduce

1. Setup a Hello World example
2. Insert the code shown below in the hello_world_main.c
3. Setup a JTAG flashing debug configuration (see attachement)
4. Setup a JTAG debug configuration with openocd tracing (see attachement)
5. Enable application level tracing in sdkconfig
6. Build
7. Run by stepping through
8. Reset
9. Run without stepping
10. Compare console output of step 7 and 9

### Code to reproduce this issue

```cpp
/* Hello World Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "esp_app_trace.h"
#include "esp_log.h"

void app_main(void)
{
    char *ptr = NULL;

    ptr = (char *)esp_apptrace_buffer_get(ESP_APPTRACE_DEST_TRAX, 32, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
    if (ptr == NULL) {
        ESP_LOGE("SGG", "Failed to get buffer!");
    }
    sprintf(ptr, "__Hello World!\n");
    esp_err_t res = esp_apptrace_buffer_put(ESP_APPTRACE_DEST_TRAX, (uint8_t*) ptr, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
    res += esp_apptrace_flush(ESP_APPTRACE_DEST_TRAX, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
    if (res != ESP_OK) {
        /* in case of error host tracing tool (e.g. OpenOCD) will report incomplete user buffer */
        ESP_LOGE("SGG", "Failed to put buffer!");
    }

    ptr = (char *)esp_apptrace_buffer_get(ESP_APPTRACE_DEST_TRAX, 32, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
    if (ptr == NULL) {
        ESP_LOGE("SGG", "Failed to get buffer!");
    }
    sprintf(ptr, "__Hello Again!\n");
    res = esp_apptrace_buffer_put(ESP_APPTRACE_DEST_TRAX, (uint8_t*) ptr, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
    res += esp_apptrace_flush(ESP_APPTRACE_DEST_TRAX, ESP_APPTRACE_TMO_INFINITE/*tmo in us*/);
    if (res != ESP_OK) {
        /* in case of error host tracing tool (e.g. OpenOCD) will report incomplete user buffer */
        ESP_LOGE("SGG", "Failed to put buffer!");
    }
}
```

## Debug Logs
Stepping through:
```
[...]
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Warn : cpu0: No free slots to add HW breakpoint!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
__Hello World!
Info : Target halted. CPU0: PC=0x400D4024 (active)
Info : Target halted. CPU1: PC=0x400E2B1E 
Warn : cpu0: HW breakpoint not found!
[...]
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Warn : cpu0: No free slots to add HW breakpoint!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
__Hello Again!
Info : Target halted. CPU0: PC=0x400D407C (active)
Info : Target halted. CPU1: PC=0x400E2B1E 
[...]
```

Running:
```
Warn : cpu0: No free slots to add HW breakpoint!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Target halted. CPU0: PC=0x400D3FE0 (active)
Info : Target halted. CPU1: PC=0x400E2B1E 
Warn : cpu0: HW breakpoint not found!
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
__Hello Again!
```

## Other items if possible
[hello_world flash Configuration.launch.txt](https://github.com/espressif/esp-idf/files/6130161/hello_world.flash.Configuration.launch.txt)
[hello_world debug Configuration.launch.txt](https://github.com/espressif/esp-idf/files/6130162/hello_world.debug.Configuration.launch.txt)
[esp-huzzah-32.cfg.txt](https://github.com/espressif/esp-idf/files/6130168/esp-huzzah-32.cfg.txt) (if interesting - nothing special here I think)

Alvin1Zhang, 2021-03-15:
Thanks for reporting and sorry for the inconvenience, we will look into.

gerekon, 2021-03-15:
Hi @TobiasBoeschMiele!

1) What OpenOCD version do you use? The default one specified by IDF?
2) I could not find when you invoke `esp apptrace ...` command to start tracing. Search in debug configuration returned no results.

TobiasBoeschMiele, 2021-03-15:
I start it in the `hello_world debug Configuration.launch.txt` on line 19

gerekon, 2021-03-15:
> I start it in the hello_world debug Configuration.launch.txt on line 19:

Ahh, sorry I see. Missed this. I will check the issue.

gerekon, 2021-03-15:
@TobiasBoeschMiele I can reproduce your problem. 
For now as a temporary solution I'd recommend you to start tracing upon hitting breakpoint at `app_main`. To do this you need to remove `esp32 apptrace start con: 0` from OpenOCD command line and use the following GDB inti commands
```
mon reset halt
flushregs
set remote hardware-watchpoint-limit 2

b app_main
commands
mon esp32 apptrace start con: 0
c
end
```