espressif/esp-idf #6700: Custom MODBUS read/write handlers

allard-potma, 2021-03-12:
I would like to write my own modbus slave register callbacks. The stack and the Espressif port already allows this, see the slave interface in mbc_slave.h :

```
typedef struct
{
    mb_slave_options_t opts;                                 

    // Functional pointers to internal static functions of the implementation (public interface methods)
    iface_init init;                        /*!< Interface method init */
    iface_destroy destroy;                  /*!< Interface method destroy */
    iface_setup setup;                      /*!< Interface method setup */
    iface_start start;                      /*!< Interface method start */
    iface_check_event check_event;          /*!< Interface method check_event */
    iface_get_param_info get_param_info;    /*!< Interface method get_param_info */
    iface_set_descriptor set_descriptor;    /*!< Interface method set_descriptor */

    // Modbus register calback function pointers
    reg_discrete_cb slave_reg_cb_discrete;  /*!< Stack callback discrete rw method */
    reg_input_cb slave_reg_cb_input;        /*!< Stack callback input rw method */
    reg_holding_cb slave_reg_cb_holding;    /*!< Stack callback holding rw method */
    reg_coils_cb slave_reg_cb_coils;        /*!< Stack callback coils rw method */
} mb_slave_interface_t;
```

If i'm correct the **handler in `esp_err_t mbc_slave_init(mb_port_type_t port_type, void **handler)` points to a mb_slave_interface_t object, but i'm not able to cast the void pointer to this type because the type is defined in a private h file.
My suggestion is to move mb_slave_interface_t to a public h file, or provide a function to edit the register callback function pointers.

The second feature request is to add the option the change the core affinity of the modbus tasks when initializing the stack, inline with other ESP-IDF components that allow this.

Alvin1Zhang, 2021-03-15:
Thanks for raising this feature request.

alisitsyn, 2021-03-15:
@allard-potma,

Thank you for your proposals which sound useful. 
My point of view:
The `mb_slave_interface_t` type and callbacks were kept private intentionally. If the user defined callback functions and interface will be public it will be very hard to support this functionality in Modbus component. The user callbacks (usually are ) will be called in context of Modbus task and any errors or blockings in the callback function may cause critical errors and will prevent Modbus to function correctly due to deterministic aspects of the protocol.
I would propose you to override functionality of modbus controller object on your side in this case. Would you agree with this?
This will keep the interface hidden for public but will allow to override just required functionality for applications which allow mapping the external peripheral addresses for example. If yes, I can propose the solution for this. 

The feature to add the option to change the core affinity of the modbus tasks will be useful and will be implemented.

allard-potma, 2021-03-15:
@alisitsyn 
Sound reasonable, I am open to your proposal.

alisitsyn, 2021-03-16:
@allard-potma,

I prepare and check the example for you. I will share it as soon as possible (could be today).

alisitsyn, 2021-03-16:
@allard-potma ,

The example to override the callback function in Modbus stack is here: https://github.com/alisitsyn/modbus_support/tree/mb_support/mb_override_callbacks/mb_override_slave
The goal of this example is to demonstrate possible approach to customize the existing Modbus stack support. The example currently overrides just one function to read Modbus input registers.
[override the modbus controller object](https://github.com/alisitsyn/modbus_support/blob/mb_support/mb_override_callbacks/mb_override_slave/components/freemodbus/readme.md)
[example notes](https://github.com/alisitsyn/modbus_support/blob/mb_support/mb_override_callbacks/mb_override_slave/notes.md)

The folder components/freemodbus in this example contains the files required to override the modbus controller object represented in the freemodbus component folder of ESP-IDF. The special CMakeLists.txt is used to override the private files in the component folder. This technique can be used to allow definition of custom read/write callbacks for Modbus stack in your project and not limited only for this purpose.
Note: This functionality is not public and not officially supported.

Below is the patch to add affinity kconfig option for modbus tasks:
[0001_freemodbus_add_affinity.diff.log](https://github.com/espressif/esp-idf/files/6148467/0001_freemodbus_add_affinity.diff.log)

Please check the example project and let me know the results.
Thank you.

allard-potma, 2021-03-17:
@alisitsyn Thanks, I got it working. Do you want me to close this issue, or keep it open until the affinity patch is added to the master branch?

alisitsyn, 2021-03-17:
@allard-potma,

Good to know it helps. Please leave the issue opened. Once the affinity MR is merged the issue will be closed automatically. Please be patient because merging of these fixes usually delayed. Thanks.

allard-potma, 2021-03-18:
@alisitsyn The affinity patch works, but when I create a serial slave the "uart_queue_task" created in `esp-idf/components/freemodbus/port/portserial.c ` still runs with affinity -1. I will patch this in my local component, but you might want to add this to the affinity MR.

alisitsyn, 2021-03-18:
@allard-potma, Thank you for this reminder. I missed one commit and updated this patch again:

[0001_freemodbus_add_affinity.diff.log](https://github.com/espressif/esp-idf/files/6162787/0001_freemodbus_add_affinity.diff.log)

closed 2021-07-25 by espressif-bot

alisitsyn, 2024-07-11:
@thiagoprati , I think you are looking to this [mb_override_slave project](https://github.com/alisitsyn/modbus_support/blob/mb_support/mb_override_callbacks/mb_override_slave)