espressif/esp-idf #6701: ESP32  UART rx interrupt occur every 96 bytes.

thedayu, 2021-03-12:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
     v4.0
- Build System:         windows + vscode + esp32 extension

## Problem Description
if use uart_read_bytes() to read single byte, it blocks until more than 96 bytes recieved.
if use costom interrupt routine instead of default handler, uart rx interrupt occurs every 96 bytes.

//Detailed problem description goes here.

### Expected Behavior
uart_read_bytes() to read single byte, it should return at any data.
also uart rx interrupt should occur as soon as data arrived.

### Actual Behavior
see debug code below.

### Code to reproduce this issue

```cpp
//setup of uart2
  const uart_config_t uart_config = {
        .baud_rate = baudrate,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .rx_flow_ctrl_thresh = 0,
        .use_ref_tick = 0,
    };
    // We won't use a buffer for sending data.
    ERROR_CHECK(uart_driver_install(UART_NUM_2, 256 * 2, 0, 0, NULL, 0));
    ERROR_CHECK(uart_param_config(UART_NUM_2, &uart_config));
    int err = uart_set_pin(UART_NUM_2, txPin, rxPin, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
    uart_enable_rx_intr(UART_NUM_2);
    uart_isr_free(UART_NUM_2);
    uart_isr_register(UART_NUM_2, uart_intr_handle, 0, 0, 0);

....

/////ISR ROUTINE
uint8_t rxbuf[1024];
uint16_t rxlen = 0;
static void IRAM_ATTR uart_intr_handle(void *arg)
{
    uint16_t rx_fifo_len, status;
    uint16_t i = 0;

    status = UART2.int_st.val;             // read UART interrupt Status
    rx_fifo_len = UART2.status.rxfifo_cnt; // read number of bytes in UART buffer
    rxlen = rx_fifo_len;
    while (rx_fifo_len)
    {
        rxbuf[i++] = UART2.fifo.rw_byte; // read all bytes
        rx_fifo_len--;
    }

    int evt = 1;
    xQueueSendFromISR(que_handle, (void *)&evt, 0);
    uart_clear_intr_status(UART_NUM_2, UART_RXFIFO_FULL_INT_CLR|UART_RXFIFO_TOUT_INT_CLR);
}

....
/// READ BYTES, try to get data as soon as possible.
        char data[RX_BUF_SIZE];
        int rxBytes = uart_read_bytes(UART_NUM_1, data, 1, 1000000 / portTICK_RATE_MS);
        if (rxBytes == 1)rxBytes = 1+ uart_read_bytes(UART_NUM_1, data + 1, RX_BUF_SIZE, 0);

...

//LOGGING CODE 
...
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

//SENDING 1234567890 to uart and repeat.

```
I (1212456)  UART RECIVED 96bytes: 345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678
I (1213416) UART RECIVED 96bytes: 901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234
I (1214366) UART RECIVED 96bytes: 567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890
I (1215306)  UART RECIVED 96bytes: 123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456
I (1216266) UART RECIVED 96bytes: 789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012
I (1217216) UART RECIVED 96bytes: 345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2021-03-12:
> //SENDING 1234567890 to uart every second

How could you be receiving 96bytes per second if you are sending 10 bytes per second? What is your sending code?

thedayu, 2021-03-13:
my bad, sending 1234567890 to uart manually, multiple time.

And problem solved after explictly call uart_intr_config() with default thresholds.
ISSUE CLOSED.

> > //SENDING 1234567890 to uart every second
> 
> How could you be receiving 96bytes per second if you are sending 10 bytes per second? What is your sending code?
> 
> > also uart rx interrupt should occur as soon as data arrived
> 
> Rx interrupt occurs when:
> 
> 1. Rx idle timeout (end of frame)
> 2. Rx fifo full threshhold

closed 2021-03-13 by thedayu