espressif/esp-idf #6702: Incorrect Kconfig documentation for BT_CTRL_BLE_MAX_ACT

Emill, 2021-03-12:
## Environment

- Module or chip used:  ESP32-C3 / ESP32-S3
- IDF version: v4.4-dev-4-g73db14240

## Problem Description

In `components/bt/controller/esp32c3/Kconfig.in` and `components/bt/controller/esp32s3/Kconfig.in` we can find the following:

```
config BT_CTRL_BLE_MAX_ACT
    int "BLE Max Instances"
    default 10
    range 1 10
    help
        BLE maximum instances of bluetooth controller.
```

The documentation here is incorrect. This config option does not refer to "maximum instances of bluetooth controllers" at all (the chip only has one bluetooth controller), but to the maximum number of air activities.

So if you have it set to 10, it means you can have 1 ongoing scanning and 9 ongoing connections.

Alvin1Zhang, 2021-03-15:
Thanks for reporting, we will look into.

xiewenxiang, 2021-03-19:
This configuration option refers to the ble maximum activities of bluetooth controller，both of connections, scan , sync and adv(periodic adv, multi-adv).

closed 2021-03-22 by Alvin1Zhang