espressif/esp-idf #6704: idf.py tool: monitor -B / --monitor-baud crashes idf.py (in 4.2 stable release)

a2800276, 2021-03-12:
- Development Kit:      [none]

- IDF version (run ``git describe --tags`` to find it): v4.2
- Build System:         [idf.py]
- Operating System:     [Linux]

## Problem Description

passing the the -B / --monitor-baud parameter to the `mointor` subcommand causes the following stack-trace:

```
$ idf.py monitor --monitor-baud 115200
Executing action: monitor
Choosing default port b'/dev/ttyUSB0' (use '-p PORT' option to set a specific serial port)
Traceback (most recent call last):
  File "/...projects/esp32/esp-idf/tools/idf.py", line 788, in <module>           
    main()
  File "/...projects/esp32/esp-idf/tools/idf.py", line 706, in main           
    cli(sys.argv[1:], prog_name=PROG)
  File "/....espressif/python_env/idf4.2_py3.8_env/lib/python3.8/site-packages/click/core.py", line 829, in __call__           
    return self.main(*args, **kwargs)
  File "/....espressif/python_env/idf4.2_py3.8_env/lib/python3.8/site-packages/click/core.py", line 782, in main           
    rv = self.invoke(ctx)
  File "/....espressif/python_env/idf4.2_py3.8_env/lib/python3.8/site-packages/click/core.py", line 1290, in invoke           
    return _process_result(rv)
  File "/....espressif/python_env/idf4.2_py3.8_env/lib/python3.8/site-packages/click/core.py", line 1224, in _process_result           
    value = ctx.invoke(self.result_callback, value, **ctx.params)
  File "/....espressif/python_env/idf4.2_py3.8_env/lib/python3.8/site-packages/click/core.py", line 610, in invoke           
    return callback(*args, **kwargs)
  File "/...projects/esp32/esp-idf/tools/idf.py", line 622, in execute_tasks           
    task(ctx, global_args, task.action_args)
  File "/...projects/esp32/esp-idf/tools/idf.py", line 211, in __call__           
    self.callback(self.name, context, global_args, **action_args)
  File "/...projects/esp32/esp-idf/tools/idf_py_actions/serial_ext.py", line 114, in monitor           
    run_tool("idf_monitor", monitor_args, args.project_dir)
  File "/...projects/esp32/esp-idf/tools/idf_py_actions/tools.py", line 74, in run_tool           
    display_args = " ".join(quote_arg(arg) for arg in args)
  File "/...projects/esp32/esp-idf/tools/idf_py_actions/tools.py", line 74, in <genexpr>           
    display_args = " ".join(quote_arg(arg) for arg in args)
  File "/...projects/esp32/esp-idf/tools/idf_py_actions/tools.py", line 70, in quote_arg           
    if " " in arg and not (arg.startswith('"') or arg.startswith("'")):
```

### Expected Behavior
no stacktrace

### Actual Behavior
see above

### Steps to reproduce
$ idf.py monitor --monitor-baud 115200

a2800276, 2021-03-12:
setting via environment variable, i.e. `MONITORBAUD=115200 idf.py monitor` works and from what I can tell, the current master also works, though I'm having difficulties finding a difference that would explain the change in behavior...

martinespgano, 2021-03-16:
Hi, @a2800276, could you please provide your current commit of version 4.2? I was not able to reproduce the issue with the latest commit of v4.2.

a2800276, 2021-03-21:
My apologies, I checked out the v4.2 **tag** assuming this is the `stable` release. If I check out out the `release/v4.2` **branch** the --monitor-baud works as expected. I assumed a tagged release would remain stable, but the documentation clearly states that bug fixes are applied.

martinespgano, 2021-03-23:
Hi, @a2800276, thank you for the information. I am closing this issue for now. If you had any other issues don't hesitate to contact us.

closed 2021-03-23 by Alvin1Zhang