espressif/esp-idf #6706: Failed to compile example of rotary_encoder

zsjlovestudy, 2021-03-13:
Could you tell me the solution to this
[2/9] Building C object esp-idf/rotary_encoder/CMakeFiles/__idf_rotary_encoder.dir/src/rotary_encoder_pcnt_ec11.c.obj
FAILED: esp-idf/rotary_encoder/CMakeFiles/__idf_rotary_encoder.dir/src/rotary_encoder_pcnt_ec11.c.obj 

My current environment is version 4.2 esp-idf
But I couldn't find the example of rotary_encoder .

[image]
[image]
[image]

suda-morris, 2021-03-13:
Hi @zsjlovestudy 
That example and the new API was introduced in v4.3, you can switch to release/v4.3 or master branch.

zsjlovestudy, 2021-03-14:
Thank you for your help！！

zsjlovestudy, 2021-03-17:
Could you tell me how to do this.
[image]
[image]

brianignacio5, 2021-03-17:
It seems something is wrong in the extension.

Can you try to delete:

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION`

and install [VSIX installer](https://github.com/espressif/vscode-esp-idf-extension/suites/2259097242/artifacts/47026344)

zsjlovestudy, 2021-03-17:
I am using the VSIX you just sent, The version of the ESP IDF is up to date

[image]

It's the same mistake

[image]
[image]

In addition, I did not see the release version of 4.3release in these options. 

I chose to Find ESP-IDF in Your System,   and this directory already has the 4.3release file I just downloaded

brianignacio5, 2021-03-18:
About the version not being part of the list, v4.3 is still in beta.

I'm not sure why this happens. I've tested on Windows 10 and it is working fine, also the connection error is not there.

Maybe you can close this issue (since conversation changed to a vscode extension issue) and open a [new issue](https://github.com/espressif/vscode-esp-idf-extension/issues/new?assignees=&labels=bug-report&template=bug_report.md&title=%5BBug+Report%5D%3A+) with details about your setup (OS, Visual Studio Code information, nodeJS)

closed 2021-03-24 by Alvin1Zhang