espressif/esp-idf #6707: v3.3.1

zhy102, 2021-03-13:
i used by git.And now,i try to run again from cmd,its same erros as follows:
C:\msys32\esp-idf-v3.3>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@1.22.0-97-gc752ad5d-5.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Skipping mconf@v4.6.0.0-idf-20190628 (already installed)
Skipping ninja@1.9.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Setting up Python environment
fatal: No names found, cannot describe anything.
Traceback (most recent call last):
  File "C:\msys32\esp-idf-v3.3\tools\idf_tools.py", line 1358, in <module>
    main(sys.argv[1:])
  File "C:\msys32\esp-idf-v3.3\tools\idf_tools.py", line 1354, in main
    action_func(args)
  File "C:\msys32\esp-idf-v3.3\tools\idf_tools.py", line 1144, in action_install_python_env
    idf_python_env_path, _, virtualenv_python = get_python_env_path()
  File "C:\msys32\esp-idf-v3.3\tools\idf_tools.py", line 840, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '--work-tree=' + global_idf_path, 'describe'], cwd=global_idf_path, env=os.environ).decode()
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 512, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['git', '--work-tree=C:\\msys32\\esp-idf-v3.3', 'describe']' returned non-zero exit status 128.

dobairoland, 2021-03-13:
@zhy102 You don't have a `git` installed for Windows. The one installed in MSYS32 is not working from Windows command line.

You can use the Windows tool istaller to install everything you need:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#esp-idf-tools-installer

dobairoland, 2021-03-15:
I'm linking here my previous comment: https://github.com/espressif/esp-idf/issues/6705#issuecomment-797962639

zhy102, 2021-03-20:
I am sorry, but it still  when  i  run install.bat from Windows command line. Later versions of ESP-IDF support powershell as well.

 
I'm linking here my previous comment: #6705 (comment)

dobairoland, 2021-03-20:
Have you tried the Windows tool installer linked above?

dobairoland, 2021-03-25:
I'm closing this in favor of https://github.com/espressif/esp-idf/issues/6775

closed 2021-03-25 by dobairoland