espressif/esp-idf #6708: undefined reference to `app_main'    idf-v4.4

xenpac, 2021-03-13:
I just updated to latest esp-idf, did install.sh and source export.sh
(python2.7.2)
My project was originally done on idf-V3.3.1

idf.py build
......
getting some warnings about "depreciated..stuff", but no errors. However linker fails.
All sources are specified in CMakelist.txt. There is definitely a "void app_main(void)".

/home/tom/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../
	../xtensa-esp32-elf/bin/ld: esp-idf/freertos/libfreertos.a(port_common.c.obj):(.literal.main_task+0x1c): undefined reference to `app_main'
/home/tom/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../
	../xtensa-esp32-elf/bin/ld: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':
/home/tom/esp/esp-idf/components/freertos/port/port_common.c:122: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

...

i deleted the build directory and tryied "make"

make

...creates a new sdkconfig....not getting "depreciated warnings"...
success

hmmm...

chegewara, 2021-03-13:
Is your main file `.c` or `.cpp`? Did you add file to CMakeLists.txt?

xenpac, 2021-03-13:
its all c code so .c
https://github.com/xenpac/esp32-camera-driver

chegewara, 2021-03-13:
https://github.com/espressif/esp-idf/blob/master/examples/get-started/hello_world/main/CMakeLists.txt

xenpac, 2021-03-14:
thanks, works

closed 2021-03-14 by xenpac