espressif/esp-idf #6710: Ledc driver bug, after fast switching from the 0 duty and 100% duty with fade, the leds stays on

mvarchdev, 2021-03-13:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32U
- IDF version: v4.3-beta1
- Build System:         idf.py
- Compiler version: (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: Yes - Visual Studio code with installed esp-idf extension, but building and flashing from the server (Ubuntu wth esp-idf, same version)
- Power Supply:         USB

## Problem Description

I am working on a project, something like smart home, just school project... I created IFTTT (If this then this) mechanism where conditions can be added while running. There is one loop checking if something was changed (button high, etc...), I added ledc to make led bars easily controllable with it. gpio_set_level works fine, and ledc_set_fade_with_time with low click frequency too, but as soon as I start clicking faster, the ledc_set_fade_with_time bugs, and some leds will stay on. (The main function is just from duty 0% to 100%, then inverted after button click (100% to 0%) and so on). The debug output showing some weird things, like there would be actual duty 10000 and more. You will see in the videos:

First: [VIDEO1](https://cloud.sunray.sk/index.php/s/Ky9wRJoB9K5JYs5)
Second: [VIDEO2](https://cloud.sunray.sk/index.php/s/DnismH6nCxEtKFM)

Can you please help me?

Here is debug log with ledc:
**NORMAL**
D (00:00:30.093) ledc: cur duty: 0; target: 2047, step: 1023, cycle: 1; scale: 1; dir: 1

D (00:00:31.061) ledc: cur duty: 2047; target: 0, step: 1023, cycle: 1; scale: 1; dir: 0

D (00:00:31.062) ledc: cur duty: 0; target: 2047, step: 1023, cycle: 1; scale: 1; dir: 1

D (00:00:31.985) ledc: cur duty: 2047; target: 0, step: 1023, cycle: 1; scale: 1; dir: 0

D (00:00:31.986) ledc: cur duty: 0; target: 2047, step: 1023, cycle: 1; scale: 1; dir: 1

D (00:00:32.975) ledc: cur duty: 2047; target: 0, step: 1023, cycle: 1; scale: 1; dir: 0

D (00:00:32.976) ledc: cur duty: 0; target: 2047, step: 1023, cycle: 1; scale: 1; dir: 1

D (00:00:33.415) ledc: cur duty: 0; target: 2047, step: 1023, cycle: 1; scale: 1; dir: 1

**GLITCHED**
D (00:00:41.912) ledc: cur duty: 9587; target: 2047, step: 1023, cycle: 1; scale: 1; dir: 0

D (00:00:42.034) ledc: cur duty: 9098; target: 0, step: 1023, cycle: 1; scale: 2; dir: 0

D (00:00:42.035) ledc: cur duty: 291; target: 2047, step: 1023, cycle: 2; scale: 1; dir: 1

D (00:00:42.158) ledc: cur duty: 291; target: 0, step: 291, cycle: 13; scale: 1; dir: 0

D (00:00:42.159) ledc: cur duty: 193; target: 2047, step: 1023, cycle: 2; scale: 1; dir: 1

D (00:00:42.264) ledc: cur duty: 507; target: 2047, step: 1023, cycle: 2; scale: 1; dir: 1

D (00:00:42.265) ledc: cur duty: 193; target: 0, step: 193, cycle: 20; scale: 1; dir: 0

D (00:00:42.403) ledc: cur duty: 783; target: 0, step: 783, cycle: 5; scale: 1; dir: 0

D (00:00:42.404) ledc: cur duty: 10243; target: 2047, step: 1023, cycle: 1; scale: 2; dir: 0

D (00:00:42.509) ledc: cur duty: 9399; target: 0, step: 1023, cycle: 1; scale: 2; dir: 0

D (00:00:42.510) ledc: cur duty: 224; target: 2047, step: 1023, cycle: 2; scale: 1; dir: 1

D (00:00:42.633) ledc: cur duty: 224; target: 0, step: 224, cycle: 17; scale: 1; dir: 0

D (00:00:42.634) ledc: cur duty: 149; target: 2047, step: 1023, cycle: 2; scale: 1; dir: 1

D (00:00:42.754) ledc: cur duty: 502; target: 2047, step: 1023, cycle: 2; scale: 1; dir: 1

D (00:00:42.756) ledc: cur duty: 149; target: 0, step: 149, cycle: 26; scale: 1; dir: 0

V (00:00:42.892) esp_adapter: thread sem get: sem=0x3ffccb50
D (00:00:43.111) ledc: cur duty: 705; target: 0, step: 705, cycle: 5; scale: 1; dir: 0

D (00:00:43.112) ledc: cur duty: 8680; target: 2047, step: 1023, cycle: 1; scale: 1; dir: 0

D (00:00:43.234) ledc: cur duty: 8191; target: 0, step: 1023, cycle: 1; scale: 2; dir: 0

D (00:00:43.235) ledc: cur duty: 114; target: 2047, step: 1023, cycle: 2; scale: 1; dir: 1

Alvin1Zhang, 2021-03-15:
Thanks for the very detailed report and sorry for the inconvenience, we will look into.

mvarchdev, 2021-04-02:
> Thanks for the very detailed report and sorry for the inconvenience, we will look into.

Hey, What is your progress?

mvarchdev, 2021-04-02:
> > Thanks for the very detailed report and sorry for the inconvenience, we will look into.
> 
> Hey, What is your progress?

@Alvin1Zhang

mvarchdev, 2021-04-02:
> Thanks for the very detailed report and sorry for the inconvenience, we will look into.

Hey, what is your progress? @Alvin1Zhang Thanks in advance.

Alvin1Zhang, 2021-07-29:
@mvarchdev Thanks for reporting and sorry for the slow turnaround. Thanks for being patient, we had a fix under internal reviewing. The issue will be closed automatically once the fix is available on GitHub. Thanks.

mvarchdev, 2021-09-05:
Hello. How is the fix going on? Thanks in advance.

closed 2022-01-26 by espressif-bot