espressif/esp-idf #6711: mdns_service_add crash with ESP_ERR_INVALID_ARG

kotik-0, 2021-03-14:
Environment information:
I (46) boot: ESP-IDF v4.4-dev-4-g73db142403-dirty 2nd stage bootloader
I (46) boot: compile time 01:15:04
I (46) boot: chip revision: 0

Hardware: ESP32S2-WROOM

Description:
I'm trying to use the example restful_server:
esp-idf/examples/protocols/http_server/restful_server/

but the code generates an error once I add a mdns service:

```
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4002868c
0x4002868c: _esp_error_check_failed at ...esp/esp-idf/components/esp_common/src/esp_err.c:41

file: "../main/main.c" line 309
func: initialise_mdns
expression: mdns_service_add("WDResetter-WebServer", "_http", "_tcp", 80, NULL,0)

abort() was called at PC 0x4002868f on core 0
0x4002868f: _esp_error_check_failed at ..../esp/esp-idf/components/esp_common/src/esp_err.c:42

Backtrace:0x40029a03:0x3ffd1380 0x4002a291:0x3ffd13a0 0x4003116e:0x3ffd13c0 0x4002868f:0x3ffd1430 0x4008af98:0x3ffd1450 0x4008b8ab:0x3ffd1480 0x4002d1e1:0x3ffd14a0
0x40029a03: panic_abort at ..../esp/esp-idf/components/esp_system/panic.c:356

0x4002a291: esp_system_abort at .../esp/esp-idf/components/esp_system/system_api.c:112

0x4003116e: abort at ...esp/esp-idf/components/newlib/abort.c:46

0x4002868f: _esp_error_check_failed at .../esp/esp-idf/components/esp_common/src/esp_err.c:42

0x4008af98: initialise_mdns at ....esp/esp-idf/examples/get-started/wd_resetterv2/build/../main/main.c:309 (discriminator 1)

0x4008b8ab: wifi_task at ..../esp/esp-idf/examples/get-started/wd_resetterv2/build/../main/main.c:522 (discriminator 13)

0x4002d1e1: vPortTaskWrapper at /.../esp/esp-idf/components/freertos/port/xtensa/port.c:168
```

Code:
```
 //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    
    
    ESP_LOGI(TAGW,"Init HTTP test");
    ESP_ERROR_CHECK(esp_netif_init());
    //ESP_ERROR_CHECK(esp_event_loop_create_default());

    initialise_mdns();
    ESP_LOGI(TAGW,"Init netbiosns_init");
    netbiosns_init();
    netbiosns_set_name(CONFIG_EXAMPLE_MDNS_HOST_NAME);
```

I'm not able to find any information about how deal with it. I'm just using the example. The Wi-Fi is not yet initialised.

Thanks.

Alvin1Zhang, 2021-03-15:
Thanks for reporting, we will look into.

david-cermak, 2021-03-15:
Hi @kotik-0 

Are you sure that you're not adding the service multiple times? One of the reason for receiving the `ESP_ERR_INVALID_ARG` error could be that a service of the same name and protocol is already up and running. Looking at your code and the logs, this seems like a possibility, as you're adding the service with *some* TXT data:
```
mdns_service_add("ESP32-WebServer", "_http", "_tcp", 80, serviceTxtData,sizeof(serviceTxtData) / sizeof(serviceTxtData[0])
```
but it crashes while adding a service with *no* TXT data:
```
expression: mdns_service_add("WDResetter-WebServer", "_http", "_tcp", 80, NULL,0)
```

kotik-0, 2021-03-15:
Hi @david-cermak ,
Thanks for your answer and your feedback.
It seems that the problem was that I didn't start:
`ESP_ERROR_CHECK(esp_event_loop_create_default());` before call:
`initialise_mdns();`

Problem fixed.

closed 2021-03-15 by kotik-0