espressif/esp-idf #6713: Sending UDP packet lost when immediately call set_channel afterward.

rtmtree, 2021-03-14:
I am trying to create a pair of AP and STA that can change their channel (from 1-13) continuously together while always keeping connection.
If you have a suggestion on this ,please share.

My idea is 
 - AP tell STA via UDP socket when the channel need to change.
 - STA receive UDP socket from AP then change the channel accordingly.

I wrote a code for communication successfully.

AP :
```
   int newChannel = next_channel();
   printf("tell client to change channel to %d.\n",newChannel);  
   while(1){   
        if (sendto(sockfd, &newChannel, sizeof(newChannel), 0, &cliaddr, len) != sizeof(newChannel)) {
            printf("send failed.\n");  
            continue;
        }else{
            printf("send success. \n"); 
       	    // change_channel(newChannel);
            break;   
        }
    }
```

STA : 
```
    int newChannel;
    int n, len;
    n = recvfrom(sockfd, &newChannel, sizeof(newChannel), 0, (struct sockaddr *)&servaddr, (socklen_t *)&len);
    if (n > 0)
    {
        outprintf("Told by server to change channel to %d \n", newChannel);
        // change_channel(newChannel);
    }
```

change_channel:
```
void change_channel(int newChannel)
{
    while (1)
    {
        outprintf("change ch to %d\n", newChannel);
        if (esp_wifi_set_channel(newChannel, WIFI_SECOND_CHAN_NONE) == ESP_OK)
        {
            uint8_t primaryChan;
            wifi_second_chan_t secondChan;
            esp_wifi_get_channel(&primaryChan, &secondChan);
            printf("check ch  %d %d.\n", primaryChan, secondChan);
            printf("esp_wifi_set_channel success");
            if (primaryChan == newChannel)
            {
                printf("ful\n");
                break;
            }
            else
            {
                printf("less\n");
                continue;
            }
        }
        else
        {
            printf("esp_wifi_set_channel fail\n");
            continue;
        }
    }
}
```

When I commented the `// change_channel(newChannel);` in AP and STA , they communicate successfully.

but when I call `change_channel(newChannel);` (or `esp_restart()`) immediately after sending UDP packet. the UDP packet seems unsent. Even after it print `printf("send success. \n");`.

I research a lot and realize in ESP32 the `sendto(..........)` doesn't send data right a way. But it wait for some kernel interval to really send to.

So, I would like to know if there is anyway to call some function after the `sendto(..........)` really finished. Where I can do like 
`esp_wifi_set_channel` or `esp_restart`

Thank you in advance

zhangyanjiaoesp, 2021-03-16:
@rtmtree Can you provide a packet capture when the problem happen?

xueyunfei998, 2021-08-02:
hi @rtmtree 
It is recommended to delay for a period of time before switching channels after sending data

jack0c, 2022-04-24:
If the switch channel period is long(at least 3x beacon interval(102.4ms)), We can use CSA to change channel. 
That is when AP switch channel with CSA, station will switch to the new channel automatically.

jack0c, 2022-05-20:
@rtmtree I will close this issue, feel free to reopen it.

closed 2022-05-20 by jack0c