espressif/esp-idf #6714: How to use Multicore compilation？

zsjlovestudy, 2021-03-14:
The environment I use is VSCODE and ESPIDF development。
I found that the CPU usage was very low at compile time，but I can't find a place to add -J16.

igrr, 2021-03-16:
Hi @zsjlovestudy, ESP-IDF CMake based build system (which I assume you are using) will run `ninja` as the build tool. `ninja` auto-detects the number of CPU cores, and launches multiple build processes based on that. Please see https://github.com/espressif/esp-idf/issues/6406 for the answer to the same question.

Which host operating system are you using? If you are on Windows, then Windows Defender may be limiting the build performance.

zsjlovestudy, 2021-03-17:
While compiling the same project, I found that when I closed Window Defender it was much the same as when I had it on.
When I used MDK to compile STM32, CPU utilization is almost always 100%。
Now this only hits 100% once in a while.

projectgus, 2021-03-24:
Hi @zsjlovestudy,

Can you share please some log output from the build process? It will help confirm that it is running ninja as the build tool and not something else.

Is the average CPU utilization close to 100% during the build, or very low?

Do you run any other monitoring/anti-virus software? ESP-IDF has a *lot* of source files and spawns a lot of compilation processes, so anything that hooks/interrupts file access and/or process creation tends to slow the build down a lot.

Angus

Alvin1Zhang, 2021-04-02:
@zsjlovestudy Thanks for reporting, would you please help share more updates/details? Thanks.

KaeLL, 2021-04-05:
There is something wrong, though. The CMake based (CMake -> Make one, **not** CMake -> Ninja) build fails to saturate the cores and building [this](https://github.com/KaeLL/ESP32-W5100) simple app consistently ends up being over 20% slower than with Make on a 6 core machine. Each was run 10 times and I even tested with the app, repo and tools (except python stuff) on a tmpfs partition.
Commands:
Make : `make -j 8 all 1> /dev/null`, which takes ~18.5s
CMake: `idf.py build 1> /dev/null`, which takes ~23.1s

chegewara, 2021-04-05:
On my Ubuntu 20.04 CPU (all 12 cores) utilization is 100% almost whole build process, except few short times, but that is expected. I am checking with `System monitor` graphic tool.

projectgus, 2021-04-06:
@KaeLL Does the `idf.py build` step in your example include re-running CMake? This typically takes about 5 seconds but shouldn't be needed for any incremental builds, only the first one.

KaeLL, 2021-04-06:
@projectgus yes. Both CMake and Make tests I performed are complete rebuilds, as I rm the build/ folder before running them.

But I still don't think that's the problem because the poor saturation of the CPU is during the build itself, not when CMake is run in the beginning. Hopefully the images below will give some context. The following is the `htop` graph of CPU utilization during the build using `idf.py build`:
[image]
And this is running `make -j 8 all`
[image]
The CMake one shows gaps up until the 4th quarter of the test.

Let me know if there's an actual useful way of debugging this.

igrr, 2021-04-06:
The graphs look very interesting @KaeLL!

Another thing to try might be passing the same -j8 flag to Make, after using CMake to generate the Makefiles:
```
idf.py fullclean
idf.py -G "Unix Makefiles" reconfigure
time make -j8 -C build
```

For comparison, I'm seeing the following build times on my computer:

* `make -j8` takes 43 seconds
* `idf.py reconfigure` (which is the part of `build` that only generates the build scripts) takes 7.5 seconds on its own.
* `make -j8 -C build` takes 45 seconds (after doing `idf.py -G "Unix Makefiles" reconfigure`)
* `idf.py build` takes 42 seconds (after doing `idf.py -G "Unix Makefiles" reconfigure`)
* idf.py sets -j flag to 14 by default (`multiprocessing.cpu_count() + 2`). Ninja uses the same logic.
* `ninja -C build` takes 30 seconds (after doing `idf.py reconfigure`)
* `ninja -C build` (or `idf.py build`) takes 9 seconds (after doing `idf.py reconfigure`) with warm ccache.

KaeLL, 2021-04-06:
`make -j 8 -C build` still has poor saturation, and resulted in ~21.1s, which added to `idf.py -G "Unix Makefiles" reconfigure`'s ~2.1s, results on the >~23s I mentioned. I even tried the solo solo `-j` for infinite amount of jobs, and still the same. I don't have Ninja on my machine so I won't be testing it.

projectgus, 2021-04-08:
Hi @KaeLL,

That's very interesting.

Unfortunately I'm not sure what we can do to debug or work on this from the IDF side. After the initial "CMake" run ("idf.py reconfigure") finishes then we don't have much control over the Makefiles - CMake generates these to be simple/machine-generated and any inefficiencies in the CPU utilization are going to boil down to some interaction between particular version(s) of Make, other environment factors, and patterns in the Makefiles generated by CMake.

It might be possible to raise this as an issue with the CMake project themselves. I didn't find any similar issues in a cursory search, but it's possible everyone who cares about build efficiency on CMake has moved to Ninja.

> I don't have Ninja on my machine so I won't be testing it.

This is obviously up to you, but would you consider installing it? Ninja has very few dependencies, is an excellent match for CMake (where the build "generator" stage doesn't need to include complex logic), and I've never seen a system where it doesn't shave significant time off the build. Especially incremental builds, these are amazingly fast. However clean builds get faster as well.

KaeLL, 2021-04-08:
Ok, so I installed ninja and ran the same test again. Turns out you're right, it is much faster and apparently, it saturates the CPU completely, though still slower than Make by ~2.7% taking ~18.7s to build against Make's ~18.2, but still faster than CMake-Make combo by ~23.5%. There's definitely something wrong with the latter.

zsjlovestudy, 2021-04-09:
I am using the same machine in Ubuntu 20.04 as long as more than 10s, the whole CPU is 100%, and in Win10 takes a few minutes.

ajsb85, 2023-02-08:
Hi, @zsjlovestudy

Just try `ninja -C build -j 8`

Cheers.

ESP-Marius, 2024-08-15:
Seems like the original question here has already been answered here a while ago. So I think we can consider this one closed.

If there are still any outstanding issues with compilation speed not related to the original question that is still valid (and that we have control over and can fix) on IDF master then feel free to open a new issue regarding it.

closed 2024-08-15 by ESP-Marius