espressif/esp-idf #6715: ESP32S2 more RAM

wastene, 2021-03-14:
Hi,

i had no more space for RAM in my ESP32S2, which is why I searched for a solution. I have only used round about 180 kB with my application + sdk and in the data sheet is written that the ESP32S2 has internal SRAM of 320 kB. 

I have found out, that this is bounded by the boot loader, in this case at 0x3FFE0000, which results in 196608 Bytes (cache size not removed) available for the application. So im going to deeper and come to following solution for me:

I have just shift all blocks since the bootloader a few blocks back, so that the application has more space (=> 237568 Bytes).

Following are the values that I have changed in some linker files:
esp32s2.ld: 
- DATA_RAM_END: 0x3FFF0000 (old: 0x3FFE0000)

bootloader.ld:
- iram_seg: 0x4005C000 (old: 0x4004C000)
- iram_loader_seg: 0x40060000 (old: 0x40050000)
- dram_seg: 0x3FFF6000 (old: 0x3FFE6000)

esp32s2.rom.ld:
- _dram0_0_start: 0x3FFFAB00 (old: 0x3FFEAB00)

For me this changes are working fine. What is the reason, that the configuration is not like that. Or better what is the reason that the bootloader is at that specific location?

chegewara, 2021-03-14:
Hi @wastene 
did you consult it with this datasheet, page 19:
https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf

wastene, 2021-03-14:
I have used the chapter 1 „System and Memory“ from the technical reference (https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf) to calculate the new values.

chegewara, 2021-03-14:
@wastene i have to do something wrong, because all changes does not impact on free heap. Neither 2nd stage bootloader logs or heap tracking in app does not chow any changes.

wastene, 2021-03-14:
@chegewara cleaned the whole project and rebuilt it?

chegewara, 2021-03-14:
Yes, including delete build folder. Tested with my own project and with hello-world example.
```
diff --git a/components/bootloader/subproject/main/ld/esp32s2/bootloader.ld b/components/bootloader/subproject/main/ld/esp32s2/bootloader.ld
index 62b51c919..8020f1b17 100644
--- a/components/bootloader/subproject/main/ld/esp32s2/bootloader.ld
+++ b/components/bootloader/subproject/main/ld/esp32s2/bootloader.ld
@@ -5,9 +5,9 @@
 
 MEMORY
 {
-  iram_seg (RWX) :                  org = 0x4004c000, len = 0x4000  /* SRAM Block 13 */
-  iram_loader_seg (RWX) :           org = 0x40050000, len = 0x6000  /* SRAM Block 14 & part of 15 */
-  dram_seg (RW) :                   org = 0x3FFE6000, len = 0x4B00  /* Part SRAM Blocks 15 & 16, ROM static buffer starts at end of this region (reclaimed after app runs) */
+  iram_seg (RWX) :                  org = 0x4005c000, len = 0x4000  /* SRAM Block 13 */
+  iram_loader_seg (RWX) :           org = 0x40060000, len = 0x6000  /* SRAM Block 14 & part of 15 */
+  dram_seg (RW) :                   org = 0x3FFF6000, len = 0x4B00  /* Part SRAM Blocks 15 & 16, ROM static buffer starts at end of this region (reclaimed after app runs) */
 }
 
 /*  Default entry point:  */
diff --git a/components/esp32s2/ld/esp32s2.ld b/components/esp32s2/ld/esp32s2.ld
index 2ae61149f..f70234bb8 100644
--- a/components/esp32s2/ld/esp32s2.ld
+++ b/components/esp32s2/ld/esp32s2.ld
@@ -34,7 +34,7 @@
 #define RAM_IRAM_START    0x40020000
 #define RAM_DRAM_START    0x3FFB0000
 
-#define DATA_RAM_END      0x3FFE0000  /* 2nd stage bootloader iram_loader_seg starts at SRAM block 14 (reclaimed after app boots) */
+#define DATA_RAM_END      0x3FFF0000  /* 2nd stage bootloader iram_loader_seg starts at SRAM block 14 (reclaimed after app boots) */
 
 #define IRAM_ORG    (RAM_IRAM_START + CONFIG_ESP32S2_INSTRUCTION_CACHE_SIZE \
                                     + CONFIG_ESP32S2_DATA_CACHE_SIZE)
diff --git a/components/esp_rom/esp32s2/ld/esp32s2.rom.ld b/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
index 8781a9c8c..e2e6c646b 100644
--- a/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
+++ b/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
@@ -190,7 +190,7 @@ PROVIDE ( dmadesc_tx = 0x3fffeb40 );
 PROVIDE ( _DoubleExceptionVector = 0x400003c0 );
 PROVIDE ( _DoubleExceptionVector_text_end = 0x400003c6 );
 PROVIDE ( _DoubleExceptionVector_text_start = 0x400003c0 );
-PROVIDE ( _dram0_0_start = 0x3ffeab00 );
+PROVIDE ( _dram0_0_start = 0x3fffab00 );
 PROVIDE ( _dram0_rtos_reserved_start = 0x3ffffa10 );
 PROVIDE ( dummy_len_plus = 0x3ffffd54 );
 PROVIDE ( Enable_QMode = 0x40016690 );
```
hello-world example:
```
I (243) cpu_start: Project name:     hello-world
I (248) cpu_start: App version:      v4.3-dev-2136-gb0150615d-dirty
I (255) cpu_start: Compile time:     Mar 14 2021 19:22:01
I (261) cpu_start: ELF file SHA256:  badca86b52359b88...
I (267) cpu_start: ESP-IDF:          v4.3-dev-2136-gb0150615d-dirty
I (274) heap_init: Initializing. RAM available for dynamic allocation:
I (281) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (287) heap_init: At 3FFBF160 len 0003CEA0 (243 KiB): DRAM
I (294) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (300) spi_flash: detected chip: generic
I (305) spi_flash: flash io: dio
W (308) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (322) cpu_start: Starting scheduler on PRO CPU.
Hello world!
This is esp32s2 chip with 1 CPU core(s), WiFi, silicon revision 0, 2MB external flash
Minimum free heap size: 257712 bytes
```

wastene, 2021-03-14:
@chegewara I haven’t check it with heap. Need to check if there is a variable for it. Just check it with static ram, by putting a large variable in ram.

chegewara, 2021-03-14:
Can you provide sdkconfig and example code to test it?

chegewara, 2021-03-14:
When i remove changes i am having exactly the same logs. As you can see it is esp-idf clean v4.3 (no dirty):
```
I (46) boot: ESP-IDF v4.3-dev-2136-gb0150615d 2nd stage bootloader
I (46) boot: compile time 20:34:25
I (46) boot: chip revision: 0
I (49) boot.esp32s2: SPI Speed      : 80MHz
I (54) boot.esp32s2: SPI Mode       : DIO
I (59) boot.esp32s2: SPI Flash Size : 2MB
I (63) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  2 factory          factory app      00 00 00010000 00100000
I (102) boot: End of partition table
I (106) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=05adch ( 23260) map
I (120) esp_image: segment 1: paddr=00015b04 vaddr=3ffbc3a0 size=02624h (  9764) load
I (126) esp_image: segment 2: paddr=00018130 vaddr=40022000 size=00404h (  1028) load
I (132) esp_image: segment 3: paddr=0001853c vaddr=40022404 size=07adch ( 31452) load
I (149) esp_image: segment 4: paddr=00020020 vaddr=40080020 size=10bf8h ( 68600) map
I (163) esp_image: segment 5: paddr=00030c20 vaddr=40029ee0 size=024c0h (  9408) load
I (172) boot: Loaded app from partition at offset 0x10000
I (172) boot: Disabling RNG early entropy source...
I (185) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (186) cpu_start: Pro cpu up.
I (239) cpu_start: Pro cpu start user code
I (239) cpu_start: cpu freq: 160000000
I (240) cpu_start: Application information:
I (242) cpu_start: Project name:     hello-world
I (248) cpu_start: App version:      v4.3-dev-2136-gb0150615d
I (254) cpu_start: Compile time:     Mar 14 2021 20:34:20
I (260) cpu_start: ELF file SHA256:  1f67537359176386...
I (266) cpu_start: ESP-IDF:          v4.3-dev-2136-gb0150615d
I (273) heap_init: Initializing. RAM available for dynamic allocation:
I (280) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (286) heap_init: At 3FFBF160 len 0003CEA0 (243 KiB): DRAM
I (292) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
```

For big static data i am suggesting to check this option in menuconfig:
```
Name: ESP32S2_USE_FIXED_STATIC_RAM_SIZE
Prompt: Use fixed static RAM size
Type: bool
Value: n

Help:

  If this option is disabled, the DRAM part of the heap starts right after the .bss section,
  within the dram0_0 region. As a result, adding or removing some static variables
  will change the available heap size.

  If this option is enabled, the DRAM part of the heap starts right after the dram0_0 region,
  where its length is set with ESP32S2_FIXED_STATIC_RAM_SIZE
```

wastene, 2021-03-14:
@chegewara 
Im tested it now with the hello-world example, too. 

What I have added:
`
char test_buffer[150000] DRAM_ATTR = {0};

void app_main(void)
{
printf("TEST: %d\n", test_buffer[0]);
....
}
`

Without the changes I get:
region `dram0_0_seg' overflowed by 16344 bytes

With the changes it builds successfully.

chegewara, 2021-03-14:
You see, it is very important difference. I am sure espressif devs will explain it better, but i think there is reason whay those values are setup that way. Probably it is because upper 64kB is reserver for bss etc. The important part is here i think (read excluding part):
```
  /* Shared data RAM, excluding memory reserved for bootloader and ROM bss/data/stack. */
  dram0_0_seg (RW) :                 org = DRAM_ORG, len = I_D_RAM_SIZE - STATIC_RAM_SIZE
```

wastene, 2021-03-14:
Yeah that’s why I’m moved back the addresses for the bootloader and the ROM.

projectgus, 2021-03-24:
Hi @wastene,

Thanks for being patient while someone from our team got back to you.

The reason memory above the ROM symbol `_dram0_0_start` (0x3ffeab00) on ESP32-S2 is reserved is that some of this memory is used by the chip's internal ROM during early boot (i.e. when the chip loads the ESP-IDF bootloader). We try to place the IDF bootloader as close as possible to the end of RAM, but leave enough room that it doesn't overlap with this address. 

Then, when the bootloader loads the app we need to avoid overwriting the bootloader's own memory address range so the app must end before the bootloader starts.

Once the app starts up, both the bootloader memory range and most of the internal ROM memory range (everything up to _dram0_rtos_reserved_start 0x3ffffa10) are reclaimed by the app and added to the heap. I am guessing this may be why the total free heap size didn't change when @chegewara applied the same changes.

I can't say for sure why your bootloader and app still worked fine with the moved addresses, but my guess is that it would have failed to boot if Secure Boot was enabled - and possibly some other boot configurations won't work as expected either.

Unfortunately when linking static variables in the app, the regions for bss & data need to be single contiguous blocks so we can't put any of them after the start address of the bootloader. We may be able to do some tricks in the bootloader eventually to load the app much closer to the _dram0_rtos_reserved_start boundary, but for now the easiest way to use this memory is to make one or more of the large buffer(s) in your application allocated from heap at startup instead of allocated statically.

projectgus, 2021-05-20:
We have a Performance Guide section pending to be added to the ESP-IDF Programmers Guide. This includes some tips on how to optimize RAM usage. This issue will be updated once it merges.

closed 2021-06-06 by espressif-bot