espressif/esp-idf #6716: I2C lcd 1602 display library

mvarchdev, 2021-03-14:
**Is your feature request related to a problem? Please describe.**

Hello I searched all internet and have unsuccessfully found anything like library to controll i2c lcd 1602 16x2 display with esp idf.
Is there some example on how to use one? Thanks in advance.

**Describe the solution you'd like**

I would like to see an example of how to use esp idf with i2c lcd 1602 (PCF8574T)

**Describe alternatives you've considered**

Try to create library by my own.

Alvin1Zhang, 2021-03-15:
Thanks for raising this feature request.

igrr, 2021-03-15:
@mvarchdev Not sure why this didn't show up in search, please check it out: https://esp-idf-lib.readthedocs.io/en/latest/groups/pcf8574.html. It is part of https://github.com/UncleRus/esp-idf-lib.

mvarchdev, 2021-03-15:
I meant something more like library with functions "PrintString","BliknCursorOn" etc... I am working on mine library so once it will be done, I will post here repo link.

igrr, 2021-03-16:
Thanks @mvarchdev, I'm sure others looking for the same library will appreciate a link to your code, when you have it ready.

I'm going to close the issue since we are unlikely to add such peripheral driver components to esp-idf.

closed 2021-03-16 by igrr

mvarchdev, 2021-03-17:
So here is the link to library -> https://github.com/mvarchdev/lcd_i2c