espressif/esp-idf #6719: flash.rodata linked at 0x3f400040, how ?

rqbh, 2021-03-15:
I have done nothing to the ld script of esp-idf, but my cpp project cannot be made into .bin by esptool.
after my look for the reason carefully, found the reason:
flash.rodata section linked start from 3f400040, it should be start at 3f400020,  so esptool cannot do sha256 part,

link map of rodata section:

.flash.rodata   0x000000003f400040    0xae4a4
                       0x000000003f400040                _rodata_start = ABSOLUTE (.)
 *(.rodata_desc .rodata_desc.*)
 .rodata_desc  0x000000003f400040      0x100 esp-idf/app_update/libapp_update.a(esp_app_desc.c.obj)
                        0x000000003f400040                esp_app_desc
 *(.rodata_custom_desc .rodata_custom_desc.*)

esp32_out.ld part

/* Shared data RAM, excluding memory reserved for ROM bss/data/stack.

     Enabling Bluetooth & Trace Memory features in menuconfig will decrease
     the amount of RAM available.

     Note: Length of this section *should* be 0x50000, and this extra DRAM is available
     in heap at runtime. However due to static ROM memory usage at this 176KB mark, the
     additional static memory temporarily cannot be used.
  */
  dram0_0_seg (RW) : org = 0x3FFB0000 + 0xdb5c,
                                     len = (0x40000000 - 0x3FFB0000) - 0xdb5c
  /* Flash mapped constant data */
  drom0_0_seg (R) : org = 0x3F400020, len = 0x400000-0x20

igrr, 2021-03-15:
Hi @rqbh, we will need a bit more information to troubleshoot this. Please check the issue template you were presented when opening the issue, and fill in things such as: IDF version (commit ID), build system, your host OS, project sdkconfig file, etc.

Does the issue happen when building IDF example projects, or only your custom project?

rqbh, 2021-03-15:
Hi @igrr  , the project is based on master branch esp-mdf, all the file please ref https://github.com/rqbh/nav_light.git. 
Thanks a lot!

AxelLin, 2021-03-15:
@rqbh 

master branch could mean different commit at different timing.
Please provide "git describe --tags" output.

BTW, you can also run git bisect to figure out the bad commit.

igrr, 2021-03-15:
Could you attach the .map file you get in the build directory once the build is done, please?

As for my other questions:

* which build system are you using? The linked project has both a Makefile and CMakeLists.txt, not sure if you use `make` or `idf.py build`.
* what host OS are you using?

The linked project also disables -ffunction-sections and -fdata-sections compiler flags. I'm not sure this is related to this issue, but I recommend not disabling these flags globally, as IDF is not tested with these flags disabled.

If you need these flags to be disabled for some of your code, you can do so at component level or even source file level using `target_compile_options` or `set_source_files_properties` CMake functions (assuming you do use CMake).

rqbh, 2021-03-15:
@igrr  System os is  4.19.0-14-amd64 #1 SMP Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux
Using idf.py all
disable ffunction-sections and fdata-section is not my wanted, this maybe a my mistake

rqbh, 2021-03-15:
@AxelLin the output is v1.0-123-g26d134c

rqbh, 2021-03-16:
Hi, @igrr @AxelLin , any progress about the link problem ?

igrr, 2021-03-16:
It seems that some of the static libraries you are linking into the application are forcing larger section alignment than 32 bytes:

```
176 .bss._ZZN2cv7details12getLogTab32fEvE8logTab_f 00000800  00000000  00000000  0000a340  2**6
177 .rodata._ZN2cv7detailsL6logTabE 00001000  00000000  00000000  0000a340  2**6
179 .bss._ZZN2cv7details12getExpTab32fEvE8expTab_f 00000100  00000000  00000000  0000b340  2**6
180 .rodata._ZN2cv7detailsL6expTabE 00000200  00000000  00000000  0000b340  2**6
176 .bss._ZZN2cv7details12getLogTab32fEvE8logTab_f 00000800  00000000  00000000  0000a340  2**6
177 .rodata._ZN2cv7detailsL6logTabE 00001000  00000000  00000000  0000a340  2**6
179 .bss._ZZN2cv7details12getExpTab32fEvE8expTab_f 00000100  00000000  00000000  0000b340  2**6
180 .rodata._ZN2cv7detailsL6expTabE 00000200  00000000  00000000  0000b340  2**6
```
(2**6 means 64 byte alignment).

Seems to correspond to these lines:

https://github.com/opencv/opencv/blob/b19f8603843ac63274379c9e36331db6d4917001/modules/core/src/mathfuncs.cpp#L2055

https://github.com/opencv/opencv/blob/b19f8603843ac63274379c9e36331db6d4917001/modules/core/src/mathfuncs.cpp#L2145

https://github.com/opencv/opencv/blob/b19f8603843ac63274379c9e36331db6d4917001/modules/core/src/mathfuncs.cpp#L2411

and possibly some more, search for CV_DECL_ALIGNED(64) in the source code.

Supporting 64-byte rodata section alignment in ESP-IDF is non-trivial at the moment, since the offset of the application metadata is fixed, and esptool has certain restrictions on placement of rodata section.

You can try removing the alignment macros in OpenCV source code, as a workaround.

rqbh, 2021-03-16:
@igrr thanks a lot, i will give it a try tonight.

closed 2021-04-27 by espressif-bot

AxelLin, 2021-06-18:
Hi @igrr 

I found the commit "freertos: fix TLS run-time address calculation" https://github.com/espressif/esp-idf/commit/05351959837567d0badc9f4f4519cb14550ce7b1
is not included in v4.3. It's not clear to me if v4.3 needs fix or not.
(Since dd849ff is included in v4.3 so I think it might need fix).

 Can you clarify if v4.3 needs the "TLS run-time address calculation" fix or not?

igrr, 2021-06-18:
Hi @AxelLin. I think the fix https://github.com/espressif/esp-idf/commit/dd849ffc2628e7b1842f0ea5f3c5e7c655973de4 isn't backported to release/v4.3 yet. If you do see some related commit in release/v4.3 branch, please leave a link to it, I'll check again.
We have postponed merging the backport for this fix until after v4.3 is released, because at that time it was estimated that the occurrence of the issue is low, and the risk of merging linker script changes so late in the QA process wasn't justified. I am sorry if that caused you inconvenience. Now that v4.3 release has been done, we will start merging fixes into that branch again.