espressif/esp-idf #6720: flash.rodata link to wrong place < 3f400040>

rqbh, 2021-03-15:
I have done nothing to the ld script of esp-idf, but my cpp project cannot be made into .bin by esptool.
after my look for the reason carefully, found the reason:
flash.rodata section linked start from 3f400040, it should be start at 3f400020,  so esptool cannot do sha256 part,

link map of rodata section:

.flash.rodata   0x000000003f400040    0xae4a4
                       0x000000003f400040                _rodata_start = ABSOLUTE (.)
 *(.rodata_desc .rodata_desc.*)
 .rodata_desc  0x000000003f400040      0x100 esp-idf/app_update/libapp_update.a(esp_app_desc.c.obj)
                        0x000000003f400040                esp_app_desc
 *(.rodata_custom_desc .rodata_custom_desc.*)

esp32_out.ld part

/* Shared data RAM, excluding memory reserved for ROM bss/data/stack.

     Enabling Bluetooth & Trace Memory features in menuconfig will decrease
     the amount of RAM available.

     Note: Length of this section *should* be 0x50000, and this extra DRAM is available
     in heap at runtime. However due to static ROM memory usage at this 176KB mark, the
     additional static memory temporarily cannot be used.
  */
  dram0_0_seg (RW) : org = 0x3FFB0000 + 0xdb5c,
                                     len = (0x40000000 - 0x3FFB0000) - 0xdb5c
  /* Flash mapped constant data */
  drom0_0_seg (R) : org = 0x3F400020, len = 0x400000-0x20

Alvin1Zhang, 2021-03-15:
@rqbh Thanks for reporting, seems this ticket duplicates https://github.com/espressif/esp-idf/issues/6719, could we close one? Thanks.

rqbh, 2021-03-15:
@Alvin1Zhang Yes, please close this, network delay make mistake.

Alvin1Zhang, 2021-03-15:
@rqbh Thanks for the info.

closed 2021-03-15 by Alvin1Zhang