espressif/esp-idf #6724: Adding components to AWS FreeRTOS ESP32

Yoimer, 2021-03-16:
Is there any documentations that explains how can I add components to `FreeRTOS V202012.00`. A library such as [UncleRus](https://github.com/UncleRus/esp-idf-lib) is not easy to install on FreeRTOS.

Yoimer, 2021-03-16:
@Alvin1Zhang Thanks for taking your time to respond. I was able to install the `DHT11` sensor to AWS FreeRTOS ESP32 (after struggling a lot). and I can even send the data to AWS using the coreMQTT example. If that can help, please let me know.

chegewara, 2021-03-16:
@Yoimer have you seen this and related issues in AWSFreeRTOS repo?
https://github.com/aws/amazon-freertos/issues/1652

Yoimer, 2021-03-16:
@chegewara Hello. Thanks for replying. Actually I looked at your link, and unfortunately it is a bit outdated and I could not get too much help from it. However, I handled adding the `DHT` library to the latest AWS FreeRTOS ESP32. Please check my repo on the next link. There I explained how I handled it. Hope that can help anyone in those needs. There is not too much information available regarding this. 
https://github.com/Yoimer/aws_freertos_esp32_components/commit/f62417d901d8d34004929838f58e35a29dc850ad

Yoimer, 2021-03-16:
@chegewara @Alvin1Zhang Now my goal is to bring the `bmp180` from [bmp180](https://github.com/UncleRus/esp-idf-lib/tree/master/components/bmp180). I tried already using the very same procedure I executed on the previous link, but it failed due to a `i2cdev` library issue.

chegewara, 2021-03-16:
I didnt work with AWS for some time, but i believe it is still possible to start project with AWS as component. When you do that then you can add any other component just like in regular esp-idf.
As far as i remember this was very good starting point to do esp32 project with AWS as component:
https://github.com/chegewara/amazon-freertos-examples

Yoimer, 2021-03-16:
@chegewara Thanks for replying. Actually the link you sent about using the `amazon-freertos-examples` was a start point in order to accomplish what I did on the full `FreeRTOS V202012.00`. However when I tried using the BMP180 from https://github.com/UncleRus/esp-idf-lib, that failed. Do you happen to have any available (and reliable) `bmp180` library for ESP32 that I could add on AWS FreeRTOS ESP32?

shubhamkulkarni97, 2021-03-16:
Hi @Yoimer,

There are two ways of adding components in AFR:
* First one is similar to what you have done in https://github.com/Yoimer/aws_freertos_esp32_components/commit/f62417d901d8d34004929838f58e35a29dc850ad
* As @chegewara mentioned, second one is using AFR is a library. Please find https://github.com/tgsong/amazon-freertos-examples repo which uses AFR 202011.00. You can add external ESP-IDF components as demonstrated in the repo.

When creating a new component, you can add dependencies by setting COMPONENT_REQUIRES cmake variable. For instance, you can add `driver` in COMPONENT_REQUIRES to use I2C specific APIs.

If it doesn't work for you, please feel free to share error log.

Thanks,
Shubham

Yoimer, 2021-03-16:
@shubhamkulkarni97 Hello. Thanks for your support. I tried adding `bmp180` as seen on my commit
https://github.com/Yoimer/aws_freertos_esp32_components/commit/5d8cab8da03719045cdd2d57d9b10bd4381be48d

The error I am getting is this:
`[691/897] Building C object freertos/esp-idf/i2cdev/CMakeFiles/idf_component_i2cdev.dir/src/i2cdev.c.obj
FAILED: freertos/esp-idf/i2cdev/CMakeFiles/idf_component_i2cdev.dir/src/i2cdev.c.obj
C:\Users\Yoimer\.espressif\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DGCC_NOT_
_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v4.2\" -I../freertos-configs -I../components/i2cdev/include -Iconfig -I../components/esp_idf_lib_helpers/include -
/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/esp32/include -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/driver/in
de -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/esp_ringbuf/include -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/compone
/tcpip_adapter/include -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/lwip/include/apps -IC:/esp-idf-v4.2/FreeRTOS/vendors/espress
esp-idf/components/lwip/include/apps/sntp -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/lwip/lwip/src/include -IC:/esp-idf-v4.2/F
RTOS/vendors/espressif/esp-idf/components/lwip/port/esp32/include -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/lwip/port/esp32/i
ude/arch -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/lwip/include_compat -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/c
onents/vfs/include -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/esp_event/include -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/e
idf/components/log/include -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/efuse/include -IC:/esp-idf-v4.2/FreeRTOS/vendors/espress
esp-idf/components/efuse/esp32/include -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/newlib/platform_include -IC:/esp-idf-v4.2/Fr
TOS/vendors/espressif/esp-idf/components/newlib/include -I../freertos/vendors/espressif/boards/esp32/components/freertos/include -I../freertos/free
s_kernel/include -I../freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/include -I../freertos/vendors/espressif/boards/esp32/aws_d
s/config_files -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/heap/include -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/co
nents/soc/esp32/include -IC:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/soc/include -mlongcalls   -mlongcalls -Og -ffunction-sectio
-fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=un
d-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -MD -MT f
rtos/esp-idf/i2cdev/CMakeFiles/idf_component_i2cdev.dir/src/i2cdev.c.obj -MF freertos\esp-idf\i2cdev\CMakeFiles\idf_component_i2cdev.dir\src\i2cdev
obj.d -o freertos/esp-idf/i2cdev/CMakeFiles/idf_component_i2cdev.dir/src/i2cdev.c.obj   -c ../components/i2cdev/src/i2cdev.c
In file included from ../freertos/vendors/espressif/boards/esp32/components/freertos/include/freertos/semphr.h:1:0,
                 from C:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/components/driver/include/driver/i2c.h:26,
                 from ../components/i2cdev/include/i2cdev.h:15,
                 from ../components/i2cdev/src/i2cdev.c:14:
../components/i2cdev/src/i2cdev.c: In function 'i2cdev_done':
../components/i2cdev/src/i2cdev.c:27:62: error: 'CONFIG_I2CDEV_TIMEOUT' undeclared (first use in this function)
         if (!xSemaphoreTake(states[port].lock, pdMS_TO_TICKS(CONFIG_I2CDEV_TIMEOUT))) \
                                                              ^
../freertos/freertos_kernel/include/semphr.h:296:92: note: in definition of macro 'xSemaphoreTake'
 #define xSemaphoreTake( xSemaphore, xBlockTime )    xQueueSemaphoreTake( ( xSemaphore ), ( xBlockTime ) )
                                                                                            ^
../components/i2cdev/src/i2cdev.c:27:48: note: in expansion of macro 'pdMS_TO_TICKS'
         if (!xSemaphoreTake(states[port].lock, pdMS_TO_TICKS(CONFIG_I2CDEV_TIMEOUT))) \
                                                ^
../components/i2cdev/src/i2cdev.c:67:13: note: in expansion of macro 'SEMAPHORE_TAKE'
             SEMAPHORE_TAKE(i);
             ^
../components/i2cdev/src/i2cdev.c:27:62: note: each undeclared identifier is reported only once for each function it appears in
         if (!xSemaphoreTake(states[port].lock, pdMS_TO_TICKS(CONFIG_I2CDEV_TIMEOUT))) \
                                                              ^
../freertos/freertos_kernel/include/semphr.h:296:92: note: in definition of macro 'xSemaphoreTake'
 #define xSemaphoreTake( xSemaphore, xBlockTime )    xQueueSemaphoreTake( ( xSemaphore ), ( xBlockTime ) )
                                                                                            ^
../components/i2cdev/src/i2cdev.c:27:48: note: in expansion of macro 'pdMS_TO_TICKS'
         if (!xSemaphoreTake(states[port].lock, pdMS_TO_TICKS(CONFIG_I2CDEV_TIMEOUT))) \
                                                ^
../components/i2cdev/src/i2cdev.c:67:13: note: in expansion of macro 'SEMAPHORE_TAKE'
             SEMAPHORE_TAKE(i);
             ^
[... cut ...]
                                                                                            ^
../components/i2cdev/src/i2cdev.c:27:48: note: in expansion of macro 'pdMS_TO_TICKS'
         if (!xSemaphoreTake(states[port].lock, pdMS_TO_TICKS(CONFIG_I2CDEV_TIMEOUT))) \
                                                ^
../components/i2cdev/src/i2cdev.c:231:5: note: in expansion of macro 'SEMAPHORE_TAKE'
     SEMAPHORE_TAKE(dev->port);
     ^
[693/897] Linking C static library freertos\esp-idf\foo\libfoo.a
ninja: build stopped: subcommand failed.
`
Any ideas? Thanks in advance.

shubhamkulkarni97, 2021-03-17:
Hi @Yoimer,

Are you using FreeRTOS V202012.00 or master branch of AFR?

Support for ESP-IDF v4.2 was merged after FreeRTOS V202012.00 and I can see that you are using ESP-IDF v4.2 from the logs.

There are a lot of build system changes that should be considered when using ESP-IDF v4.2

Yoimer, 2021-03-17:
@shubhamkulkarni97 Hello. I am using `FreeRTOS V202012.00` from AWS Console

Yoimer, 2021-03-17:
@shubhamkulkarni97 Technically speaking, I could also use this [202012.00]https://github.com/aws/amazon-freertos/releases/tag/202012.00) from Github too. But of course, compiling time would increase since this zip file includes all the libraries such as `greengrass` `cli` etc.

shubhamkulkarni97, 2021-03-17:
@Yoimer, I tried cloning your repo and found that IDF code base uses v3.3. I'm not sure why `IDF_VER=\"v4.2\"` is passed in the build logs you shared above.

I tried debugging build failures and found that value for `CONFIG_I2CDEV_TIMEOUT` is not set In source file `i2cdev.c`. Creating a macro by `#define CONFIG_I2CDEV_TIMEOUT 10` in `i2cdev.c` should fix build issues.

Hope this helps!

Yoimer, 2021-03-17:
@shubhamkulkarni97 Hello there and thanks for replying, I will try this right now. In fact, I am using `IDFV4.2`. However, I integrated it with `FreeRTOS V202012.00` as explained next:
On `C:\Users\Yoimer\.espressif\idf_cmd_init.bat` I modified the `IDF_PATH` variable as seen on snapshot. The first time I ran `ESP-IDF Command Prompt`, it automatically executed the `install.bat` file from `FreeRTOS V202012.00` and install the toolchain and dependencies needed. If I ever want to stop using FreeRTOS (and use ESP-IDF examples), I just go to `C:\Users\Yoimer\.espressif\idf_cmd_init.bat` and set the `IDF_PATH` variable as it came by default.

[image]

Yoimer, 2021-03-17:
@shubhamkulkarni97 By the way. Where can I see the `IDF code base uses v3.3` from my repo? Thanks in advance

shubhamkulkarni97, 2021-03-17:
@Yoimer Please check IDF version defined in esp_idf_version.h at https://github.com/Yoimer/aws_freertos_esp32_components/blob/7a21cc7a87148719e6c10ba9d1df4874496b7403/vendors/espressif/esp-idf/components/esp32/include/esp_idf_version.h#L21

It shows you are using IDF v3.3.2

In Amazon FreeRTOS, IDF build system is plugged in to AFRs build system and IDF is used as a library. So, changes in [above comment](https://github.com/espressif/esp-idf/issues/6724#issuecomment-801056317) should not be required and changing IDF_PATH to point to v4.2 will not work.

With FreeRTOS V202012.00, you will have to use IDF v3.3.2 that is downloaded with Amazon FreeRTOS.
If you want to use IDF v4.2, you will have to switch to master branch of Amazon FreeRTOS (https://github.com/aws/amazon-freertos/tree/master)

Please find getting started guide for ESP32 with Amazon FreeRTOS: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

Yoimer, 2021-03-17:
@shubhamkulkarni97 Thanks a lot for you quick and efficient support. You have no idea how valuable this has been to me. Thanks a lot

Yoimer, 2021-03-17:
@shubhamkulkarni97 It works now!! Thanks a lot again. Have a wonderful day.

Alvin1Zhang, 2021-03-18:
Thanks for reporting and glad it works now, feel free to reopen.

closed 2021-03-18 by Alvin1Zhang

Yoimer, 2021-03-18:
@shubhamkulkarni97 Hello there again. Checking the link you recommended  me https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html, they say their current AWS FreeRTOS ESP32 supports `IDFV3.3` but we can see here https://github.com/Yoimer/aws_freertos_esp32_components/blob/7a21cc7a87148719e6c10ba9d1df4874496b7403/vendors/espressif/esp-idf/components/esp32/include/esp_idf_version.h#L21 that it says it supports `IDFV3.2` as you gently mentioned. So which one is the right to use?. That part is a little bit confusing. Thanks in advance

[image]

shubhamkulkarni97, 2021-03-19:
@Yoimer,

In [above comment](https://github.com/espressif/esp-idf/issues/6724#issuecomment-801073105), I had mentioned AFR uses IDF v3.3.2 which is incremental version of IDF v3.3. You can consider it as v3.3.

https://github.com/Yoimer/aws_freertos_esp32_components/blob/7a21cc7a87148719e6c10ba9d1df4874496b7403/vendors/espressif/esp-idf/components/esp32/include/esp_idf_version.h#L21 shows `ESP_IDF_VERSION_MAJOR` is 3, `ESP_IDF_VERSION_MINOR` is 3 and `ESP_IDF_VERSION_PATCH` is 2

Yoimer, 2021-03-19:
@shubhamkulkarni97 Thanks a lot for replying. So, one last question. Since `AFR uses IDF v3.3.2`,

Which `IDF` is the best to use? https://docs.espressif.com/projects/esp-idf/en/v3.3.2/get-started/index.html `IDFV3.3.2` or

`IDFV3.3` https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started/index.html?

Or either both work fine?

Thanks for your support.

shubhamkulkarni97, 2021-03-19:
@Yoimer, We have tested AFR with IDF v3.3.2, so v3.3.2 is recommended.

Yoimer, 2021-03-19:
@shubhamkulkarni97 Thanks a lot for clarifying. I rate you 5 starts on this.

chegewara, 2021-03-19:
Can this discussion be moved to forum, as issue seems to be resolved?

Yoimer, 2021-03-19:
@chegewara it's already resolved. Thanks to @shubhamkulkarni97

Yoimer, 2021-03-20:
@chegewara It could be. I've got a few doubts yet (sorry)

Yoimer, 2021-03-20:
@shubhamkulkarni97 Hello there. This is getting really exciting :). 

I just installed `IDFV3.3.2` as explained here https://docs.espressif.com/projects/esp-idf/en/v3.3.2/get-started/index.html, including the Windows Toolchain https://docs.espressif.com/projects/esp-idf/en/v3.3.2/get-started/windows-setup.html on Windows 7 64Bits. All went fine. However, `AWS`  mentions here https://docs.aws.amazon.com/freertos/latest/userguide/getting-started-cmake.html that `AFR` needs `CMake` for its installation, and `IDFV3.3.2` did not come with any CMake ( at least using the https://docs.espressif.com/projects/esp-idf/en/v3.3.2/get-started/windows-setup.html toolchain mention here).
So my question is: How could I possible compile `AFR FreeRTOS V202012.00` which is from my repo https://github.com/Yoimer/aws_freertos_esp32_components/blob/7a21cc7a87148719e6c10ba9d1df4874496b7403/vendors/espressif/esp-idf/components/esp32/include/esp_idf_version.h#L21 without `CMake`?.

I am asking all of this since I am quite curious about how to compile `AFR using IDFV3.3.2`. I have done that but using `IDFV4.2` as seen here https://github.com/espressif/esp-idf/issues/6724#issuecomment-801056317.

Thanks for your time.