espressif/esp-idf #6725: needed by 'hello-world.elf', missing and no known rule to make it

JerryCloud, 2021-03-16:
system：ubantu16.04
idf-version:v4.3
tool chain: extensa-esp32-elf-gcc:xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

build hello_world example error

idf.py set-target esp32    # OK
idf.py menuconfig           #OK
idf.py build                       #error

ninja: error: '/home/jerry/esp/esp-idf/components/esp_wifi/lib/esp32/libwapi.a', needed by 'hello-world.elf', missing and no known rule to make it

please help me

chegewara, 2021-03-16:
Hi @JerryCloud 
i think this issue is similar to the one you are having:
https://github.com/espressif/esp-idf/issues/6690

You didnt clone esp-idf properly with `--recursive` flag or didnt update submodules.

JerryCloud, 2021-03-16:
@chegewara thanks.
I use this,but error too.

git submodule update --init --recursive

fatal: reference is not a tree: b967ca054f48a36f82d8fcdd32e54ec5144f2751
Unable to checkout 'b967ca054f48a36f82d8fcdd32e54ec5144f2751' in submodule path 'components/nghttp/nghttp2/third-party/neverbleed'
Failed to recurse into submodule path 'components/nghttp/nghttp2'

chegewara, 2021-03-16:
Probably that module is somehow changed and cant be updated. Please try:
`git status'
You can also try `git stash`. When you fix changes in that submodule, you can try to update submodules again.

chegewara, 2021-03-16:
This may help you even better:
https://stackoverflow.com/questions/2155887/git-submodule-head-reference-is-not-a-tree-error

ankit-thealchemist, 2021-03-16:
> system：ubantu16.04
> idf-version:v4.3
> tool chain: extensa-esp32-elf-gcc:xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
>
> build hello_world example error
>
> idf.py set-target esp32 # OK
> idf.py [...]

Exactly this problem is faced by me. I just deleted whole esp idf directory and freshly cloned from the repo.

JerryCloud, 2021-03-16:
@ankit-thealchemist thanks.
 let me try again

JerryCloud, 2021-03-16:
@chegewara I've already fixed the problem.
it's my process:
1.
#download  esp-idf v4.2
git clone -b v4.2 --recursive https://github.com/espressif/esp-idf.git  

#it takes a long time
/install. sh 

#make sure all submodules are installed
git submodule update --init --recursive  

#switch v4.3
git fetch
git checkout release/v4.3
git pull
git submodule update --init --recursive

The following questions will appear:

fatal: reference is not a tree: b967ca054f48a36f82d8fcdd32e54ec5144f2751
Unable to checkout 'b967ca054f48a36f82d8fcdd32e54ec5144f2751' in submodule path 'components/nghttp/nghttp2/third-party/neverbleed'
Failed to recurse into submodule path 'components/nghttp/nghttp2'

so，I switch v4.2：
git fetch
git checkout release/v4.2
git pull
git submodule update --init --recursive

idf.py set-target esp32 # OK
idf.py menuconfig #OK
idf.py build #ok

How should I switch to the latest stable version？

chegewara, 2021-03-16:
@JerryCloud Please try this:
```
rm -rf components/nghttp/nghttp2
git submodule update --init --recursive
```

projectgus, 2021-03-22:
Hi @JerryCloud,

I'm not sure why you're seeing the "fatal: reference is not a tree: b967ca054f48a36f82d8fcdd32e54ec5144f2751". I tried following the steps above but in my version of git it can fetch, it may be some issue with git.

If it's still not working after trying @chegewara's suggestion and you get the same error, try this:

```
cd components/nghttp/nghttp2/third-party/neverbleed
git fetch
git checkout b967ca054f48a36f82d8fcdd32e54ec5144f2751
```

Then go back and try to run `git submodule update --init --recursive` again.

> How should I switch to the latest stable version？

For what it's worth, the `v4.2` is the latest stable release. The `release/v4.x` branches are what we call "release branches" which contain pre-release code, they are more stable than master branch but less stable than a release.

So if you want the latest stable version, checking `v4.2` and using it is the best approach.

You can read more about this here: 
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/versions.html

projectgus, 2021-04-06:
Closing as the original issue seems to be resolved, and no follow-up from reporter.

closed 2021-04-06 by projectgus